freeacs/freeacs #13: Checking public IP in Kick class must be configurable

pingwiniasty, 2018-05-10:
This line is checking if the connection URL is on public IP. But in modern networks, communication with device to configure it, is on dedicated VLAN with private IP. Maybe You can add option in configuration to disable checking Public IP or remove this totally from code?

https://github.com/freeacs/freeacs/blob/efc4a61149fcbbe4b77f1d3cb6367e3b19954c14/stun/src/main/java/com/owera/xaps/stun/Kick.java#L75

 In our modification I removed logic from method isPublic from 
`common com/owera/common/util/IPAddress.class`

jarlah, 2018-05-10:
I will take a look at it :) Its on the top on my list, but its also a low hanging fruit that you could potentially open a pull request for yourself. See my other comment about intellij in the other issue you made.

jarlah, 2018-05-13:
I will make a release candidate soon, that contains this fix.

PR #39 (2018-05-13): adding feature toggle for checking if ip is public before kicking
Closes #13 
When deploying stun service, change freeacs-stun.conf like this:
```
JAVA_OPTS="-DLOG_PATH=logs/ -Dkick.check-public-ip=false"
```
We might make it default, but now it must be explicitely disabled.

I have made unit tests that prove the fix.
closed 2018-05-13 by jarlah