freeacs/freeacs #16: Issues regarding rewrite to spring boot and lightbend config <- place them here

jarlah, 2018-05-10:
I would like to avoid creating many issues, so lets try to summarize them here and extract those issues that are real issues.

jarlah, 2018-05-13:
I have fixed a bug with the quirks properties in tr069 in another issue. Needed to replace the @ delimiter for [Unittype]@[Version] with [Unittype]_[Version]. Also, spaces in unittype name must be unicode like \u0020

pingwiniasty, 2018-05-14:
How to enable logging to tr069-conversation.log?
I'm trying to discover 2 different devices, but it not working

jarlah, 2018-05-14:
What do you get (that is relevant) in tr069.lov?

jarlah, 2018-05-14:
I introduced a quick fix on the method name extractor that i have now refined. So i am planning to release å quick RC5. To filter out the possobility that the method name extractor in RC4 is bad. (I know it works but just for good safety).

Conversation log should work by default. But maybe there is some error in tr069 log.

jarlah, 2018-05-14:
Try this candiate:
https://github.com/freeacs/freeacs/releases/tag/freeacs-2.0.0-RC5
Just copy the jar file and restart service.

jarlah, 2018-05-14:
You might need to set auth.mode to none in freeacs-tr069.conf with JAVA_OPTS=".... -Dauth.mode=none" . I think default is that discover is on by default. But your devices will get basic challenge even if auth mode is none. So they must get pass authentication anyhow

pingwiniasty, 2018-05-14:
In tr069.log I have:

```
2018-05-14 11:11:07 [localhost-startStop-1] INFO  c.github.freeacs.tr069.Provisioning - Server starts...
2018-05-14 11:11:07 [localhost-startStop-1] WARN  com.github.freeacs.base.db.DBAccess - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 11:11:08 [localhost-startStop-1] WARN  com.github.freeacs.base.db.DBAccess - The value of db.syslog.maxage was not specified, instead using default value 600000
2018-05-14 11:11:08 [localhost-startStop-1] WARN  com.github.freeacs.base.db.DBAccess - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 11:11:08 [localhost-startStop-1] WARN  com.github.freeacs.base.db.DBAccess - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 11:11:09 [StabilityLogger] WARN  com.github.freeacs.tr069.Properties - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 11:11:09 [ActiveDeviceDetection TR069] INFO  c.g.f.t.b.ActiveDeviceDetectionTask - ActiveDeviceDetectionTask: Will check if some devices scheduled to return before Mon May 14 10:11:09 CEST 2018 are too late
2018-05-14 11:11:09 [ActiveDeviceDetection TR069] INFO  c.g.f.t.b.ActiveDeviceDetectionTask - ActiveDeviceDetectionTask: Have found 0 inactive devices
2018-05-14 11:11:11 [main] INFO  com.github.freeacs.App - Started App in 9.836 seconds (JVM running for 10.634)
2018-05-14 11:11:18 [StabilityLogger] WARN  com.github.freeacs.tr069.Properties - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 11:11:22 [http-nio-8085-exec-1] INFO  c.g.f.base.http.BasicAuthenticator - Send challenge to CPE, located on IP-address 10.90.7.169
2018-05-14 11:11:22 [http-nio-8085-exec-1] INFO  c.g.f.base.http.BasicAuthenticator - Send challenge to CPE, located on IP-address 10.90.7.169
2018-05-14 11:11:28 [StabilityLogger] WARN  com.github.freeacs.tr069.Properties - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 11:11:38 [StabilityLogger] WARN  com.github.freeacs.tr069.Properties - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 11:11:40 [http-nio-8085-exec-3] WARN  com.github.freeacs.base.db.DBAccess - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 11:11:48 [StabilityLogger] WARN  com.github.freeacs.tr069.Properties - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 11:11:57 [http-nio-8085-exec-2] INFO  c.g.f.base.http.BasicAuthenticator - Send challenge to CPE, located on IP-address 10.90.7.169
2018-05-14 11:11:57 [http-nio-8085-exec-2] INFO  c.g.f.base.http.BasicAuthenticator - Send challenge to CPE, located on IP-address 10.90.7.169
2018-05-14 11:11:58 [StabilityLogger] WARN  com.github.freeacs.tr069.Properties - The value of db.xaps.maxage was not specified, instead using default value 600000
```
In old ACS I have `auth.mode=digest` `discovery.mode = true` and this device was successful added.

jarlah, 2018-05-14:
Yeah dont mind the auth mode tips. I am just wondering how i could reproduce this problem. At least you are not getting any error. I need to read up on the docs regarding digest mode. I havent changed anything regarding this so i need to figure out how it works and what it expects

jarlah, 2018-05-14:
AND .. we need to remove all these warnings.. its pretty much DEBUG more than WARN for these default value warning messages...

jarlah, 2018-05-14:
@pingwiniasty pro tip. Extract the logback.xml from the jar file and place it in /var/freeacs/tr069. Change the loglevel for com.github.freeacs to DEBUG and add -Dlogging.config=logback.xml to freeacs-tr069.conf. Then restart

jarlah, 2018-05-14:
Planning to Add support for overriding standard log Levels like this via environment variables without having to dig out the log file and manipulate it

pingwiniasty, 2018-05-14:
I'm trying to find differences between old and new version in packet.
New version sends response with doubled WWW-Authenticate header
[image]

and device not respond to it

pingwiniasty, 2018-05-14:
I don't know why, but it communicate now with ACS, but without adding new device.
```
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid 604BA5E3CBAA1A041675C6ABD8FBED4D did not return a SessionData object from cache, must create a new SessionData object
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid 604BA5E3CBAA1A041675C6ABD8FBED4D created: 1526294444931, lastAccess:1526294444931, mxInactiveInterval:1800
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.f.base.http.BasicAuthenticator - Basic verification of CPE starts, located on IP-address 10.90.7.164
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.f.base.http.DigestAuthenticator - Basic verification identifed unit id 000EB4-R3621-000EB4100020 from CPE IP-address 10.90.7.164
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG com.github.freeacs.tr069.SessionData - Will load unit data
2018-05-14 12:40:44 [http-nio-80-exec-8] WARN  com.github.freeacs.base.db.DBAccess - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.f.common.db.ConnectionProvider - Reusing connection com.mysql.cj.jdbc.ConnectionImpl@1e3aedff
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.f.common.db.ConnectionProvider - Connection returned in 1 ms
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0,84 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, up.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = '000EB4-R3621-000EB4100020' 
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.f.common.db.ConnectionProvider - Connection com.mysql.cj.jdbc.ConnectionImpl@1e3aedff is returned to the pool
2018-05-14 12:40:44 [http-nio-80-exec-8] WARN  com.github.freeacs.tr069.SessionData - Did not find unit in unit-table, nothing exists on this unit
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG com.github.freeacs.tr069.SessionData - No unit data found & discovery mode true -> first-connect = true, allow to continue
2018-05-14 12:40:44 [http-nio-80-exec-8] WARN  com.github.freeacs.tr069.Properties - The value of discovery.block was not specified, instead using default value null
2018-05-14 12:40:44 [http-nio-80-exec-8] WARN  c.g.f.base.http.DigestAuthenticator - Authentication not verified, but accepted since in Discovery Mode
2018-05-14 12:40:44 [http-nio-80-exec-8] INFO  c.g.f.base.http.BasicAuthenticator - Authentication verified (CPE IP address: 10.90.7.164)
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG com.github.freeacs.common.util.Cache - Key 10.90.7.164Basic MDAwRUI0LVIzNjIxLTAwMEVCNDEwMDAyMDo4MTAxMjM5OTc5Nzk= will be removed from Cache (explicitely - not by Cache.Cleanup.run())
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.g.f.t.methods.HTTPRequestProcessor - Will process method Empty (incoming request/response from CPE)
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG com.github.freeacs.tr069.Properties - The unittypename (null) could not be found. The quirk prettyprint will return default false
2018-05-14 12:40:44 [http-nio-80-exec-8] ERROR c.g.freeacs.tr069.methods.EMDecision - EM-Decision is EM since the CPE did not send an INFORM (or sessionId was not sent by client)
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG com.github.freeacs.tr069.Properties - The unittypename (null) could not be found. The quirk termination will return default false
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG c.github.freeacs.tr069.Provisioning - End of session is reached, will write queued unit parameters if unit (null) is not null
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG com.github.freeacs.common.util.Cache - Key 000EB4-R3621-000EB4100020SESSION will be removed from Cache (explicitely - not by Cache.Cleanup.run())
2018-05-14 12:40:44 [http-nio-80-exec-8] DEBUG com.github.freeacs.common.util.Cache - Key 604BA5E3CBAA1A041675C6ABD8FBED4DSESSION will be removed from Cache (explicitely - not by Cache.Cleanup.run())
```
In conversation.log there are no response from ACS
https://pastebin.com/saA75Prg

pingwiniasty, 2018-05-14:
Maybe 
```
2018-05-14 12:40:44 [http-nio-80-exec-8] ERROR c.g.freeacs.tr069.methods.EMDecision - EM-Decision is EM since the CPE did not send an INFORM (or sessionId was not sent by client)
```
is related to https://github.com/freeacs/freeacs/issues/16#issuecomment-388765891

jarlah, 2018-05-14:
I'll look at it

jarlah, 2018-05-14:
@pingwiniasty simple to reproduce ..
[image]
This will be very simple to fix. I think its becaue of changes in how the servlet api works. Old tr069 used servlet 2.5 ... this version uses the latest version servlet api on modern embedded tomcat inside latest spring boot.

jarlah, 2018-05-14:
Cookie path is following the spring boot context. But i can investigate how to always set it /

jarlah, 2018-05-14:
And we should either always log requests or have a feature toggle. Do you agree @pingwiniasty ?

jarlah, 2018-05-14:
@pingwiniasty  i propose a change. To log the cpe request before checking authentication. This way we will see all requests coming from any device. This will hugely increase the log size if the server is ddos attacked, but will likely be more informative in terms of transparency of whats going on between server and client. What do you think? Feature toggle it?  Or hard code always log cpe request before anything else is done with the request?

jarlah, 2018-05-14:
This is definitely interesting. It's a major pain to not see what content the cpe is sending without first having an authenticated seesion. By feature toggling eager logging of request data we can stop being reliable upon wireshark for capturing cpe request payloads. In addition to the request xml we could also log the request headers sent from the devices to enable easy replication of request in Postman

pingwiniasty, 2018-05-14:
@jarlah logging requests may be configurable, it's only needed when something doesn't work. If you want logging all req. before authentication - this is must be configurable too, we don't want to kill disk on server :D Some CPEs sends stupid request (like ZyXEL) without checking inform time. 

I will test RC6

pingwiniasty, 2018-05-14:
Not working, the same issue with both things. Maybe my device not accept HttpOnly in set-cookie?

```

2018-05-14 20:12:57 [http-nio-80-exec-4] WARN  com.github.freeacs.tr069.SessionData - Did not find unit in unit-table, nothing exists on this unit
2018-05-14 20:12:57 [http-nio-80-exec-4] WARN  com.github.freeacs.tr069.Properties - The value of discovery.block was not specified, instead using default value null
2018-05-14 20:12:57 [http-nio-80-exec-4] WARN  c.g.f.base.http.DigestAuthenticator - Authentication not verified, but accepted since in Discovery Mode
2018-05-14 20:12:57 [http-nio-80-exec-4] INFO  c.g.f.base.http.BasicAuthenticator - Authentication verified (CPE IP address: 10.90.7.164)
2018-05-14 20:12:57 [http-nio-80-exec-4] ERROR c.g.freeacs.tr069.methods.EMDecision - EM-Decision is EM since the CPE did not send an INFORM (or sessionId was not sent by client)
```

pingwiniasty, 2018-05-14:
Sure, I know that httpOnly is for security reasons, maybe think about write "quirk" for this issue and disables httpOnly only for some devices?

jarlah, 2018-05-14:
Right now making a quirk for this is too much work. Lets try getting a stable artifact and then we can think about advanced stuff like this quirks :) And AddObject  ;)

jarlah, 2018-05-14:
@pingwiniasty try this
https://github.com/freeacs/freeacs/releases/tag/freeacs-2.0.0-RC7
Its a lot of back and forth, but i am doing this on my spare time with dslam and no devices at home, so I am grasping at straws you throw at me :)

jarlah, 2018-05-14:
but i have easycwmp in docker that is. Nice testing tool

pingwiniasty, 2018-05-14:
HttpOnly is not a problem.
Does ACS can response at first time with 204 code? Not 200 with inform response?

If you need some devices to test ACS, maybe I can get you access to some our devices? But I don't want to promise that it will be tomorrow. Or we can talk about it, when you done more important changes, then I can send info about project reactivation to some my colleagues

pingwiniasty, 2018-05-14:
I looking at my pcaps from old service and after 401, device sends the same request but with auth header and acs responses with 200
```
    <soapenv:Envelope
        xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/"
        xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/"
        xmlns:xsd="http://www.w3.org/2001/XMLSchema"
        xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
        xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
        <soapenv:Header>
            <cwmp:ID
                soapenv:mustUnderstand="1">
                inform
                </cwmp:ID>
            </soapenv:Header>
        <soapenv:Body>
            <cwmp:InformResponse>
                <MaxEnvelopes>
                    1
                    </MaxEnvelopes>
                </cwmp:InformResponse>
            </soapenv:Body>
        </soapenv:Envelope>
```

jarlah, 2018-05-14:
is it possible for you capture the headers and payload that the CPE sends to the ACS server? Then I can try to replicate the behaviour locally. So basically, each request + headers that the cpe sends. Not so much important what the server sends, but by all means if no problem send acs responses as well. you can choose to paste it here or send to jarl.andre (at) gmail.com

jarlah, 2018-05-14:
i need to learn how to open pcacp files, but until then I more happy with good old text

pingwiniasty, 2018-05-14:
pcap file you can open with wireshark :)

jarlah, 2018-05-14:
ok @pingwiniasty attach it here and i will install wireshark .. (very simple task in manjaro hopefully)

jarlah, 2018-05-14:
by the way .. the acs server only sends Empty aka no content (204) on error .. or when it resolves that the method is Empty. For ex when client sends Empty the acs sends Empty. If an error occurs it sends Empty. But if you send the pcap files i can investigate and begin to create a postman collection to reproduce the behaviour.

pingwiniasty, 2018-05-14:
@jarlah check mail :)

pingwiniasty, 2018-05-14:
If you don't want waste your time to this, I will try to debug it by myself tomorrow in work. Maybe I find some solution :thinking: 

I know, why this device sometimes sends auth header at inform. When I'm looking for differences, I replaced ACS URL from NEW to OLD, then challenge was correct, device was added to OLD ACS and CPE saved info about this, and sends authorization header in each request. When I changed it back to NEW URL, device sends Inform with auth header, and ACS logs this
`2018-05-14 12:40:44 [http-nio-80-exec-8] ERROR c.g.freeacs.tr069.methods.EMDecision - EM-Decision is EM since the CPE did not send an INFORM (or sessionId was not sent by client)`

jarlah, 2018-05-14:
decernable  difference is that old server sent tomcat error message on 401 Unauthorized. New server sends zero content on 401

jarlah, 2018-05-14:
Line-based text data: text/html (1 lines)
     [truncated]<html><head><title>Apache Tomcat/7.0.68 (Ubuntu) - Error report</title><style>

jarlah, 2018-05-14:
but thanks for requests .. they helped me understand. I might need to return something on 401.

jarlah, 2018-05-14:
Ah and the new server also doesnt send 401 message .. it just says 401 ... i will try to send with a text message, and set 401 Unauthorized instead of just 401

jarlah, 2018-05-14:
Try this. It will now always send body "Unauthorized if not authenticated. It also sets content type to text/html;charset=utf-8
[tr069-2.0.1-SNAPSHOT-bin.zip](https://github.com/freeacs/freeacs/files/2002584/tr069-2.0.1-SNAPSHOT-bin.zip)
Not releasing this because its just a test.

pingwiniasty, 2018-05-14:
Nope, this is not this. 
I will run test server on my laptop and debug something. Maybe we have some troubles with test server, I don't know

jarlah, 2018-05-15:
I still think that the reason may be that the server does not pass a http status code message. I will try to make the server return  correct status aka HTTP/1.1 401 Unauthorized. Right now it only sends HTTP/1.1 401

pingwiniasty, 2018-05-15:
There are any simple option to add this to response?

jarlah, 2018-05-15:
yes, i am working on it. I have found the culprit. Its the modern embedded tomcat version that has skipped this because HTTP/2 doesnt need it anymore. So even if it uses HTTP/1.1 the embedded tomcat ignores custom messages in status codes. So, I have a fix for this. To not use embedded tomcat. Simple :) In regards of running the service, it should have no impact. Actually it means less hacks because the httpOnly fix i did was quite a mess with the embedded tomcat but works out of the box with jetty. Also remember that even if i talk about jetty and tomcat, parameters to tune the service is generalized through spring boot. So the underlying container doesnt mean much.

jarlah, 2018-05-15:
@pingwiniasty Try this artiact:
[tr069-2.0.1-SNAPSHOT-bin.zip](https://github.com/freeacs/freeacs/files/2004138/tr069-2.0.1-SNAPSHOT-bin.zip)
The changes in this package is currently in review. So when its confirmed, I will merge it. I have now tested with curl to check that the status code is correct. I see that there is an expires header on the initial 401 but this means in effect that the device should revoke any JSESSIONID it has lying around.
There is a lot more debug log in tro069.log in this package, so it should more simpler to see whats going on.

pingwiniasty, 2018-05-15:
Nope, still from ACS:
```
	HTTP/1.1 401 
	Set-Cookie: JSESSIONID=35F7F9E6A0835610270386E8CFAECD28; Path=/
	WWW-Authenticate: Basic realm="xaps"
	Content-Type: text/html;charset=utf-8
	Content-Length: 12
	Date: Tue, 15 May 2018 09:11:57 GMT
	
	Unauthorized
```

And any new req. from device :(

pingwiniasty, 2018-05-15:
But other device (GPON ZTE F600) was successful added. Maybe this device has problem with this header :/

jarlah, 2018-05-15:
```
$ curl -X POST -v http://localhost:8085/tr069
*   Trying ::1...
* TCP_NODELAY set
* Connected to localhost (::1) port 8085 (#0)
> POST /tr069 HTTP/1.1
> Host: localhost:8085
> User-Agent: curl/7.59.0
> Accept: */*
> 
< HTTP/1.1 401 Unauthorized
< Date: Tue, 15 May 2018 09:29:35 GMT
< Set-Cookie: JSESSIONID=node0vvj2t0lrf0gmxx3hfaqozgoz0.node0;Path=/;HttpOnly
< Expires: Thu, 01 Jan 1970 00:00:00 GMT
< WWW-Authenticate: Basic realm="xaps"
< Content-Length: 0
< 
* Connection #0 to host localhost left intact
```

jarlah, 2018-05-15:
note that the artifact has same name and version as the previous. So it may be confusing.

pingwiniasty, 2018-05-15:
Yep, you are right. I'm upload old package :D Sorry.

pingwiniasty, 2018-05-15:
Ok, after 401, CPE sends new req. with Auth header and maybe correct Inform :D

https://pastebin.com/v2WGMMbQ

But ACS receives 204 No Content
```
HTTP/1.1 204 No Content
Date: Tue, 15 May 2018 09:44:04 GMT
Set-Cookie: JSESSIONID=node011r92o6ch851fe4w8wl45j8qw12.node0;Path=/
Expires: Thu, 01 Jan 1970 00:00:00 GMT
Connection: close
```

and in tr069.log
```
2018-05-15 11:50:04 [qtp686466458-17] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node01daztz4goob8s58cnnijouf5y14 created: 1526377804692, lastAccess:1526377804692, mxInactiveInterval:1800
2018-05-15 11:50:04 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 0 elements out of 0 in 0 ms.
2018-05-15 11:50:04 [qtp686466458-17] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-05-15 11:50:04 [qtp686466458-17] DEBUG c.g.f.base.http.BasicAuthenticator - Basic verification of CPE starts, located on IP-address 10.90.7.156
2018-05-15 11:50:04 [qtp686466458-17] DEBUG c.g.f.base.http.DigestAuthenticator - Basic verification identifed unit id 000EB4-R3621-000EB4100020 from CPE IP-address 10.90.7.156
2018-05-15 11:50:04 [qtp686466458-17] DEBUG com.github.freeacs.tr069.SessionData - Will load unit data
2018-05-15 11:50:04 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache.Cleanup.run() starts
2018-05-15 11:50:04 [qtp686466458-17] WARN  com.github.freeacs.base.db.DBAccess - The value of db.xaps.maxage was not specified, instead using default value 600000
2018-05-15 11:50:04 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 0 elements out of 0 in 0 ms.
2018-05-15 11:50:04 [qtp686466458-17] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0,74 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, up.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = '000EB4-R3621-000EB4100020' 
2018-05-15 11:50:04 [qtp686466458-17] WARN  com.github.freeacs.tr069.SessionData - Did not find unit in unit-table, nothing exists on this unit
2018-05-15 11:50:04 [qtp686466458-17] DEBUG com.github.freeacs.tr069.SessionData - No unit data found & discovery mode true -> first-connect = true, allow to continue
2018-05-15 11:50:04 [qtp686466458-17] WARN  c.g.f.base.http.DigestAuthenticator - Authentication not verified, but accepted since in Discovery Mode
2018-05-15 11:50:04 [qtp686466458-17] INFO  c.g.f.base.http.BasicAuthenticator - Authentication verified (CPE IP address: 10.90.7.156)
2018-05-15 11:50:04 [qtp686466458-17] DEBUG com.github.freeacs.common.util.Cache - Key 10.90.7.156Basic MDAwRUI0LVIzNjIxLTAwMEVCNDEwMDAyMDo4MTAxMjM5OTc5Nzk= will be removed from Cache (explicitely - not by Cache.Cleanup.run())
2018-05-15 11:50:04 [qtp686466458-17] DEBUG c.g.f.t.methods.HTTPRequestProcessor - Will process method Empty (incoming request/response from CPE)
2018-05-15 11:50:04 [qtp686466458-17] ERROR c.g.freeacs.tr069.methods.EMDecision - EM-Decision is EM since the CPE did not send an INFORM (or sessionId was not sent by client)
2018-05-15 11:50:04 [qtp686466458-17] DEBUG c.github.freeacs.tr069.Provisioning - End of session is reached, will write queued unit parameters if unit (null) is not null
2018-05-15 11:50:04 [qtp686466458-17] DEBUG com.github.freeacs.common.util.Cache - Key 000EB4-R3621-000EB4100020SESSION will be removed from Cache (explicitely - not by Cache.Cleanup.run())
2018-05-15 11:50:04 [qtp686466458-17] DEBUG com.github.freeacs.common.util.Cache - Key node01daztz4goob8s58cnnijouf5y14SESSION will be removed from Cache (explicitely - not by Cache.Cleanup.run())
```

jarlah, 2018-05-15:
Very Nice .  I will take a break now. You investigate further. Ill check in later today:)

pingwiniasty, 2018-05-15:
Ok :)

jarlah, 2018-05-15:
gosh... fun :) i found the problem. Another corner case for the method name parser.

Thanks the payload. The response 204 is correct, but code is incorrect. The 204 is because the code didnt find the method name. Because of incomplete regex :)

jarlah, 2018-05-15:
@pingwiniasty new artifact: https://github.com/freeacs/freeacs/releases/tag/freeacs-2.0.0-RC8

In this artifact i have merged the changes you tested recently which got the unit further into the protocol communication with the server in that it accepted the 401 challenge. Now, the acs server should accept its Inform also.

pingwiniasty, 2018-05-15:
yep, everything works!