freeacs/freeacs #42: Add support for AddObject

jarlah, 2018-05-13:
I might need to digg deep in tr069 buts definitely doable.

pingwiniasty, 2018-12-11:
HI @jarlah, I will try to add this functionality after new year. I have some concept to implement this in code

pingwiniasty, 2018-12-13:
I think little bit about this issue.
Now, all unit parameters inherits by unit type parameters. In database we have relations from unit_param to unit_type_param. Is this good way to implement this in UnitType? In case, when we want to add 10  firewall rules to Unit, we must add 10 new objects to UnitType, which is unintuitive, because we must create 2 additional steps to add it for only one unit (firstly add it to unit type, secondly set parameter in unit)

What do you think about it?

jarlah, 2018-12-13:
I want to flip the questions around a bit. How can we make it simpler to add unit specific params? Maybe we should instead think about how we could simplify unit parameter creation? If its one, two or three steps. Its all about how you make the process. The unittype is the blueprint aka class. The profile is the abstract shared implementation and the units is the actual implementation. So each unit can only be defined with a parameter defined in its unittype defintion.

I will make en ER diagram for the whole database and post it on wiki soon.

closed 2021-09-04 by jarlah