freeacs/freeacs #66: Suggest a better way to monitor services, now that they are all bundled with actuator

jarlah, 2018-05-19:
I am going to remove monitor. I need to suggest a better way  monitor modules. Maybe inter module (cluster?) monitoring? Need to think about this. One thing for sure, I cannot maintain a monitor application. Those that want to use it, should use old artifact.

jarlah, 2018-05-20:
cron job?

Cougar, 2018-07-30:
What about adding [Prometheus client](https://github.com/prometheus/client_java) to modules? Of course this needs to some more investigation what metrics are meaningful and should be collected. Any thoughts?

jarlah, 2018-08-02:
it might be a good idea. But right now I have brought back the old monitor back from the dead, stripped for all unnecessary features. It only checks if modules are up. Closing this issue since.

closed 2018-08-02 by jarlah