freeacs/freeacs #92: Should manual xml creation be replaced with something more readable

jarlah, 2018-05-21:


dror27, 2018-10-07:
Could you please provide a clue as to manually create the xml files? Thanx

jarlah, 2018-10-07:
in code
```
String xml = "<Whatever>" + OK + "</Whatever>";
```
service should just send an instance of Whatever to a SOAP marshaller.

jarlah, 2018-10-07:
but i am not sure if i am ever going to do this

dror27, 2018-10-07:
Sorry - I though this issue related to the creation of the classes in com.github.freeacs.ws.xml ... I have since generated them manually using xjc (as I could not find the automation for it). Very nice project. Feel free to erase this whole thread.

jarlah, 2018-10-07:
Install sbt and run sbt clean compile. Can go without clean also. And use intellij with Scala plugin (provides sbt) to open project

jarlah, 2018-10-07:
Then you will get automatic xsd to webservice classes

jarlah, 2018-10-07:
If you just need the classes you can of course skip intellij. But any contributions is Nice:)

dror27, 2018-10-07:
Thank you for the quick response. Enjoy :)

jarlah, 2018-10-07:
No problem. I usually respond fast only when i immediately know the context and what to reply. Closing.

closed 2018-10-07 by jarlah