freeacs/freeacs #116: next step after FreeACS installation

jcmosquera2001, 2018-05-29:
Hi,

I was able to install FreeACS with all components on Centos 7.5 + MySQL 8 (it is a Linode VM).
So far so good! :)

We have an small installation with 20- ONTs ZTE-F660 that I want to start configuring using TR069. 

Do I have to define any king of Unit Type at GUI ? 
Which is the default user/pass to configure on the devices ?

Regards,
Jose

pingwiniasty, 2018-05-29:
Jarlah is right. Set TR069 in discovery mode
`discovery.mode = true`
 then new units will be available in web panel.

In discovery mode you not need user and pass for devices. Devices pass it at first request

jarlah, 2018-06-10:
There has been made a great deal of improvements to the install process. We now have deb and rpm packages. And docker images.

I suggest download the rpms from the latest release and install them in centos. But disable the old services with systemctl disable freeacs-[module]. And you might need to Port some custom config to /etx/default/freeacs-[module]

jarlah, 2018-06-10:
Aaaaand now I have added install script for CentOS. See wiki.

closed 2021-09-04 by jarlah