freeacs/freeacs #117: Fresh install RC11 on Ubuntu 16.04 LTS and Failed to create Users object i Web interface

jaroslavsafranek, 2018-05-31:
Hi, 

I have problem with FreeACS RC11.
I used wget https://raw.githubusercontent.com/freeacs/freeacs/master/scripts/install.sh -v -O install.sh && chmod +x install.sh && ./install.sh;

And FreeACS is installed without errors, and px ax show FreeACS running.
I can connect to host:8081 but with login admin/freeacs as you wrote in #116 show Failed to create Users object i Web interface. 

You or script wrote about steps from PDF, but it's old version.
Or are here some steps for installation make done?

Best regards Jaroslav

jarlah, 2018-05-31:
Great to hear that the script worked! 

Next step in resolving this issue is to look into the logs. You can find the log in /var/freeacs/web/logs. The message in web ui is very limited. 

It can be many things. One of them being misconfigured database password for the acs user. 

We will just have to see what gives in the logs :)

Cheers.

jarlah, 2018-05-31:
I will btw veeery soon make a guide or document on how to get things up after install. And yes old docs are for tomcat. What you can take from old docs is functionality and property docunentation

jarlah, 2018-06-01:
But hikari has automatic leak detection. But it can't guard against human errors (possible errors while porting old connection pool, which in fact was even more exhausting)

jarlah, 2018-06-05:
@jaroslavsafranek sorry my bad. I noticed while testing that i didnt log the error to log file while logging. Regarding the users object.

jogermech, 2018-06-09:
found the same issue (Ubuntu 16.04, fresh install) and from logs it seems related to using wrong mysql user/password
I've changed it in /var/freeacs/web/application-config,properties to proper ones
Then found that MySQL cannot normally recognize system timezone settings which was "MSK",
changed it by logging into mysql and running command
SET GLOBAL time_zone = '+3:00';
Then logged in with admin/freeacs.

jarlah, 2018-06-09:
there is a similar issuee regarding CEST timezone talked about in the forums. I think this is more deployment issue. Regarding CEST timezone, this was solved by adding parameters to jdbcUrl in application-config.properties.

jogermech, 2018-06-09:
It seems timezone settings doesn't survive system restart. I'm quite newbie in both Ubuntu and mysql, could you advise how to store settings, so it survived restart?

jarlah, 2018-06-09:
But if you have wrong username and password, how is this possible? You should always choose to select acs as the password for the acs user, otherwise it wont work fresh install. It think the script warns about this?

jarlah, 2018-06-09:
I can make it so that the script only asks for root password, and that it uses acs/acs behind the scenes. And then you can change and configure other password after install.

jogermech, 2018-06-09:
I really a bit messed in multiple passwords requested by script, and I tried several times install setting all passwords ="acs" and all passwords ="freeacs". And I only check logs at last install when used "freeacs". Maybe first time it was ok, but failed because of timezone

jarlah, 2018-06-09:
haha... i can see the confusion. The solution is really to not ask for acs password, and manage this in the script. I will think about it. See https://www.tapatalk.com/groups/freeacs/i-can-install-freeacs-for-you-t300-s50.html#p1423 for info about fix for CEST

jarlah, 2018-06-10:
No problems @jogermech. Someone needs to drive this bus :) After a while it will drive it self ;)

Btw, I have fixed a the UTC/CEST timezone thing .. AND the useSSL in latest release.

In addition i have added nginx to the script, so that you can call http://localhost/[module] after installation.

jarlah, 2018-09-05:
I consider this outdated and its propably also fixed. Please verify. Closing. Open new issue if still present

closed 2018-09-05 by jarlah