freeacs/freeacs #132: Cannot handle device parameters in WEB (2.0.0R12)

jogermech, 2018-06-13:
When try to modify existing parameter of a CPE or a profile via the Web GUI and press "Update parameters" button, I am instantly taken to a "Whitelabel Error Page", with the following mesage:

"This application has no explicit mapping for /error, so you are seeing this as a fallback.
There was an unexpected error (type=Bad Request, status=400).
Unable to parse form content."

Web log attached
[web.log](https://github.com/freeacs/freeacs/files/2097430/web.log)

jarlah, 2018-06-13:
Ok i see the problem. I will plan a fix.

jarlah, 2018-06-13:
Web has undergone massive refactoring, But this one is about the switch of underlying servlet container. And default config for max allowed payload sent to the server.

jogermech, 2018-06-13:
Ok, it's good that you know the root cause

jarlah, 2018-06-13:
Can you try to update ```/etc/default/freeacs-web``` with:

```SPRING_SERVLET_MULTIPART_MAX-REQUEST-SIZE=20MB```

You can try to increase it to 100MB (just to make sure) if it doesnt work. If 20 and 100 doesnt work. Then I will have to try something else.

If it works however I will hardcode it in.

jarlah, 2018-06-13:
Well not hardcode, but in properties file. that resides in jar file

jarlah, 2018-06-13:
And systemctl restart freeacs-web

jogermech, 2018-06-14:
Hi, Jarl.I tried your suggestion, but it didn't help
Could you help to check whether my original suggestion is correct (regarding "Form too large" log)?
I found this log it is related to org.eclipse.jetty.server.Request.maxFormContentSize property, it is 200000 by default, but I'm not sure whether it is possible to workaround by changing some environment variable. I even found a way to change it via JVM_OPTS variable, and tried to add it in freeacs-web defaults (not sure it is correct to place it there though) but in log I still see that form is bigger than 200000, so it didn't help.

jogermech, 2018-06-14:
Can't say exactly, number of parameter names is about 1000, and if consider instances of some nodes then I believe it will be more than 2K

jarlah, 2018-06-14:
I think we used wrong and maybe deprecated property. I will test another and create å bunch of parameters

jarlah, 2018-06-15:
Nailed it. I found the property, set it to 1, tried to do anything like login and falled hard. Then i increased the value much higher than the default value (think it's 20mb or something or 2mb), and then i pushed it to master. Will generate a new minor release and update artifacts on RC12 and also publish docker images. This goes for all services. Also tr069. The default limit is too low for how the stack is implemented. 

It might be that i wait with release until i have found out what gives with the connection leakage issue.

jarlah, 2018-06-15:
Try again, rerunning the install script should work. You might get a warning that you need to update nginx config, if using install.sh, but you ignore this. It just finds out that nginx is installed and its not doing anything with nginx.conf. You could also just download the web .deb file and install it directly. There is actually no need to rerun the installation script to fix the web problem. After downloaded install it with dpkg -i freeacs-web_2.0.0.rc.12.3_all.deb

Web, tr069 and webservice has gotten the new property set. Syslog, stun and core doesnt depend on multipart post body or rest for that matter.

Before the default was 200000 now its configured to be 5000000

jogermech, 2018-06-15:
Hi, it still fails, but log is different now:
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.IllegalStateException: Form with too many keys [1001 > 1000]

jogermech, 2018-06-15:
Here is log, if you need it
[web.log](https://github.com/freeacs/freeacs/files/2105866/web.log)

jarlah, 2018-06-16:
I found a way to fix it. But Ill hard code it with no way to override. I set the max form keys limit to 2000. We need to place some hard limit, or else we might get ddos attacks.

Currently preparing release

jarlah, 2018-06-16:
Fixed in deb and rpm packages. Havent had time yet to push to docker images.

Please verify if not using docker

jarlah, 2018-06-16:
I will increase it. But why do you have so many params?

jogermech, 2018-06-16:
I still didn't get the point regarding parameter values displaying in ACS. 
From my point in real application it will never need to display them all,  and suppose there should be a way to create some limited set of params to get. But currently I cannot do any action within unit, unit type or profile pages, as it fails.
And regarding number of parameters on device - it is a normal ethernet WiFi router (HGW). I think other devices I have on hands should have even greater number of params, as they additionally have SIP and dual-band WiFi. So, multiple WAN, multiple WLAN (2 bands, 4 SSIDs each) and SIP nodes according to TR098 data model are giving so many parameters.
These devices are managed by Axiros and Friendly solutions in a billions subscriber real networks with no problems. And I'm sorry for putting such a long story here, it is more suitable for a forum rather than for issue reporting )

jarlah, 2018-06-16:
I think tomcat doesnt care. At least old versions. But I will completely disable the limit by setting to a number so high that if the limit is reached will make web so unusable and slow for changing params that you could use Shell instead. Lets say 10K. If that limit is reached. You can't possibly change parameters in web without waiting minutes for client side js to complete render

jarlah, 2018-06-16:
If you have installed freeacs via Shell script. Shell should be available via freeacs-shell command. I think it's better solution to change unit parameter for such large sets of params. 

But i will up the limit as said so much that it will be effectively disabled. But as i said web is not optimized for rendering and updating one of 5000 params. The problem is client sends all params back

jarlah, 2018-06-16:
Old software. I have thought about adding some async small updates for params.  But its not prio

jarlah, 2018-06-16:
@jogermech try again. Updated web deb and rpm and pushed to docker hub. I have now (with the help of freeacs shell) created a unit type with 5000 parameters and tried to add a unit to this unit type and then update one of the parameters. IT worked, but it was so horribly slow, that I feel bad for anyone trying to load and display so many parameters. But you should at least be able to update the parameters, actually al the way up 100K params. Which is totally bonkers. However. This MUST be revisited and we need to see if it is possible to only submit those parameters that have actually changed.

jarlah, 2018-06-17:
@jogermech  💯 Can we close issue?

jogermech, 2018-06-17:
yes, close it

closed 2018-06-17 by jarlah