freeacs/freeacs #133: TR069 always 401 authentication error,don't know setting username and password?

zhouxiaohai, 2018-06-13:
hi:
TR069 always fails to connect. The catch sees the 401 authentication error.
Installation has never seen the account password settings. Now how do I get my phone?

[image]

jarlah, 2018-06-13:
I will check.

jarlah, 2018-06-13:
I checked to see if I had accidentally added spring security to tr069 server, which would have been a total blunder. But I hadnt. I think this is a question more in line with "How do I provision my unit for the first time?". So, before I continue with more advice, is this assumption correct?

But hey! great to see that you have deployed the release and that it works :+1: (except for this problem ..:P )

zhouxiaohai, 2018-06-14:
I've changed application.properties to "auth.method=none" and now I'm reporting a redirect error! It doesn't seem to work.

[image]

zhouxiaohai, 2018-06-14:
I used both "http:server:8085/tr069" and "http:server/tr069" to post connection debugging and both returned redirection errors. How do I use it? Seek guidance

jarlah, 2018-06-14:
Most propably because you need a slash at the end. I would try that e.g. ```http://server/tr069/```

You could also check curl at command line (if you have linux or linux subsystem or cygwin) with ```curl -v http://server/tr069``` and ```curl -v http://server/tr069/``` to verify my point. 

You can call either ```http://server/tr069/``` or ```http://server/tr069/ok``` in browser to check if its running. But its obviously running. Just a tip if you are unsure if its running.

I wouldn't set auth mode to none. I would let auth mode be what it was (most propably basic) and set discovery.mode=true. But Are you modifying inside the jar file, or running pre RC12? In RC12 you should place override configs in ```/etc/default/freeacs-tr069``` like adding ```DISCOVERY_MODE=true```. In RC11 and earlier you must update ```/var/freeacs/tr069/application-config.properties``` with adding or changing ```discovery.mode=true```.

jarlah, 2018-06-14:
add a slash to the end of the url in the cpe device configuration. Actually the part /tr069 is just a context path. Not an actual servlet. You could add anything behind it, like /tr069/prov. See curl output below to see my point:

```
➜  scripts git:(master) curl -v http://localhost/tr069
*   Trying ::1...
* TCP_NODELAY set
* Connected to localhost (::1) port 80 (#0)
> GET /tr069 HTTP/1.1
> Host: localhost
> User-Agent: curl/7.52.1
> Accept: */*
> 
< HTTP/1.1 301 Moved Permanently
< Server: nginx/1.15.0
< Date: Thu, 14 Jun 2018 06:33:12 GMT
< Content-Type: text/html
< Content-Length: 185
< Location: http://localhost/tr069/
< Connection: keep-alive
< X-Clacks-Overhead: GNU Terry Pratchett
< 
<html>
<head><title>301 Moved Permanently</title></head>
<body bgcolor="white">
<center><h1>301 Moved Permanently</h1></center>
<hr><center>nginx/1.15.0</center>
</body>
</html>
* Curl_http_done: called premature == 0
* Connection #0 to host localhost left intact
➜  scripts git:(master) curl -v http://localhost/tr069/
*   Trying ::1...
* TCP_NODELAY set
* Connected to localhost (::1) port 80 (#0)
> GET /tr069/ HTTP/1.1
> Host: localhost
> User-Agent: curl/7.52.1
> Accept: */*
> 
< HTTP/1.1 200 OK
< Server: nginx/1.15.0
< Date: Thu, 14 Jun 2018 06:33:20 GMT
< Content-Length: 194
< Connection: keep-alive
< X-Clacks-Overhead: GNU Terry Pratchett
< 
<title>xAPS TR-069 Server Monitoring Page</title><h1>Monitoring of the TR-069 Server v. 3.1.2</h1><h1>Memory Usage</h1>
* Curl_http_done: called premature == 0
* Connection #0 to host localhost left intact
The JVM uses 217 MB of memory. 217 MB of memory available on the heap.<br>%                                                                                                                                        ➜  scripts git:(master) curl -v http://localhost/tr069/prov     
*   Trying ::1...
* TCP_NODELAY set
* Connected to localhost (::1) port 80 (#0)
> GET /tr069/prov HTTP/1.1
> Host: localhost
> User-Agent: curl/7.52.1
> Accept: */*
> 
< HTTP/1.1 200 OK
< Server: nginx/1.15.0
< Date: Thu, 14 Jun 2018 06:33:32 GMT
< Content-Length: 194
< Connection: keep-alive
< X-Clacks-Overhead: GNU Terry Pratchett
< 
<title>xAPS TR-069 Server Monitoring Page</title><h1>Monitoring of the TR-069 Server v. 3.1.2</h1><h1>Memory Usage</h1>
* Curl_http_done: called premature == 0
* Connection #0 to host localhost left intact
The JVM uses 230 MB of memory. 204 MB of memory available on the heap.<br>%                                                                                                                                        ➜  scripts git:(master) curl -v -X POST http://localhost/tr069/prov
*   Trying ::1...
* TCP_NODELAY set
* Connected to localhost (::1) port 80 (#0)
> POST /tr069/prov HTTP/1.1
> Host: localhost
> User-Agent: curl/7.52.1
> Accept: */*
> 
< HTTP/1.1 401 Unauthorized
< Server: nginx/1.15.0
< Date: Thu, 14 Jun 2018 06:33:39 GMT
< Transfer-Encoding: chunked
< Connection: keep-alive
< Set-Cookie: JSESSIONID=node01jkv3c06xgpar1h198c9cfaj00.node0;Path=/tr069;HttpOnly
< Expires: Thu, 01 Jan 1970 00:00:00 GMT
< WWW-Authenticate: Digest realm="xaps", qop="auth", nonce="4e28960836949f38ad5af5f79890e54f", opaque="2bfcc6bf9ad3df2434098091482fcb8f"
< 
* Curl_http_done: called premature == 0
* Connection #0 to host localhost left intact
➜  scripts git:(master) 
```

It hits me however that it returns Digest by default. I will look into it, but now you see how it works.

closed 2018-06-14 by zhouxiaohai

jarlah, 2018-06-14:
I havent said the test is success or that your problem is fixed. Currently freeacs is rewritten to be maintainable and future proof. And is currently a release candidate.  We lack up to date documentation for the New plattform but you can still the old documentation under each repo. This old documents will cover everything except new way of installing and maintaining install. Please do go read what you can find in the repo and maybe you will understand more