freeacs/freeacs #134: Error time out [Core] to DB connection Docker and normal installation

penaivanalejandro, 2018-06-14:
Good day, after installing FreeACS and start creating Unit Type, Profile and unit.
In the logs it begins to mark the following error, both in Dockers and traditional installation:

[DBI for Core] ERROR com.github.freeacs.dbi.DBI - An error occurred in DBI.run () (stacktrace printed earlier): mainPool - Connection is not available, request timed out after 30000ms.

And the creation of more units no longer works or the CPEs can register.

Marking a series of errors, this log is from Docker but this happen to in normal installation...

[log_freeacs.log](https://github.com/freeacs/freeacs/files/2103554/log_freeacs.log)

jarlah, 2018-06-14:
run docker logs freeacs_core_1 > core.log
the combined log is very much unreadable ;)
I have two scenarios to investigate 1) connection leakage and 2) misconfigured connection pool in core.
I need some steps. Let me guess your steps:
1) You createu unit type, profile and unit in gui
2) You point some units to the tr069 server
3) Errors occurs? 

Is this correct? Or is there another set of steps you are doing to reproduce?

jarlah, 2018-06-14:
I have had some doubt about the connection pooling in the code, sometimes the code takes a connection and maybe not closing it properly when finished. If I can create a routine that I can use to reproduce then maybe i can find the culprit

penaivanalejandro, 2018-06-14:
Exactly that happens ...

When I start creating what you mention in the same order the first CPE is registered in ACS, when I try to create the second unit already in the logs it shows the connection error time out and it no longer allows to create anything.

In docker I only reboot the container and in the normal installation I need to restart the whole system.

I will leave the log running and having the error again (because I already reincie the container and has not been presented) I upload it.

penaivanalejandro, 2018-06-14:
This is the result from the command docker logs freeacs_core_1 > core.log
[core.log](https://github.com/freeacs/freeacs/files/2103658/core.log)

penaivanalejandro, 2018-06-14:
Again the same error...  this is the log where show the time out

[core2.log](https://github.com/freeacs/freeacs/files/2103840/core2.log)
[stun.log](https://github.com/freeacs/freeacs/files/2103841/stun.log)
[tr069.log](https://github.com/freeacs/freeacs/files/2103842/tr069.log)

jarlah, 2018-06-15:
Ive fixed a max http body size problem with jetty in RC12. Just docker pull freeacs-web and docker pull freeacs-tr069 and then docker-compose -p freeacs --force-recreate up -d ... not completely sure about the order of the things in the last command, since i am not able to run it myself atm.

jarlah, 2018-06-15:
@penaivanalejandro question:

The stack is built around separating syslog into its own datasource. But i am not really convinced that there is any point in this anylonger. I have gone away from the old custom made connection pooling to hikari. And if we configure hikari correctly I am definitely sure its possible to run with syslog tables in same database as configuration and unit data. 

With this is am proposing to no longer having two separate hikaricp datasources, but to only have one datasource per service. This will consume less connections and should be more future proof. If there is slowness in regards to syslog server accessing database, then that is a completely different issue.

What do you think?

penaivanalejandro, 2018-06-15:
@jarlah 

Good day Jarl, update the container you mention and I confirm you.
And I totally agree to unify the connection that is made with Hikari to the database.
It would be healthier when making connections transactions to the database.

penaivanalejandro, 2018-06-15:
I try the commands both commands:

- docker pull freeacs-web
- docker pull freeacs-tr069

and response:

Using default tag: latest
Error response from daemon: pull access denied for freeacs-web, repository does not exist or may require 'docker login'

Using default tag: latest
Error response from daemon: pull access denied for freeacs-tr069, repository does not exist or may require 'docker

i don't know what i doing wrong.

jarlah, 2018-06-15:
Ah sorry. My bad. I forgot freeacs/ prefix. Hehe. Docker pull freeacs/freeacs-etc

penaivanalejandro, 2018-06-15:
Thanks...   thats work

root@freeacs:/home/adminfree# docker pull freeacs/freeacs-tr069
Using default tag: latest
latest: Pulling from freeacs/freeacs-tr069
cc1a78bfd46b: Already exists 
d2c05365ee2a: Already exists 
231cb0e216d3: Already exists 
3d2aa70286b8: Already exists 
b621e52d9d65: Already exists 
f12314f6054a: Already exists 
f5ce6397976f: Already exists 
965859e0769b: Already exists 
4a76f2995e39: Already exists 
8459a3d1edf8: Already exists 
0e357338455b: Pull complete 
9a0615b25cea: Pull complete 
e861571c7b6a: Pull complete 
Digest: sha256:bb98d59826aa7ed7708c7144ad7e64ed8a584628bca7296e58884c64e7907b04
Status: Downloaded newer image for freeacs/freeacs-tr069:latest
root@freeacs:/home/adminfree#

jarlah, 2018-06-15:
bam 100% reproducable. currently testing on windows .., had to fix my linux laptop and got a windows machine in backup from work .. and hey ... can run freeacs in docker. . and i reproduced the problem veeery quickly.... i will look into it. About the core and connection timeout

jarlah, 2018-06-15:
The title on the project IS true... the ACS that can run almost everywhere ;)

[image]

jarlah, 2018-06-15:
I knew it. Sadly enough I knew it. When i refactored the code I thought the connection handling in the DBI object was poorly implemented. But I thought to myself that hey i bet its a good reason. But no, it was not a good reason. It might also be that i managed to destroy a working business logic that cleaned up its resources correctly.  It might be. But now I will fix the connection leakage problems for all modules. Because DBI is used by all modules.

jarlah, 2018-06-15:
DBI is an object/system that preloads data and listens to messages from other modules for when to reload. So when you change something in web core and tr060 and all other modules will reload.

jarlah, 2018-06-15:
solved it. Shitty connection management solved by asking for new connection and closing it when it done when needed. Problem was that the old connection pooling had a special technique for checking connection idling and then autoreturned it to the pool. But hikari needs explicit connection returning. 

Need to find a linux computer to build the debs and rpms. And to push docker images. Cant trust windows. Its a hell to work in. The terminal is like a torture. Even powershell makes me cringe.

jarlah, 2018-06-15:
fixed. release updated. docker images updated. please verify

penaivanalejandro, 2018-06-15:
@jarlah  is a pleasure and honor to inform .....

That the issue is no longer presented in this release.
Tested on Docker ... !!! Again good work. 
Issue resolved .. !!!

penaivanalejandro, 2018-06-15:
Issue Resolved by @jarlah in Docker...!

closed 2018-06-15 by penaivanalejandro