freeacs/freeacs #135: Freeacs-web Provisioning history question?

zhouxiaohai, 2018-06-15:
1. Changing unit values cannot be Provisioning history effective?

[image]

2. Provistion ,Read all, Reboot, Reset..button Click anomaly!
```
path:   http://192.168.78.159/web/index?page=unit-configuration

page reason:

Whitelabel Error Page
This application has no explicit mapping for /error, so you are seeing this as a fallback.

Fri Jun 15 11:03:24 CST 2018
There was an unexpected error (type=Bad Request, status=400).
Unable to parse form content

```

So, I can't get the parameters of phone from freeacs-web!

zhouxiaohai, 2018-06-15:
I found 1 #132  problem was the same as mine.

jarlah, 2018-06-15:
Will fix soon. Will release fix today. Hopefully:)

jarlah, 2018-06-15:
See my last comment in #132

jarlah, 2018-06-16:
Fixed max http form size problem and max form keys problem in deb and rpm packages. Havent had time yet to push to docker images.

Please verify if not using docker

jarlah, 2018-06-16:
Please see https://github.com/freeacs/freeacs/issues/132#issuecomment-397842303

jarlah, 2018-06-17:
Please verify that this is now fixed in latest release binary of web module in RC12.

jarlah, 2018-06-20:
@zhouxiaohai can you verify the fix

jarlah, 2018-09-05:
I consider this outdated. If this is still an issue, please open new issue.

closed 2018-09-05 by jarlah