freeacs/freeacs #136: Prepare 2.0.0 release

jarlah, 2018-06-16:
There has been done major contributions to the codebase both by you the users and me as a developer. 

I feel we have come to a point where i need to make a list of things that needs to be done/fixed before we can officially go live with a community supported version.

1. Documentation needs to be updated especially installation and Configuration and first login

2. Datasource defintions is now only one datasource and i need to remove config and update code to represent this. 

3. Automatic docker image publishing when travis runs 

With these in place i will be confident enough to make a major release.

And if we find more bugs before release we just fix them

jarlah, 2018-06-17:
Automatic docker image publishing when travis runs
**Done**

jarlah, 2018-06-18:
Release is dependending on #135 to be resolved.

jogermech, 2018-08-07:
Hi, Jarl. I'm not sure how to report correctly the issue I met on 2.0.0, so I will post here for a while. Maybe you will help to sort it out:
TR069 conversations log is empty, as well as tr069 events log. I installed new release by dpkg -i over old RC..19 with config files replacement (answer Y when asked). Then I added quirks for my test device, but failed to get it working. Then I noticed there is no conversation in log.
After that I decided to delete unit and create it again. I added new unit of existing type and profile, but it didn't help

jarlah, 2018-08-07:
Interesting. Might be logback. I will check. There should be a logback file in config. I think i set root level on tr069 to logback (same as for all other modules) but conversation and event and stuff should override. I will check

jarlah, 2018-09-05:
2.0.0 release was success.

closed 2018-09-05 by jarlah