freeacs/freeacs #139: No device parameter values found, ACS parameterkey = "No data in DB"

penaivanalejandro, 2018-06-19:
Hello, again requesting your support with a particular CPE ... a Huawei B310 where it shows the following message in the log of tr069.log

[tr069-conversation.log](https://github.com/freeacs/freeacs/files/2116157/tr069-conversation.log)

```XML
2018-06-19 11:30:18 [qtp399699227-29] INFO  Conversation - ============== FROM CPE ( B123456 ) TO ACS ===============
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/" xmlns:SOAP-ENC="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
	<SOAP-ENV:Header>
		<cwmp:ID SOAP-ENV:mustUnderstand="1">37</cwmp:ID>
	</SOAP-ENV:Header>
	<SOAP-ENV:Body>
		<cwmp:Inform>
			<DeviceId>
				<Manufacturer>Huawei Technologies Co., Ltd.</Manufacturer>
				<OUI>00E0FC</OUI>
				<ProductClass>B310s-518</ProductClass>
				<SerialNumber>B9K7S17713003443</SerialNumber>
			</DeviceId>
			<Event SOAP-ENC:arrayType="cwmp:EventStruct[1]">
				<EventStruct>
					<EventCode>0 BOOTSTRAP</EventCode>
					<CommandKey/>
				</EventStruct>
			</Event>
			<MaxEnvelopes>1</MaxEnvelopes>
			<CurrentTime>2018-06-20T00:30:18</CurrentTime>
			<RetryCount>0</RetryCount>
			<ParameterList SOAP-ENC:arrayType="cwmp:ParameterValueStruct[7]">
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.DeviceInfo.SpecVersion</Name>
					<Value xsi:type="xsd:string">1.0</Value>
				</ParameterValueStruct>
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.DeviceInfo.HardwareVersion</Name>
					<Value xsi:type="xsd:string">WL1B310FM</Value>
				</ParameterValueStruct>
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.DeviceInfo.SoftwareVersion</Name>
					<Value xsi:type="xsd:string">21.329.01.00.1494</Value>
				</ParameterValueStruct>
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.DeviceInfo.ProvisioningCode</Name>
					<Value xsi:type="xsd:string"/>
				</ParameterValueStruct>
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.ManagementServer.ParameterKey</Name>
					<Value xsi:type="xsd:string"/>
				</ParameterValueStruct>
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.ManagementServer.ConnectionRequestURL</Name>
					<Value xsi:type="xsd:string">http://100.111.253.120:8081/xusfdeanzgejnicyfgqxeaohrzlom</Value>
				</ParameterValueStruct>
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.WANDevice.1.WANConnectionDevice.1.WANIPConnection.3.ExternalIPAddress</Name>
					<Value xsi:type="xsd:string">100.111.253.120</Value>
				</ParameterValueStruct>
			</ParameterList>
		</cwmp:Inform>
	</SOAP-ENV:Body>
</SOAP-ENV:Envelope>
```
```XML
<`2018-06-19 11:30:18 [qtp399699227-29] INFO  Conversation - =============== FROM ACS TO ( B123456 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
	<cwmp:ID soapenv:mustUnderstand="1">37</cwmp:ID>
</soapenv:Header>
<soapenv:Body>
		<cwmp:InformResponse>
			<MaxEnvelopes>1</MaxEnvelopes>
[... 39 lines cut ...]
2018-06-19 11:30:18 [qtp399699227-29] DEBUG c.g.f.t.methods.HTTPRequestProcessor - Will process method Inform (incoming request/response from CPE)
2018-06-19 11:30:18 [qtp399699227-29] DEBUG c.github.freeacs.tr069.methods.INreq - Parsed INreq params, found keyroot:InternetGatewayDevice., parameterkey:, swver:21.329.01.00.1494
2018-06-19 11:30:18 [qtp399699227-29] DEBUG c.github.freeacs.tr069.ParameterKey - No device parameter values found, ACS parameterkey = "No data in DB"
2018-06-19 11:30:18 [qtp399699227-29] DEBUG o.s.b.w.s.f.OrderedRequestContextFilter - Cleared thread-bound request context: Request(POST //localhost:8085/tr069/)@274f219
```

On the Huawei page, just bring these parameters to configure:

[image]

I attach the log of the different modules, and I probe it in Dockers and Normal installation (via script "install.sh") and I get the same result.

I need some help to register this CPE and can register.

The way I do it in FreeACS is using:
- Unit as user
- System.X.Secret as a password

and add the option of:
- System.X.Discover = true
- System.X.Provisioning = READALL / REGULAR (both and it is the same error)

Others Logs:
[stun.log](https://github.com/freeacs/freeacs/files/2116190/stun.log)
[syslog.log](https://github.com/freeacs/freeacs/files/2116191/syslog.log)

penaivanalejandro, 2018-06-19:
I put only the parameter:

System.X.Discover = true
System.X.Provisioning = REGULAR

and the log from stun show me this:

```LOG
2018-06-19 12:30:14 [ActiveDeviceDetection] INFO  c.g.f.stun.ActiveDeviceDetection - ActiveDeviceDetection: Have removed 0 devices from sentSyslog map - should be approx 1/60 of activeDevices.size() (0)
2018-06-19 12:30:14 [ActiveDeviceDetection] INFO  c.g.f.stun.ActiveDeviceDetection - ActiveDeviceDetection: Processed 0 active device syslog messages to the syslog server. Sent 0 syslog messages. SentSyslogMap.size() = 0, ActiveDevices.size() = 0
2018-06-19 12:30:18 [KickRunnable] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - -SQL: SELECT u1.unit_id, u1.profile_id, u1.unit_type_id FROM unit u1 WHERE (u1.unit_id IN (SELECT up.unit_id FROM unit_param up WHERE value = 'READALL') OR u1.unit_id = 'READALL') 
2018-06-19 12:30:44 [mainPool housekeeper] DEBUG com.zaxxer.hikari.pool.HikariPool - mainPool - Before cleanup stats (total=2, active=0, idle=2, waiting=0)
2018-06-19 12:30:44 [mainPool housekeeper] DEBUG com.zaxxer.hikari.pool.HikariPool - mainPool - After cleanup  stats (total=2, active=0, idle=2, waiting=0)
2018-06-19 12:30:47 [DBI for STUN] DEBUG com.github.freeacs.dbi.Inbox - Message: [MTYPE: PUBLISH-IM, OTYPE: UNIT, FROM: Web, TO: STUN, OBJECTID: B123456] was added to an inbox
2018-06-19 12:30:48 [KickRunnable] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0.76 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, up.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = 'B123456' 
2018-06-19 12:30:48 [KickRunnable] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0.94 ms] SELECT * FROM unit_param_session WHERE unit_id = 'B123456'
2018-06-19 12:30:48 [KickRunnable] DEBUG KickSingle - B123456 is added to the watch list (from messages), will now be processed.
2018-06-19 12:30:48 [KickRunnable] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0.56 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, up.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = 'B123456' 
2018-06-19 12:30:48 [KickRunnable] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0.43 ms] SELECT * FROM unit_param_session WHERE unit_id = 'B123456'
2018-06-19 12:30:48 [KickRunnable] INFO  com.github.freeacs.dbi.ACSUnit - Updated unit parameter System.X_FREEACS-COM.ProvisioningMode with value REGULAR
2018-06-19 12:30:48 [KickRunnable] ERROR KickSingle - An error occurred in processUnitWatchList() with unit B123456, continues with next unit (if any)
java.lang.RuntimeException: No keyroot found for unit B123456, probably because no parameters are defined
	at com.github.freeacs.stun.Kick.getKeyroot(Kick.java:173)
	at com.github.freeacs.stun.Kick.kick(Kick.java:64)
	at com.github.freeacs.stun.SingleKickThread.processUnitWatchList(SingleKickThread.java:182)
	at com.github.freeacs.stun.SingleKickThread.run(SingleKickThread.java:223)
	at java.lang.Thread.run(Thread.java:748)

```

penaivanalejandro, 2018-06-22:
Hello @jarlah, I think I need a bit of your help regarding the variables that I can use in the modules, in this case in the / etc / default / freeacs-tr069 module
When I put the variable of:

```
DISCOVERY_MODE = true
```

I get the automatic creation of UnitType, Profile and Unit like this:
[image]

But when I try to add the variables of quirks I do not see that they are respected I am placing them in the following ways:

```
# Option 1:
QUIRKS_B310s-518 = parameterkey, termination, nextlevel0ingpn

# Option 2:
QUIRKS_B310s-518 = parameterkey
QUIRKS_B310s-518 = ignorevendorconfigfile
```

I try with the Option 1 and later with the option 2...  
You can support me now with the change you describe at in your [page](https://github.com/freeacs/freeacs/releases/tag/v2.0.0-rc.12)

What does it say:

**_But you can actually override any property in the application by converting lets say some.nested.property = value to SOME_NESTED_PROPERTY = value. Yes, screaming caps lock. You could do it lowercase, but its best signaling that this is variable environment, and the convention for this is upper case ._**

This is because the only parameter that is registered on the page is the password that is set:

[image]

jogermech, 2018-06-22:
@penaivanalejandro You can check [#134](https://github.com/freeacs/freeacs/issues/137), although it doesn't help in my case

jarlah, 2018-06-22:
I have thought about this. This release where i move override of props to env variables in etc default is not pretty. I could package the applications so that they will read an application-config.properties file in the working directory like it did before. Reason why it came to how it is now is actually that i found a limit in systemd. The etc default file is sourced as env variables only. In init.d we dont have to do this upper case thing. But thats another story. I propose to do it like we did before. Edit a profile dependent Configuration file. E.g. application-config.properties file and then restart. Then you can copy paste example properties from docs 

However this will not solve quirks problem. The issue with The Old code is that the properties file were invalid. They were not actual consistent properties files. So i needed to convert it to another format.

What we could do is that we can reference unit type by id. So that you get something like 

quirks.2=somequirk

Then this quirk will be hardwired to that concrete unittype . However . It would be more natural to store this in a table. And allow to assign quirks in Shell and or web. I guess we might need to have support for many more quirks as we go along . And defining and assigning them is more user friendly. And it's not actual difficult.

What do you think?
Btw RC is more like beta. Late beta... before we fix the current issues we are far away from RC

penaivanalejandro, 2018-06-25:
An apology for the delay in answering.

Thank you very much @jogermech  for linking your issue (#137 ) and this one.

I would like to explore the path suggested by @jarlah  in trying to work with the code (I do not promise much).

And I understand that there are many changes as they mention that it has to be done and we have to wait.

jarlah, 2018-07-05:
I will work on this quirks problem and changing packaging to include and reference application-config.properties next up. We solved connection leakage problem so now i just need to make things simpler to configure and simpler to understand. 

I am hoping to nail most problems in july. And maybe release in the beginning of august. 

If someone has an old working tr069 unit laying around do please send it to me, if possible. You can send it as gift and hopefully it will pass through customs

jarlah, 2018-07-06:
@penaivanalejandro and @jogermech:

See https://github.com/freeacs/freeacs/pull/150/files#diff-7706aecb1b06a2812e70a0b33ca0eccfR126

Does this looks like a working solution? Space is escaped with backslash.

jogermech, 2018-07-06:
Can not test in nearest time. But as I remember problem was also reproduced for a single-word model name

penaivanalejandro, 2018-07-07:
Hi @jarlah, ask if it was possible to send you equipment, but I was not allowed. 

But what I can do is put the devices with the URL, username and password that you provide us so that they try to register and you make the necessary changes. 

Greetings.

jarlah, 2018-07-07:
@penaivanalejandro i have added an integrationtest for the quirks functionality. So now i know it works. At least for a simple unit type name. Havent tested unit type name with space or quirks on special version.

penaivanalejandro, 2018-07-11:
Hello @jarlah ,

I have been testing the quirks these days and the result is the same with the same model that I post in this issue ...!

Use all the quirks one by one.
By doing this I restarted the tr069 service.

I already desperately put all the quirks in the following way:
[image]

Then restart the service again with the command:

  systemctl restart freeacs-tr069

Next step I restart the device and wait to see the log, but it shows the same message in the log of tr069.log:

```LOG
2018-07-11 17:31:12 [qtp99695585-25] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node04ho8ypuvndud1ifo5q4g3m27o2 did not return a SessionData object from cache, must create a new SessionData object
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache.Cleanup.run() starts
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Key node0e9gx85ja7csegkymu23yzdiq1SESSION with CacheValue com.github.freeacs.common.util.CacheValue@32eb7b99 is to be removed
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - CacheValue has a cleanupnotifier-class which will be executed
2018-07-11 17:31:12 [Cache Cleanup] ERROR c.g.f.base.SessionDataCacheCleanup - SessionData for b310v1 (key:node0e9gx85ja7csegkymu23yzdiq1) was removed from cache after timeout - indicate a session which did not terminate correctly
2018-07-11 17:31:12 [qtp99695585-25] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node04ho8ypuvndud1ifo5q4g3m27o2 created: 1531348272101, lastAccess:1531348272101, mxInactiveInterval:1800
2018-07-11 17:31:12 [qtp99695585-25] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-07-11 17:31:12 [qtp99695585-25] INFO  c.g.f.base.http.BasicAuthenticator - Send challenge to CPE, located on IP-address 0:0:0:0:0:0:0:1
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Key node0alvfg4rcjirbwlci4wq825oi0SESSION with CacheValue com.github.freeacs.common.util.CacheValue@19709f5f is to be removed
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - CacheValue has a cleanupnotifier-class which will be executed
2018-07-11 17:31:12 [Cache Cleanup] ERROR c.g.f.base.SessionDataCacheCleanup - SessionData for null (key:node0alvfg4rcjirbwlci4wq825oi0) was removed from cache after timeout - indicate a session which did not terminate correctly
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Key b310v1SESSION with CacheValue com.github.freeacs.common.util.CacheValue@e6436b is to be removed
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - CacheValue has a cleanupnotifier-class which will be executed
2018-07-11 17:31:12 [Cache Cleanup] ERROR c.g.f.base.SessionDataCacheCleanup - SessionData for b310v1 (key:b310v1) was removed from cache after timeout - indicate a session which did not terminate correctly
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 3 elements out of 1 in 7 ms.
2018-07-11 17:31:12 [qtp99695585-28] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node0dqu83g2zyd8v14d4d4qo0zhj93 did not return a SessionData object from cache, must create a new SessionData object
2018-07-11 17:31:12 [qtp99695585-28] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node0dqu83g2zyd8v14d4d4qo0zhj93 created: 1531348272208, lastAccess:1531348272208, mxInactiveInterval:1800
2018-07-11 17:31:12 [qtp99695585-28] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-07-11 17:31:12 [qtp99695585-28] DEBUG c.g.f.base.http.BasicAuthenticator - Basic verification of CPE starts, located on IP-address 127.0.0.1
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache.Cleanup.run() starts
2018-07-11 17:31:12 [qtp99695585-28] DEBUG c.g.f.base.http.DigestAuthenticator - Basic verification identifed unit id b310v1 from CPE IP-address 127.0.0.1
2018-07-11 17:31:12 [qtp99695585-28] DEBUG com.github.freeacs.tr069.SessionData - Will load unit data
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 0 elements out of 0 in 0 ms.
2018-07-11 17:31:12 [qtp99695585-28] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      1.96 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, up.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = 'b310v1'
2018-07-11 17:31:12 [qtp99695585-28] DEBUG c.g.freeacs.base.db.DBAccessSession - Found unit b310v1, unittype B310s-518, profile Default
2018-07-11 17:31:12 [qtp99695585-28] DEBUG com.github.freeacs.tr069.SessionData - Found unit in database - in total 2 params (0 unit params, 2 profile params (0 overridden), 0 always read params added)
2018-07-11 17:31:12 [qtp99695585-28] DEBUG com.github.freeacs.tr069.SessionData - Removed 2 system parameter from param-list, now contains 0 params
2018-07-11 17:31:12 [qtp99695585-28] INFO  c.g.f.base.http.BasicAuthenticator - Authentication verified (CPE IP address: 127.0.0.1)
2018-07-11 17:31:12 [qtp99695585-28] DEBUG com.github.freeacs.common.util.Cache - Key 127.0.0.1Basic YjMxMHYxOnBhc3N3b3JkMTIz will be removed from Cache (explicitely - not by Cache.Cleanup.run())
2018-07-11 17:31:12 [qtp99695585-28] DEBUG c.g.f.t.methods.HTTPRequestProcessor - Will process method Inform (incoming request/response from CPE)
[... cut ...]
2018-07-11 17:31:12 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 0 elements out of 0 in 0 ms.
2018-07-11 17:31:12 [Syslog2DB-0] DEBUG c.g.freeacs.syslogserver.Syslog2DB - Syslog message has been parsed (unitId: null)
2018-07-11 17:31:12 [Syslog2DB-0] DEBUG c.g.f.syslogserver.SyslogPackets - Returns a packet from buffer. Buffer size is 0.
2018-07-11 17:31:12 [Syslog2DB-0] DEBUG c.g.freeacs.syslogserver.Syslog2DB - Syslog message has been parsed (unitId: b310v1)
```

I do not know if this model or equipment requires any special customization to be registered by FreeACS.

jarlah, 2018-07-12:
Just to recap, since I am not really sure what the real problem is. I dont see any actual errors in the logs other than the fact that the conversation ends after InformResponse. I am now in the position where i can test this Inform request you sent, with postman. I'll be back soon ;)

jarlah, 2018-07-12:
Ok forget about test. I have a test already with this exact scenario. Your device MUST continue with EM post to tr069 server. INReq-INRes-EM-GPVReq(from server)-GPVRes(from client) etc etc. without acking the inform response nothing else is done. And you can see in syslog that the unit is not responding to the server after INRes from server: "most probably because the client did not respond".

Why doesnt your unit follow up the InformResponse with an empty post?

And No data in DB is not really important in this case. Its not the problem for the conversation not continuing. But it actually just means that you have no RW parameters on the unit, only system parameters maybe?

Anyway, if the conversation is InReq - InRes - nothing. Then nothing more will happen :) make sure your unit follows up with an empty post to server.

jarlah, 2018-07-12:
This has taken alot of time, because I have been into so much code stabilzing to make sure that I can weed out potential errors, that I havent given myself the opportunity to think hard about this. But the quirks stuff you are doing is not relevant. You shouldnt need to add quirks to get past Inform cycle.

jarlah, 2018-07-12:
You can see my test resources here, that is not that difficult from yours. And it also uses discovery. https://github.com/freeacs/freeacs/tree/master/tr069/src/test/resources/discover

jarlah, 2018-07-12:
And easycwmp works so there is no fundamentally wrong at last as far as i know with the server.

jarlah, 2018-07-12:
Even "kicking" the easycwmp works. But i did it manually with the url in the browser though. To make kick work after provisioning easycwmp, one must set the ConnectionRequestPassword parameter manually in the unit parameters. easycwmp returns blank for password, obviously ;) would be wierd if the unit just returns passwords stored in the device for remote access ;)

jarlah, 2018-07-12:
I have ordered a very cheap voip cpe with tr069 support from china. To have something real to test with. But it will  take some time to arrive. But anyway I think your problem is propably device related

penaivanalejandro, 2018-07-12:
Thanks again for the support.

As I mentioned before, I can put the devices with a URL (with a Public IP) that you indicate, so you can see the logs (TR069) or something else that you consider.

jarlah, 2018-07-14:
@penaivanalejandro  nah dont need a client now. Right now, I need you to figure out either via wireshark or via access logs in nginx if the device is connecting with an empty POST after the inform response. I have figured out a lot of problems with the server along the way with other units, so I know that the server is working. So I need you to figure out whats going on after the inform response (if that WAS the problem, that the unit never follows up the dialog). The support i give here, unless i am paid, is strictly advice and pointing fingers in hopefully right directions. When i get my own device, or when i can salvage on of the original devices built for the server, then i can verify on my own for each release that nothing is broken.

jarlah, 2018-08-03:
I can see No data in DB with my own unit. But it doesnt hinder provisioning . Going to release 2.0.0. I will check this issue with a clean install before i release. I fixed a problem with docker lately. Or multiple actually . But I'll check

jarlah, 2018-08-03:
I checked before i made  release 2.0.0. I can now override properties and especially quirks in docker compose . I have also added and fixed many other issues the last month . So i called it finished since i have been testing a lot and know that it works well enough . Have also fixed a bug with download request only present in docker environment.  

Can you verify and eventually close this issue?

Do note that you now need a .env file in same directory as docker compose file. You also need to download/update the docker compose file. Simplest is to run the one liner in wiki.

penaivanalejandro, 2018-08-31:
Hi @Jarl, first of all an apology for the delay in answering and taking the opportunity to say hello and be well.

I just installed the new version you have and has many changes, congratulations ... !!!

I will start testing with the webservice (I try to automate the CPEs that can already be read by FreeACS).

I did tests again using the installation using the configure FreeACS (web) script in the following way:
[image]

And the Log keeps showing the same:

```LOG
2018-08-31 12:27:18 [qtp36707189-23] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node01dhwl9j094uqq10zv426198mwp14 did not return a SessionData object from cache, must create a new SessionData object
2018-08-31 12:27:18 [qtp36707189-23] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node01dhwl9j094uqq10zv426198mwp14 created: 1535736438249, lastAccess:1535736438249, mxInactiveInterval:1800
2018-08-31 12:27:18 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache.Cleanup.run() starts
2018-08-31 12:27:18 [qtp36707189-23] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-08-31 12:27:18 [qtp36707189-23] INFO  c.g.f.base.http.DigestAuthenticator - Send challenge to CPE, located on IP-address 0:0:0:0:0:0:0:1
2018-08-31 12:27:18 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Key node0hctd0n15ts4v4kijey1x8l9p10SESSION with CacheValue com.github.freeacs.common.util.CacheValue@7e01d228 is to be removed
2018-08-31 12:27:18 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - CacheValue has a cleanupnotifier-class which will be executed
2018-08-31 12:27:18 [Cache Cleanup] ERROR c.g.f.base.SessionDataCacheCleanup - SessionData for null (key:node0hctd0n15ts4v4kijey1x8l9p10) was removed from cache aftertimeout - indicate a session which did not terminate correctly
2018-08-31 12:27:18 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Key node01n89gi4d53t2n5nx55h32vll611SESSION with CacheValue com.github.freeacs.common.util.CacheValue@70c62398 is to be removed
2018-08-31 12:27:18 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - CacheValue has a cleanupnotifier-class which will be executed
2018-08-31 12:27:18 [Cache Cleanup] ERROR c.g.f.base.SessionDataCacheCleanup - SessionData for izzib310 (key:node01n89gi4d53t2n5nx55h32vll611) was removed from cache after timeout - indicate a session which did not terminate correctly
2018-08-31 12:27:18 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 2 elements out of 4 in 5 ms.
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node01mnicerjke4tz18cwqasa8o85j15 did not return a SessionData object from cache, must create a new SessionData object
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node01mnicerjke4tz18cwqasa8o85j15 created: 1535736438349, lastAccess:1535736438349, mxInactiveInterval:1800
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.g.f.base.http.DigestAuthenticator - Digest verification of CPE starts, located on IP-address 127.0.0.1
2018-08-31 12:27:18 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache.Cleanup.run() starts
2018-08-31 12:27:18 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 0 elements out of 0 in 0 ms.
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.g.f.base.http.DigestAuthenticator - Digest verification identifed unit id izzib310 from CPE IP-address 127.0.0.1
2018-08-31 12:27:18 [qtp36707189-29] DEBUG com.github.freeacs.tr069.SessionData - Will load unit data
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0.34 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, up.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = 'izzib310'
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.g.freeacs.base.db.DBAccessSession - Found unit izzib310, unittype B310, profile Defaultv1
2018-08-31 12:27:18 [qtp36707189-29] DEBUG com.github.freeacs.tr069.SessionData - Found unit in database - in total 3 params (3 unit params, 0 profile params (0 overridden), 0 always read params added)
2018-08-31 12:27:18 [qtp36707189-29] DEBUG com.github.freeacs.tr069.SessionData - Removed 3 system parameter from param-list, now contains 0 params
2018-08-31 12:27:18 [qtp36707189-29] INFO  c.g.f.base.http.DigestAuthenticator - Authentication verified (CPE IP address: 127.0.0.1)
2018-08-31 12:27:18 [qtp36707189-29] DEBUG com.github.freeacs.common.util.Cache - Key 127.0.0.1"izzib310", realm will be removed from Cache (explicitely - not by Cache.Cleanup.run())
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.g.f.t.methods.HTTPRequestProcessor - Will process method Inform (incoming request/response from CPE)
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.github.freeacs.tr069.methods.INreq - Parsed INreq params, found keyroot:InternetGatewayDevice., parameterkey:, swver:21.329.01.01.1494
2018-08-31 12:27:18 [qtp36707189-29] DEBUG c.github.freeacs.tr069.ParameterKey - No device parameter values found, ACS parameterkey = "No data in DB"
```

jarlah, 2018-10-07:
lets just rule one thing. "No data in DB" means practically nothing. It just means that there are now job parameters or unit parameters with flag RW. If there were job parameters or unit parameters on this unit, then parameterKey would be the hashed version of all the values for all the RW parameters. Dont ask me why now, but thats how it is. And if your unit does not connect further after the servers InformResponse, well then thats a an issue with the cpe. Maybe it doesnt understand the content type or something? Does it expect a certain content type? 

Can we agree on that this has nothing to do with No data in DB?

penaivanalejandro, 2018-10-11:
Thank you very much @jarlah , totally in agreement with you ... we close this topic and we give you the approach referring to the conversation between the CPE and the ACS.

Thank you.

closed 2018-10-16 by penaivanalejandro