freeacs/freeacs #140: Connection Request failed

jogermech, 2018-06-20:
Every time I'm running Provision or Readall commands for unit ACS propose to reboot device to initiate provisioning. I suppose it is related with following message:
"Kick failed at 10:32:39 - require reboot to initiate provisioning :: Neither a public ConnectionRequestURL nor any UDPConnectionRequestAddress was found"
Although "IGD.ManagementServer.ConnectionRequestURL" is provisioned
Any CPE-initiated sessions are running ok, so I can check reading and provisioning by setting low periodicinforminterval value (set frequency >2000).

jarlah, 2018-06-20:
There are some difficulties regarding the use of stun service. Its not just plug and play. Network and firewalls must be open etc. But in your case maybe its regarding something much simpler. I can make a rough check on the implementation.

jogermech, 2018-06-20:
I'm testing in a local network. Stun should not be involved at all, althoug unit status shows NAT=true

jarlah, 2018-06-20:
not currently sure why but the code requires public ip address property aka ```System.X_FREEACS-COM.Device.PublicIPAddress```. Its name has changed since i have removed old name references to old products. The ip address is used to replace the host of the connection request url. As a simple test can you try to set this, and then we can discuss what we need to change

jarlah, 2018-06-20:
hmm i am not sure if my advice is totally correct. But you can try. There is also a problem in the code where it checks if the connection request url is public. It can be toggled off, the public check i mean.

jarlah, 2018-06-20:
to disable public ip check set ```KICK_CHECK-PUBLIC-IP=false``` in etc default

jarlah, 2018-06-20:
the latter is the best option to test locally. But you could also use the public ip approach

jogermech, 2018-06-20:
I updated System.X_FREEACS-COM.Device.PublicIPAddress parameter with real CPE IP, but it doesn't impact result. From logs server still believes that IP is 127.0.0.1.

jogermech, 2018-06-20:
KICK option in default also has no effect
and system public IP parameter is updating when try to ReadALL or Provision

jarlah, 2018-06-20:
if you can read the code, can you check if this test is reasonable?
https://github.com/freeacs/freeacs/pull/142/commits/c10eb269734252ced4f6595b06fa357641ffaa01#diff-cec36b535f75a4c4c5e16ffc5b0d222eR67

jarlah, 2018-06-20:
If the above test is reasonable it might be problems with dbi layer not syncing properly betwewn modules. But need to investigate if that is the case

jogermech, 2018-06-20:
Sorry, I hardly can understand this code

jarlah, 2018-06-20:
1. There exists a connection request url parameter. 
2. Kick is initated with check public ip set to false. 
3. Kick is successful.

But i will try to replicate this manually in the application too. Did you remember to restart stun service with systemctl restart freeacs-stun after changing etc default?

jogermech, 2018-06-20:
My fault. I didn't double-checked with you which module properties should include this option.
I added it to TR069 properties file. Is it wrong? 
And afair I didn't restart stun, I just tried to stop it in the end of all my attempts

jarlah, 2018-06-20:
ok, stun is the thing about disabling public ip check :) try that and then restart.

jogermech, 2018-06-21:
nope. I tried to specify CPE IP as PublicIPAddress, tried empty, with both Public-IP-Check enabled and disabled.
And PublicIPAddress parameter is auto-updated to 127.0.0.1 after CPE comes to ACS by itself (with 2 PeriodicInform)

jarlah, 2018-06-21:
We will definitely found out whats wrong. It might take some time, since I am abroad atm. But I feel that we narrowed it down to such a level that I can try to test this by making a fake stun client server (connection url). This will enable me to see that if there is a defined unit with a connection request url defined, it can be kicked. The only difference is that i am not testing provisioning. But i am sure it will uncover the problem you are facing. However, it might take some time before i can do this. 

Its great that you are testing therefore i appreciate your ongoing patience.

ujeisharon, 2018-06-27:
I got the same problem.  My CPE and FreeACS is in the same subnet not behind the NAT.  the System.Device.PublicIPAddress is the ip address of CPE exactly. But FreeACS still cannot init a provision to CPE actively. Should I do any thing more?

jarlah, 2018-06-27:
@ujeisharon can you provide logs for the kick you initated that didnt go through?

ujeisharon, 2018-07-02:
What below is tr069.log of mine after press "Provision".
============================================
2018-07-02 09:50:18.764 NOTICE wr850n                              Added/changed 644 unittype parameters
2018-07-02 09:50:18.818 WARN   wr850n                              GPVres contained error, try once more and ask for all parameters
2018-07-02 09:50:33.236 INFO   wr850n                              No job found (filter: Status/Type:0, )
2018-07-02 09:50:33.239 WARN   wr850n                              The parameter InternetGatewayDevice.DeviceInfo.VendorConfigFile. is defined in the database,but does not exist in the CPE. Delete it from the unittype or update the firmware! This situation will impact on the performance of the system.
2018-07-02 09:50:33.239 WARN   wr850n                              GPV has been issued twice, but apparantly the reason for the failure of the first GPV-response is not due to missing parameters in the CPE.
2018-07-02 09:50:33.239 NOTICE wr850n                              Sent to CPE: 1 parameters.
2018-07-02 09:50:34.000 NOTICE wr850n                              	InternetGatewayDevice.ManagementServer.PeriodicInformInterval : 82079
2018-07-02 09:50:34.001 NOTICE wr850n                              Added unit parameter InternetGatewayDevice.ManagementServer.PeriodicInformInterval with value 82079
2018-07-02 09:50:34.001 NOTICE wr850n                              Added unit parameter System.X_OWERA-COM.Device.PeriodicInterval with value 82079
2018-07-02 09:52:18.825 INFO    ActiveDeviceDetectionTask: Will check if some devices scheduled to return before Mon Jul 02 08:52:18 CST 2018 are too late
2018-07-02 09:52:18.825 INFO    ActiveDeviceDetectionTask: Have found 0 inactive devices
2018-07-02 09:57:18.825 INFO    ActiveDeviceDetectionTask: Will check if some devices scheduled to return before Mon Jul 02 08:57:18 CST 2018 are too late
2018-07-02 09:57:18.826 INFO    ActiveDeviceDetectionTask: Have found 0 inactive devices
2018-07-02 09:58:25.074 NOTICE  Send challenge to CPE, located on IP-address 10.30.192.222
2018-07-02 09:58:25.092 NOTICE wr850n                              Authentication verified (CPE IP address: 10.30.192.222)
2018-07-02 09:58:25.437 INFO   wr850n                              Periodic Inform recorded too early (-81580 sec). Deviation: -99 %
2018-07-02 09:58:25.439 INFO   wr850n                              EM-Decision is GPV since everything is normal and previous method was either IN or TC (updating LCT and possibly FCT)
2018-07-02 09:58:25.779 WARN   wr850n                              GPVres contained error, try once more and ask for all parameters
==========================================================

ujeisharon, 2018-07-02:
clean the fusion-tr069.log  and  re-press the privision on webUI, but no logs append to the fusion-tr069.log file. Did I check the wrong file? I 've checked the ConnectionRequestURL has been seen from Current value. Should I edit extra profile config for provisioning from ACS ? Thanks.

jarlah, 2018-08-03:
I have done some changes to how this works. Please set ConnectionRequestUsermame and ConnectionRequestPassword and PublicPort to port defined in cpe and PublicProtocol to https/http. It will then dynamically try to create kick url based on PublicIp if Connectionrequesturl is not present as an parameter on the unit.

jarlah, 2018-08-03:
If all units use the same public port for connection request url you can set PublicPort and PublicProtocol in profile.

jarlah, 2018-08-03:
But wait. I havent updated release yet. So it only works in docker atm. I think I'll just release 2.0.0 today. I have fixed many issues the last months and i feel it's ready for primetime . Ill add a messages here when New release is ready.

praveendee, 2018-08-20:
Is the fix you made released yet?

jarlah, 2018-08-20:
Currently there is v2.0.0 out. But i dont have any documentation for the New kick functionality. I have a couple of commits since last version so I'll make a minor release for it and inform you here when it's done.

jarlah, 2018-08-21:
I have now released a minor release candidate. Mostly minor fixes. But at least there is a current release reflecting the contents of the master branch.

jarlah, 2018-10-16:
I have been testing kicking unit a lot over the last couple of releases. So it works now. Not sure however how good local software units work. I use an actual cpe to test with. I  would like some feedback on this, and would like you test latest version 2.0.3, but best option is to close this issue and have you comment it or reopen if stil an issue

closed 2018-10-16 by jarlah