freeacs/freeacs #146: GetParameterValues after each inform from CPE

ujeisharon, 2018-06-27:
FreeACS server send a GetParameterValues with "Device." or "InternetGatewayDevice." for whole unit when recieved the CPE inform.  It cause CPE become heavy to handle the query for each informs. I found that the "unitdiscovery" in quirk parapmeter, does it releted to this? Or where can I config not to get whole parameter values from the root top? Thanks.

jarlah, 2018-06-27:
I am in the process of redesigning the quirks administration. Right now it's a mess. You could take a peek inside the jar file for tr069 (it's a zip) and see the example there. That example can be copied into /etc/default/freeacs-tr069. But note that this will change.

ujeisharon, 2018-07-02:
Sorry for late reply. Because  I could not find the install-or-update-freeacs-ubuntu.sh mentiined in install.pdf  from current https://github.com/freeacs/ . So that I install freeacs by the files download from old.freeacs.com, is it correct?  I suppose the files are old, so that is the method from your suggestion still avaliable? Or how can I install freeacs from current source at https://github.com/freeacs?
Thanks

jarlah, 2018-07-02:
If deploying to same server as old stack you need to stop and disable tomcat. Or uninstall tomcat. If you have a database setup you can download .deb files from release page and install them with sudo dpkg -i *.deb. then go into each /etc/default/freeacs-[module] and make sure connection props matches your existing db. Save and close. Then restart each service with sudo systemctl restart freeacs-[module]. Replace [module] with the module name in question you are deciding to install. 

If New install see instructions on wiki in this repo for a one liner install

ujeisharon, 2018-07-04:
May I know how to compile with the source of the old stack? Is it the same with "sbt" ?  How to do compile with the source checkout from old.freeacs.com? When I try the  latest release with dpkg -i *.deb checkout from github/freeacs , I suffered error  as "/var/lib/dpkg/ingo/freeacs-core.prerm systemctl not found". Not successfully to try the latest freeacs yet.

jarlah, 2018-07-04:
Ok. What OS you are using? I can instead create sytemv deb files in addition. Old source is way to much hassle. It has no build system and you will not get much support from on that code. I have done major changes to the codebase.

But even if you got error. I believe package is installed. Check /opt and /etc.  And you can symlink for eks /opt/freeacs-web/freeacs-web (i dont remember the exact path, since i am on train atm) to /etc/init.d/freeacs-web and then start it with /etc/init.d/freeacs-web start.

Right now the packages are prepared for systemd. If you have systemv you must have some really old Ubuntu or something completely different

ujeisharon, 2018-07-04:
I am working with Ubuntu14.04 and after dpkg -i *.deb, the /opt/ exists the freeacs-* folder and files but there's no  symlink to in /etc/  to /opt/freeacs-*.  And /etc/init.d/freeacs-web doesnot exist also.
What OS you suggest to use? Ubuntu16? Thanks.

jarlah, 2018-07-04:
ubuntu 16 or later yes. In ubuntu 16 they introduced systemd. I see that 14.04 still gets updates a little bit into 2019, but I strongly suggest to upgrade to ubuntu 16 now. In this project i needed to bump the requirements since 14.04 is so old. 

You can see in link below that 16.04 will at least have updates into 2021:

https://www.ubuntu.com/info/release-end-of-life

ujeisharon, 2018-07-05:
When I use the clean Ubuntu16.04 and dpkg -i *.deb can successfully install without error and the symlink created. But I am confused by the installation doc of github/freeacs/installation.  What's the actual openjdk and mysql version should be? If I need to re-compile the source of github/freeacs/freeacs with sbt? And the actual steps to complete the installation?
1. dpkg -i *.deb (download from https://github.com/freeacs/freeacs/releases)
2. apt install mysql-server-5.7, openjdk-jre-7 and  tomcat7
3. how to install the freeacs and create tables that freeacs required?
4. how do some config as installation document mentioned? e.g xaps-tr069.properties

jarlah, 2018-07-05:
I am sorry for the lsck of good documentation. This is next on my list. Meanwhile take a look inside scripts/install.sh.

ujeisharon, 2018-07-06:
OK, thanks for your support. I am looking forward to our update then to try the latest FreeeACS.

closed 2021-09-04 by jarlah