freeacs/freeacs #148: Unit ID on FreeACS

ujeisharon, 2018-07-02:
When I use a CPE1 connected to ACS with ACS username/password as admin/admin. I saw the unit id of CPE1 in search will be the "admin". Is it normal?  And then I use the same config on CPE2 and then the privision history of CPE1 will show status error of value change. How ACS distinguish the two CPEs? From help page, the unit id should be composed with OUI-ProdictClass-SerialNumber . When I pre-create a unit-id as <OUI-ProductClass-SerialNumber> format and then CPE conneted to ACS , ACS cannot assign CPE to be the unit-id I create but a profile  "Default" and unit type "IGD" with unit id as "admin".  Is there any reference document ? Thanks.

ujeisharon, 2018-07-02:
I found the answer from xaps Client Design pdf. The unit Id must be the ACS username web-ui of device. It mean that I can not use the same username for all CPEs, right? Thanks

jarlah, 2018-07-02:
That is correct. I think thats how it works. For auto provision the username is used to find out the unitId to create etc.

closed 2018-07-03 by ujeisharon