freeacs/freeacs #154: Fix badly reported public ip address

jarlah, 2018-07-25:
I have bought and received a cheap voip cpe from china. Just provisioned it with discover. And public url is stored as 127.0.0.1 as reported by other users (you). Now i can reproduce and fix this stupid problem.

jarlah, 2018-07-25:
Nginx problem. Have a solution for it. But there is another problem. The unit i have doesnt answer for ConnectionRequestUrl. But i suppose that if I added a new system parameter lets say PublicPort i could fall back to creating connection request url based on PublicIpAddress and PublicPort. Maybe need some PublicProtocol too. The connection request url will then be created dynamically based on these three parameters if it's not provided by the cpe

jarlah, 2018-07-25:
I would also need to remove the check for if the connection request url is local. If the url is present then try to kick the unit.

jarlah, 2018-07-25:
So a combination of two fixes. 1) dynamically try to create connectionrequesturl if not defined and protocol, port and publicipaddress is present. And 2) disable the check in the code that Kicks the unit wether or not the conn req url is public or not. If it's there use it. Possibly configure it so it can be turned back on if it's really needed to know if it's public or not

jogermech, 2018-07-26:
looking forward for the fix

jarlah, 2018-07-29:
Fixed and merged. But i need to figure out å way to smart merge property files when upgrading services with new deb files. They are currently overridden but user should either be asked what to do or the installer should smart merge if possible

closed 2018-07-29 by jarlah

jarlah, 2018-07-29:
fixed config file problem in https://github.com/freeacs/freeacs/commit/27d52bab8de5b40d340182ab12fe0bee2f675cd0

now when installing or updating and config files have been changed, you will get something like this:

```
Configuration file '/opt/freeacs-tr069/config/application-config.properties'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** application-config.properties (Y/I/N/O/D/Z) [default=N] ?
```