freeacs/freeacs #159: Docker installation monitor not available

jpVm5jYYRE1VIKL, 2018-07-31:
after you press : Monitor in web gui you will get : 

The page could not be displayed
Monitor Service is not running! 
Log extract : 

`
2018-07-31 13:29:51 [qtp2124448375-62] INFO  o.a.c.httpclient.HttpMethodDirector - I/O exception (java.net.ConnectException) caught when processing request: Connection refused (Connection refused)
2018-07-31 13:29:51 [qtp2124448375-62] INFO  o.a.c.httpclient.HttpMethodDirector - Retrying request
2018-07-31 13:29:51 [qtp2124448375-62] INFO  o.a.c.httpclient.HttpMethodDirector - I/O exception (java.net.ConnectException) caught when processing request: Connection refused (Connection refused)
2018-07-31 13:29:51 [qtp2124448375-62] INFO  o.a.c.httpclient.HttpMethodDirector - Retrying request
2018-07-31 13:29:51 [qtp2124448375-62] INFO  o.a.c.httpclient.HttpMethodDirector - I/O exception (java.net.ConnectException) caught when processing request: Connection refused (Connection refused)
2018-07-31 13:29:51 [qtp2124448375-62] INFO  o.a.c.httpclient.HttpMethodDirector - Retrying request
2018-07-31 13:29:51 [qtp2124448375-62] WARN  c.g.f.w.app.page.monitor.MonitorPage - Could not find the monitor server
java.net.ConnectException: Connection refused (Connection refused)
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at java.net.Socket.connect(Socket.java:538)
	at java.net.Socket.<init>(Socket.java:434)
	at java.net.Socket.<init>(Socket.java:286)
........
'

jarlah, 2018-07-31:
yes i am aware of it. the monitor service was removed at one point, in addition to some other services that i also dont want to continue supporting. I am currently thinking about what i can replace it with. 

If you are in docker, you can verify that services are running by issuing docker ps in command line and inspecting output.

jarlah, 2018-07-31:
Going to bring back the monitor service, but it will be redesigned to work with spring boot actuator endpoints, which are much better to assess system status.

jarlah, 2018-07-31:
Brought back to live in https://github.com/freeacs/freeacs/commit/4e4585859baca1ff086ee5badbc5d83ce47e4d7e

jarlah, 2018-08-01:
Fixed. do a docker-compose -p freeacs down and docker-compose -p freeacs up -d (i think it will pull latest images). Monitor should now work.

jpVm5jYYRE1VIKL, 2018-08-01:
nope i cannot confirm that it started to work . i pulled latest version images and 
after it i did "docker-compose -p freeacs up --force-recreate -d" and still have same issue.

jarlah, 2018-08-01:
how are you running the docker compose file? i introduced a new file, .env that needs to be in the same directory as docker-compose.yml. Either download that file or clone the repo. Its in the scripts folder. 

You need to run docker-compose -p freeacs pull .. figured that out for myself now :P

jarlah, 2018-08-01:
[image]

jarlah, 2018-08-01:
i needed an .env file to be able to override host for syslog server and generally being able to override env variables in docker-compose. Oh .. you need to redownload the docker compose file. Its changed

jarlah, 2018-08-01:
see https://github.com/freeacs/freeacs/issues/162 for how to get stuff working after my last changes.

closed 2018-08-01 by jarlah