freeacs/freeacs #160: Docker installation : application-config.properties file ignored

jpVm5jYYRE1VIKL, 2018-07-31:
After you add application-config.properties to freeacs/conf/web 
which contain : 
`monitor.location = http://localhost:8080`
and execute docker-compose -p freeacs up -d
it is just ignored and not have any influence on web instance.

jarlah, 2018-07-31:
try to change something else. Like main.datasource.driverClassName=ThizShouldBreakWeb
i fixed config file problem lately in commit https://github.com/freeacs/freeacs/commit/27d52bab8de5b40d340182ab12fe0bee2f675cd0 so it should work.

jarlah, 2018-07-31:
nevermind .. that wont work because its overridden by docker compose

jarlah, 2018-07-31:
but it should work. there is actually not many interesting properties to edit in web :P

jarlah, 2018-07-31:
I will check this problem when i bring back and redesign the monitor service.

jarlah, 2018-08-01:
I just tested by adding application-config.properties and setting something like PROPERTIES=alepo (changes the styling) or PROPERTIES=skjsjkad (breaks the web, will not start, because its not a valid profile). If its still a problem you can just reopen this issue or open a new one

closed 2018-08-01 by jarlah