freeacs/freeacs #161: Docker installation : freeacs not correctly send TR069 script url

jpVm5jYYRE1VIKL, 2018-08-01:
If you create job for upload TR69-script
it send to CPE :

> <soapenv:Body>
>                 <cwmp:Download xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
>                         <CommandKey>1</CommandKey>
>                         <FileType>3 Vendor Configuration File</FileType>
>                         <URL>http:/:8080/tr069/file/TR069_SCRIPT/1/vendor/serial/ttt.script</URL>
>                         <Username></Username>
>                         <Password></Password>
>                         <FileSize>2414</FileSize>
>                         <TargetFileName>ttt.rsc</TargetFileName>
>                         <DelaySeconds>0</DelaySeconds>
>                         <SuccessURL></SuccessURL>
>                         <FailureURL></FailureURL>
>                 </cwmp:Download>
> </soapenv:Body>
xml line with URL is wrong. May be it is good idea to make configurable . May be it is possible already to configure but not see such option in option files for project tr069.

jarlah, 2018-08-01:
thanks for the bug report. This is probably an easy bug to fix, hopefully :)

jarlah, 2018-08-01:
I have tried to fix this bug in https://github.com/freeacs/freeacs/commit/32a002dd0840892ddb03b2fb938ef3ea4baaeb0d

you need to add ```public.url = http://localhost``` with correct hostname/ip for your machine, lets say ```public.url = http://192.168.0.25```, in conf/tr069/application-config.properties. Do docker-compose -p freeacs down and then docker-compose -p freeacs up -d. Calling down just removes the containers. Not the data. The data is stored in volume containers that are not removed by default.

jarlah, 2018-08-01:
The problem was actually docker specific in that the hostname for the tr069 container was the same as the context path. If not running in docker this new property must also be configured to a sensible hostname/ip. Its impossible for the service to know where the tr069 server is running. It can even run on a different server. So defining a public url property for download is almost required. It will maybe be a part of some installation script some day or at least wiki

jpVm5jYYRE1VIKL, 2018-08-02:
issue can be closed . With changed port in nginx configuration or application.properties in web project it works.

jarlah, 2018-08-02:
even if you fixed port, and its wrong solution (see #165 ) it doesnt solve the problem that caused the bad url. I fixed the bad url to work. So download works now? after you update public.url property?

jpVm5jYYRE1VIKL, 2018-08-02:
yes can be closed. public.url property works good.

jarlah, 2018-08-02:
Closing for now, but if any new problems with Download occurs don't hesitate to open new issue :)

closed 2018-08-02 by jarlah