freeacs/freeacs #162: Docker install: Random nginx issues after restart

jpVm5jYYRE1VIKL, 2018-08-01:
Quite recent after command 
"docker-compose -p freeacs restart"
after it finish if you open page 
http://acs.server.net/web/
you will see page 

```
HTTP ERROR 404
Problem accessing /web/. Reason:
Not Found
```
In same moment NGINX container show in logs 

1.1.1.1 - - [01/Aug/2018:13:23:15 +0000] "GET /web/ HTTP/1.1" 404 236 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"
1.1.1.1 - - [01/Aug/2018:13:24:36 +0000] "GET /web/ HTTP/1.1" 404 236 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"

Normally if to execute "docker-compose -p freeacs restart" one more time it magically start to work again.

jarlah, 2018-08-01:
several things. 
1. docker-compose -p freeacs down
2. docker-compose -p freeacs pull
3. Download new docker-compose.yml file from scripts folder if not running in cloned repo
4. Download .env file from scripts folder if not running in cloned repo
5. docker-compose -p freeacs up -d

jarlah, 2018-08-01:
.env file needs to be in same directory as docker-compose.yml

jarlah, 2018-08-01:
I suppose this can be closed since you got https://github.com/freeacs/freeacs/issues/159 working. If not just reopen it.

closed 2018-08-01 by jarlah