freeacs/freeacs #167: MySQL connection with Docker containers

yarons, 2018-08-05:
I tried several different passwords after launching the system with `docker-compose up -d`.

After several failing password attempts I realized there might be a problem with the DB connection due to the following error:

    "Host '172.19.0.9' is not allowed to connect to this MySQL server"

So I figured there might be a problem while the `web/webservice` containers are trying to reach the MySQL server.
I can add more logs if needed but this is my first attempt at trying to figure out what might be wrong, I'll see if I can modify this user.

jarlah, 2018-08-05:
So you have seen the fact that I expose 3306 to the host system ;) ? Nice. It should work. I even make containers depend on the database container. Its not bulletproof, and other containers might start faster than the MySQL container anyway so. But check with docker ps that all containers are running including mysql. Then check docker logs -f freeacs_db_1 to see whats going inside the MySQL container. If you are on Unix/Linux, do a telnet [host] 3306 etc to check that the port is listening. Finally I have actually used mostly mysql -h 0.0.0.0 -u acs -p for some reason lately. Guess i just figured out that it worked, but never questioned it.

yarons, 2018-08-06:
So further configuration is needed in order to make it work? It's a little counter intuitive to the way docker-compose should work.

I experienced the loading order issue you're talking about, I just knew I had to wait a little longer to get there.
I've also explored the database with itself, there's no special schema or anything that might be related to freeacs, it looks like the upfront script cannot access the database.

The containers were all running and even the UI was working but no matter which username/password combination I've tried, nothing worked, so I launched `docker-compose logs`, this is where I got all these errors from.

BTW, am I supposed to run one of the scripts residing in the scripts directory or running `docker-compose up` should do the trick?

Thanks.

yarons, 2018-08-06:
I sure will, I cloned the git repo so I guess it was all as planned, am I right?

jarlah, 2018-08-06:
Cloning is not really necessary. But if you want just do it. So the steps you have to reproduce. Is to clone and run docker-compose up -d? And it's consistent (wierdly enough) over many attempts ?

jarlah, 2018-08-07:
I have two things to test now. Docker and database problem. And no logging when using deb files ;) i will install a local virtualbox with CentOS. What version?

jarlah, 2018-08-07:
do docker-compose pull before creating/starting docker-compose

yarons, 2018-08-08:
Sure, I will, BTW docker-compose up will pull the containers if these are not present yet.

Thank you @jarlah !

jarlah, 2018-08-08:
Thank you too :)  without anyone caring or being interested in all this, it would be a waste . I think the market for tr069 servers needs an open solution that just works. That can be deployed on many different environments. This stack can actually be deployed anywhere. Even Windows. I dont have a package for Windows yet but it's possible to create universal binaries that contains a bat and Shell script;) i wonder if any of the other servers out there is so flexible. I think my microservices use a tad too much ram but i think it's more greediness in Java than it is actual usage. But thats my next goal. To investigate actual memory usage

jarlah, 2018-09-05:
I consider this "solved" by the last two comments. If there is any more problems, please post a new issue.

closed 2018-09-05 by jarlah

yarons, 2018-09-06:
I'll test that again in a couple of weeks and post my results but I'm guessing it should be fine :smile: