freeacs/freeacs #168: Installation aborts with  "./install.sh: line 101: install.sql: No such file or directory wc: .tmp: No such file or directory"

hmeraee, 2018-08-08:
Hi all,
May be this is a stupid question but I am really stuck, I have installed mysql and tomcat jkd, and now
I'm trying to install freeacs using this  wget https://raw.githubusercontent.com/freea ... install.sh -v -O install.sh && chmod +x install.sh && sudo ./install.sh;
but I'm getting these errors

Downloads all necessary resources from freeacs.com:

error: test is not defined
.assets[] | select(.name | test("deb|tables|pdf")) | .browser_download_url
                           ^^^^
1 compile error
wget: missing URL
Usage: wget [OPTION]... [URL]...

Try `wget --help' for more options.
unzip:  cannot find or open *.zip, *.zip.zip or *.zip.ZIP.

No zipfiles found.

All necessary FreeACS resources are available.

Then it asks me for mysql root password, I write it then it gives this error

Specify/create the password for the FreeACS MySQL user.
NB! The FreeACS MySQL user name defaults to 'acs'
NB! If the user has been created before: Do not try
to change the password - this script will not handle
the change of password into MySQL, but the configuration
files will be changed - causing a password mismatch!!
Using mysql root pw

The FreeACS MySQL database user is OK.

Loads all FreeACS table defintions into MySQL
./install.sh: line 101: install.sql: No such file or directory
wc: .tmp: No such file or directory
The output from the installation of the tables indicate some
errors occurred:
------------------------------------------------
cat: .tmp: No such file or directory

and the installation aborts
I searched the forum but didn't find something similar, the installation guide is outdated

Thanks

jarlah, 2018-08-09:
using something older than ubuntu 16.04? do you have bash?

hmeraee, 2018-08-09:
Oh, I've updated ubuntu and now the installation completed :)
Many thanks to you

closed 2018-08-09 by hmeraee