freeacs/freeacs #169: Could not compile source code due to the error "unmappable character for encoding UTF-8"

sabbane, 2018-08-09:
By trying to compile the source code many character mapping errors are displayed and the build fails. Here is one of these errors:
`freeacs\webservice\target\src_managed\main\com\github\freeacs\ws\xml\AddOrChangeProfileRequest.java:4:1:` unmappable character for encoding UTF-8`

After changing the coding within "build.sbt" from "UTF-8" to "ISO-8859-1" the build process is successful. So, I think the character encoding should be changed to "ISO-8859-1".

jarlah, 2018-08-09:
Havent tested on Windows so it's just a shot in the dark.

jarlah, 2018-08-10:
i suggest using Intellij and to configure UTF-8 file encoding in Intellij settings
[image]
Try dat.

sabbane, 2018-08-10:
Indeed it was a system configuration. I'm compiling within windows powershell (I'll try with eclipse later) and now the build is successful after setting the following system environement variable:
`JAVA_TOOL_OPTIONS: -Dfile.encoding=UTF8`

jarlah, 2018-08-10:
Great :) Closing ;)

closed 2018-08-10 by jarlah