freeacs/freeacs #172: Firmware file upload fails

jogermech, 2018-08-21:
Tried to upload firmware file via Files and Scripts -> Files menu and got nginx error after press "Upload file" button:
"413 Request Entity Too Large"

jogermech, 2018-08-21:
2.0.1 fresh install by install.sh

jogermech, 2018-08-21:
FW file size is about 16M

jarlah, 2018-08-21:
Ok. Thats solvable. If it's install script based we can hot patch your nginx config while you wait for me to patch the service. If it's docker i can also fix that. See https://www.google.no/amp/s/www.cyberciti.biz/faq/linux-unix-bsd-nginx-413-request-entity-too-large/amp/ for hotfix.

jarlah, 2018-10-18:
Closing this as there is bigger fishes to fry atm. You can follow guide above to adjust nginx to allow bigger upload payloads

closed 2018-10-18 by jarlah