freeacs/freeacs #174: <SetParameterValues> via NBI/Webservice

penaivanalejandro, 2018-09-04:
Hi @jarlah , I'm sorry, I do not know which way to contact you, so I do it over here ...

I was able to start working with the STUN Server module, I configured a new parameter from the Web page, I pressed the "Provision" button and the FreeACS does the rest of the wonder.

This same I would like to do it via NBI, to be able to set the parameter and send the "Provision" that is sent via Web but by webservice.

When checking the LOG the command of <SetParameterValues> is observed, there is a way to do this without waiting for the PeriodicInformInterval to be fulfilled and the CPE come to read everything from the Server (FreeACS)

```XML
2018-09-04 18:41:03 [qtp1479100519-22] INFO  Conversation - =============== FROM ACS TO ( 434v3test ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
	<soapenv:Header>
		<cwmp:ID soapenv:mustUnderstand="1">FREEACS-1536104463012</cwmp:ID>
	</soapenv:Header>
	<soapenv:Body>
		<cwmp:SetParameterValues>	<!--How send this command via NBI-->
			<ParameterList soapenc:arrayType="cwmp:ParameterValueStruct[4]">
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.ManagementServer.PeriodicInformInterval</Name>
					<Value xsi:type="xsd:unsignedInt">73389</Value>
				</ParameterValueStruct>
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.ManagementServer.ConnectionRequestPassword</Name>
					<Value xsi:type="xsd:string">admin</Value>
				</ParameterValueStruct>
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.LANDevice.1.WLANConfiguration.2.PreSharedKey.1.KeyPassphrase</Name>
					<Value xsi:type="xsd:string">Clave5G</Value>
				</ParameterValueStruct>
				<ParameterValueStruct>
					<Name>InternetGatewayDevice.LANDevice.1.WLANConfiguration.1.PreSharedKey.1.KeyPassphrase</Name>
					<Value xsi:type="xsd:string">Clave24G</Value>
				</ParameterValueStruct>
			</ParameterList>
			<ParameterKey>af84b11484cad2507682f05738bc75d0</ParameterKey>
		</cwmp:SetParameterValues>
	</soapenv:Body>
</soapenv:Envelope>
```

jarlah, 2018-09-30:
What about POST http://localhost:8088/webservice/ws/AddOrChangeUnitRequest
You send in the same as with GetUnits, but it wil update the unit and its parameters.

jarlah, 2018-09-30:
POST http://localhost:8088/webservice/ws/AddOrChangeUnitRequest (the latter part of the url could be anything. its the method in the payload that counts)

```
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:x.="http://xml.ws.freeacs.github.com/">
   <soapenv:Header/>
   <soapenv:Body>
      <x.:AddOrChangeUnitRequest>
         <x.:login>
            <x.:username>admin</x.:username>
            <!--Optional:-->
            <x.:password>freeacs</x.:password>
         </x.:login>
         <x.:unit>
            <x.:unitId>sddsdddd</x.:unitId>
            <x.:profile>
            	<x.:name>Default</x.:name>
            </x.:profile>
            <x.:unittype>
            	<x.:name>sddd</x.:name>
            </x.:unittype>
            <x.:parameters>
	    		<x.:parameterArray>
	    			<x.:item>
	    				<x.:name>System.X_FREEACS-COM.Comment</x.:name>
	    				<x.:value>Hello world</x.:value>
	    			</x.:item>
	    		</x.:parameterArray>
            </x.:parameters>
         </x.:unit>
      </x.:AddOrChangeUnitRequest>
   </soapenv:Body>
</soapenv:Envelope>
```

gives

```
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/">
    <SOAP-ENV:Header/>
    <SOAP-ENV:Body>
        <ns2:AddOrChangeUnitResponse xmlns:ns2="http://xml.ws.freeacs.github.com/">
            <ns2:unit>
                <ns2:unitId>sddsdddd</ns2:unitId>
                <ns2:profile>
                    <ns2:name>Default</ns2:name>
                </ns2:profile>
                <ns2:unittype>
                    <ns2:name>sddd</ns2:name>
                </ns2:unittype>
                <ns2:parameters>
                    <ns2:parameterArray>
                        <ns2:item>
                            <ns2:name>System.X_FREEACS-COM.Comment</ns2:name>
                            <ns2:value>Hello world</ns2:value>
                        </ns2:item>
                    </ns2:parameterArray>
                </ns2:parameters>
            </ns2:unit>
        </ns2:AddOrChangeUnitResponse>
    </SOAP-ENV:Body>
</SOAP-ENV:Envelope>
```

jarlah, 2018-09-30:
the response is actually stupid, because it just echoes the unit info. But it does magic on the backend, updating parameters

jarlah, 2018-09-30:
This just showcase that you can set params. I need to think about the webservice module. It has terrible security mechanism. I need to assess how it can be fixed in short and long term.

jarlah, 2018-09-30:
Its very simple to add a KickUnit method in the soap interface. But i need to think about authentication and security first. First thought is that a default setup is too insecure. I need to generate a random password on each startup of each module, so that each module has it owns admin password when no admin password is set. This is how spring security works by default when no security is setup. 

Next, I need to find out why there is a Login object in the request. I havent seen this very insecure method of authentication anywhere else before. Its actually a bit nice, because the user/pass is hidden by ssl/tls, instead of readable as in digest or basic (because they are in headers). But i am not sure about this. Maybe the automatic admin password generation is enough to make sure that new installations is not easily breakable. But we should think about requiring checksummed passwords in the Login object, and not the cleartext password.

penaivanalejandro, 2018-10-16:
> Its very simple to add a KickUnit method in the soap interface. But i need to think about authentication and security first. First thought is that a default setup is too insecure. I need to generate [...]

Hello @jarlah , good day ... !!

I was testing the webservice http://localhost:8088/webservice/ws/AddOrChangeUnitRequest

There is some way to push the provisioning or some change directly to the CPE and not wait for the time when the UnitID is reported again.

And to be able to give a kick through the SOAP to the CPE.
¿?

jarlah, 2018-10-16:
https://github.com/freeacs/freeacs/pull/205

jarlah, 2018-10-16:
I have added a KickUnitRequest/KickUnitResponse request/response to webservice.

jarlah, 2018-10-16:
```
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:x.="http://xml.ws.freeacs.github.com/">
   <soapenv:Header/>
   <soapenv:Body>
      <x.:KickUnitRequest>
         <x.:login>
            <x.:username>admin</x.:username>
            <!--Optional:-->
            <x.:password>freeacs</x.:password>
         </x.:login>
         <x.:unitId>easycwmp</x.:unitId>
      </x.:KickUnitRequest>
   </soapenv:Body>
</soapenv:Envelope>
```

jarlah, 2018-10-16:
@penaivanalejandro please verify and we will close this issue. It will come in next ordinary release or you can use it in docker.

penaivanalejandro, 2018-10-16:
I started update when finish i start to test again..!
Give 20 minutes

penaivanalejandro, 2018-10-16:
Sorry I misunderstood, I thought that the change was already in place (in the webservice), the XML that you send would be excellent as a complement to the "AddOrChangeUnitRequest" command.

jarlah, 2018-10-17:
I Will release a minor minor version with this change today so it can be used on Ubuntu and centos

penaivanalejandro, 2018-10-17:
Aware,
I'm ready to test when he's released.

Greetings.

jarlah, 2018-10-17:
@penaivanalejandro ready to test on ubuntu now.

jarlah, 2018-10-17:
and now also rpm files. Its a veeery tedious process to create rpm files, so i really should consider making a docker based script that can build them for me.

penaivanalejandro, 2018-10-17:
Sir (@jarlah ) I want to congratulate you, it works correctly through the webservice ...

When sending the kick it shows the following message in the webservice log:

```LOG
2018-10-17 16:00:19 [qtp181326224-24] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0.38 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, u
p.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = 'IPHO00510BCF'
2018-10-17 16:00:20 [DBI for WebService] DEBUG com.github.freeacs.dbi.DBI - Message: [MTYPE: PUBLISH-IM, OTYPE: UNIT, FROM: WebService, TO: STUN, OBJECTID: 533660:IPHO
00510BCF] sent/inserted
```

In the CPE the change was immediately applied, thank you very much for this new module (kick).
If something comes out he reported it to you