freeacs/freeacs #175: WebService "GetUnits"

penaivanalejandro, 2018-09-05:
Hi @jarlah again me....

I try to use the webservice called "GetUnits" with the following configuration:

```XML
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:x.="http://xml.ws.freeacs.github.com/">
   <soapenv:Header/>
   <soapenv:Body>
      <x.:GetUnitsRequest>
         <x.:login>
            <x.:username>user</x.:username>
            <!--Optional:-->
            <x.:password>password</x.:password>
         </x.:login>
         <x.:unit>
            <x.:unitId>434v3test</x.:unitId>
         </x.:unit>
      </x.:GetUnitsRequest>
   </soapenv:Body>
</soapenv:Envelope>
```

And it answers the following:

```XML
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/">
   <SOAP-ENV:Header/>
   <SOAP-ENV:Body>
      <ns2:GetUnitsResponse xmlns:ns2="http://xml.ws.freeacs.github.com/">
         <ns2:units>
            <ns2:unitArray>
               <ns2:item xsi:nil="true" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"/>
            </ns2:unitArray>
         </ns2:units>
         <ns2:moreUnits>false</ns2:moreUnits>
      </ns2:GetUnitsResponse>
   </SOAP-ENV:Body>
</SOAP-ENV:Envelope>
```

And it does not return information that specifies the PDF:
https://github.com/freeacs/freeacs/blob/master/webservice/docs/Web%20Services%20User%20Manual.pdf

7.2 use case 1: search for a Single Unit, kwown Unit id   (page 17 of 32)

I pass the CPE screen that is currently registered in the platform and the parameters that it has:

[image]

jarlah, 2018-09-25:
POST http://localhost:8088/webservice/ws/GetUnitsRequest
```
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:x.="http://xml.ws.freeacs.github.com/">
   <soapenv:Header/>
   <soapenv:Body>
      <x.:GetUnitsRequest>
         <x.:login>
            <x.:username>admin</x.:username>
            <!--Optional:-->
            <x.:password>freeacs</x.:password>
         </x.:login>
         <x.:unit>
            <x.:unitId>sddsdddd</x.:unitId>
         </x.:unit>
      </x.:GetUnitsRequest>
   </soapenv:Body>
</soapenv:Envelope>
```
returns this with my in progress fix:
```
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/">
    <SOAP-ENV:Header/>
    <SOAP-ENV:Body>
        <ns2:GetUnitsResponse xmlns:ns2="http://xml.ws.freeacs.github.com/">
            <ns2:units>
                <ns2:unitArray>
                    <ns2:item>
                        <ns2:unitId>sddsdddd</ns2:unitId>
                        <ns2:profile>
                            <ns2:name>Default</ns2:name>
                        </ns2:profile>
                        <ns2:unittype>
                            <ns2:name>sddd</ns2:name>
                        </ns2:unittype>
                        <ns2:parameters>
                            <ns2:parameterArray>
                                <ns2:item>
                                    <ns2:name>System.X_FREEACS-COM.Comment</ns2:name>
                                    <ns2:value>fdfff</ns2:value>
                                </ns2:item>
                            </ns2:parameterArray>
                        </ns2:parameters>
                    </ns2:item>
                </ns2:unitArray>
            </ns2:units>
            <ns2:moreUnits>false</ns2:moreUnits>
        </ns2:GetUnitsResponse>
    </SOAP-ENV:Body>
</SOAP-ENV:Envelope>
```
is this enough?

jarlah, 2018-09-26:
I think the webservice  module has been having a hard time in the rewrite. But i can be fixed.

penaivanalejandro, 2018-09-26:
Hello @jarlah, thank you very much for the fix,

Regarding if it is enough ?, if it shows the parameters enabled and the value of the parameter if it is enough.

jarlah, 2018-09-27:
@penaivanalejandro i released 2.0.2 but i cannot make rpm files from now on. Its either docker or ubuntu. I am working on modifying install script for centos to convert deb to rpm files.

closed 2018-09-27 by jarlah