freeacs/freeacs #178: STUN service mysql driver

UnAfraid, 2018-09-25:
Hello,

I've just started the docker containers as shown in the wiki, and i've notice that monitoring was saying stun failed to start, then i checked the logs and i found

```log
root@docker02:~/docker/nginx# docker logs -f nginx_stun_1

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::        (v2.0.2.RELEASE)

2018-09-26 00:10:28 [main] INFO  com.github.freeacs.stun.App - Starting App v2.0.2-SNAPSHOT on 519d19fece2e with PID 1 (/opt/docker/lib/com.github.freeacs.freeacs-stun-2.0.2-SNAPSHOT.jar started by daemon in /opt/docker)
2018-09-26 00:10:28 [main] DEBUG com.github.freeacs.stun.App - Running with Spring Boot v2.0.2.RELEASE, Spring v5.0.6.RELEASE
2018-09-26 00:10:28 [main] INFO  com.github.freeacs.stun.App - The following profiles are active: config
Loading class `com.mysql.jdbc.Driver'. This is deprecated. The new driver class is `com.mysql.cj.jdbc.Driver'. The driver is automatically registered via the SPI and manual loading of the driver class is generally unnecessary.
2018-09-26 00:11:01 [main] WARN  c.z.hikari.util.DriverDataSource - Registered driver with driverClassName=com.mysql.jdbc.Driver was not found, trying direct instantiation.
2018-09-26 00:11:07 [main] ERROR com.zaxxer.hikari.pool.HikariPool - mainPool - Exception during pool initialization.
java.sql.SQLNonTransientConnectionException: Could not create connection to database server. Attempted reconnect 3 times. Giving up.
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:108)
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:95)
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:87)
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:61)
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:71)
        at com.mysql.cj.jdbc.ConnectionImpl.connectWithRetries(ConnectionImpl.java:932)
        at com.mysql.cj.jdbc.ConnectionImpl.createNewIO(ConnectionImpl.java:857)
        at com.mysql.cj.jdbc.ConnectionImpl.<init>(ConnectionImpl.java:444)
        at com.mysql.cj.jdbc.ConnectionImpl.getInstance(ConnectionImpl.java:230)
        at com.mysql.cj.jdbc.NonRegisteringDriver.connect(NonRegisteringDriver.java:226)
        at com.zaxxer.hikari.util.DriverDataSource.getConnection(DriverDataSource.java:117)
        at com.zaxxer.hikari.util.DriverDataSource.getConnection(DriverDataSource.java:123)
        at com.zaxxer.hikari.pool.PoolBase.newConnection(PoolBase.java:367)
        at com.zaxxer.hikari.pool.PoolBase.newPoolEntry(PoolBase.java:196)
        at com.zaxxer.hikari.pool.HikariPool.createPoolEntry(HikariPool.java:467)
        at com.zaxxer.hikari.pool.HikariPool.checkFailFast(HikariPool.java:541)
        at com.zaxxer.hikari.pool.HikariPool.<init>(HikariPool.java:115)
        at com.zaxxer.hikari.HikariDataSource.getConnection(HikariDataSource.java:112)
        at com.github.freeacs.dbi.util.ACSVersionCheck.versionCheck(ACSVersionCheck.java:48)
        at com.github.freeacs.dbi.Users.<init>(Users.java:30)
        at com.github.freeacs.stun.StunServlet.initializeDBI(StunServlet.java:61)
        at com.github.freeacs.stun.StunServlet.trigger(StunServlet.java:70)
        at com.github.freeacs.stun.StunServlet.init(StunServlet.java:47)
        at javax.servlet.GenericServlet.init(GenericServlet.java:244)
        at org.eclipse.jetty.servlet.ServletHolder.initServlet(ServletHolder.java:670)
        at org.eclipse.jetty.servlet.ServletHolder.initialize(ServletHolder.java:427)
        at org.eclipse.jetty.servlet.ServletHandler.initialize(ServletHandler.java:760)
        at org.springframework.boot.web.embedded.jetty.JettyEmbeddedWebAppContext$JettyEmbeddedServletHandler.deferredInitialize(JettyEmbeddedWebAppContext.java:46)
        at org.springframework.boot.web.embedded.jetty.JettyEmbeddedWebAppContext.deferredInitialize(JettyEmbeddedWebAppContext.java:36)
        at org.springframework.boot.web.embedded.jetty.JettyWebServer.handleDeferredInitialize(JettyWebServer.java:209)
        at org.springframework.boot.web.embedded.jetty.JettyWebServer.start(JettyWebServer.java:141)
        at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.startWebServer(ServletWebServerApplicationContext.java:300)
        at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.finishRefresh(ServletWebServerApplicationContext.java:162)
        at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:553)
        at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.refresh(ServletWebServerApplicationContext.java:140)
        at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:759)
        at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:395)
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:327)
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:1255)
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:1243)
        at com.github.freeacs.stun.App.main(App.java:24)
Caused by: com.mysql.cj.exceptions.CJCommunicationsException: Communications link failure

The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
[... 148 lines cut ...]
# is running the requests and may in some circumstance wait
# for reply from the devices. (If not TR-111 is supported)
kick.interval = 1000

# Specify the interval in minutes between each time a job/group is scanned
# for changes in the group (units added or removed). If set too
# low it may cause higher load on server. This rescan will only
# happen if the kick-process is idle (not actively kicking any devices)
# Default is 60 minutes.
kick.rescan = 60

# The kick-feature of the STUN server will try to use the public IP
# of the device in the ConnectionRequestURL, if the ConnectionRequstURL
# is private and UDPConnectionRequestAddress (TR-111) is not defined.
# Default is false
kick.expect-port-forwarding = false

# Check if ip is public
kick.check-public-ip = false

# This is for test-mode only, never used in production. Default is true.
# If set to false, the STUN-server will not start, and it will only support
# ConnectionRequest to the devices (not UDPConnectionRequest)
test.runwithstun = true
```

jarlah, 2018-10-06:
It might happen from time to time. Is it repeatable?

UnAfraid, 2018-10-08:
Always for me tho

jarlah, 2018-10-08:
i have btw updated the docker images .. so you can do a docker-compose [-p freeacs] pull

UnAfraid, 2018-10-08:
> you have a .env file in the directory where you run docker compose?

I do have .env file in my directory of course

jarlah, 2018-10-09:
Ok. I noticed that one of my services also logged errors while starting but it was just while the db  was booting. After a while all services was running. Are you running Windows og Linux/Unix?

jarlah, 2018-10-09:
Sorry about guessing game here. This is a hobby project for me (used by many) and i might not always be the most effective help. But hey we will fix it ;)

UnAfraid, 2018-10-09:
I am running Ubuntu server 18.04 and docker 18.06

jarlah, 2018-10-09:
Ok. To replicate this issue i will do the following. Install Ubuntu server 18.04 in virtualbox and then run the script as noted in wiki. Reports back when it's tested.

jarlah, 2018-10-09:
Tried it. I only gave virtualbox 1 virtual cpu. It took a great deal of time for the mysql container to boot properly. And because i cannot make the containers start in specific order you must just wait. The services will eventually catch up when they automatically restart due to failures. 

But another thing i found when doing this live field test is that there was a bug in tr069 server. https://github.com/freeacs/freeacs/pull/192 and it will soon be merged

jarlah, 2018-10-09:
i have btw changed the tr069 server to only accept provisionin requests on http://localhost/tr069/prov ... before you had to always have the forward slash. Now you must always have prov at the end but you dont need a further slash ;)

jarlah, 2018-10-09:
test tr069 is running by executing curl -X POST http://localhost/tr069/prov ;) and you should get Unauthorized

UnAfraid, 2018-10-09:
At first boot it died, after a restart of service it worked fine
[image]

```log
root@docker01:~/docker/freeacs# docker logs -f freeacs_stun_1

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::        (v2.0.2.RELEASE)

2018-10-09 18:21:29 [main] INFO  com.github.freeacs.stun.App - Starting App v2.0.3-SNAPSHOT on 67df7d2a98f6 with PID 1 (/opt/docker/lib/com.github.freeacs.freeacs-stun-2.0.3-SNAPSHOT.jar started by daemon in /opt/docker)
2018-10-09 18:21:29 [main] DEBUG com.github.freeacs.stun.App - Running with Spring Boot v2.0.2.RELEASE, Spring v5.0.6.RELEASE
2018-10-09 18:21:29 [main] INFO  com.github.freeacs.stun.App - The following profiles are active: config
Loading class `com.mysql.jdbc.Driver'. This is deprecated. The new driver class is `com.mysql.cj.jdbc.Driver'. The driver is automatically registered via the SPI and manual loading of the driver class is generally unnecessary.
2018-10-09 18:21:49 [main] WARN  c.z.hikari.util.DriverDataSource - Registered driver with driverClassName=com.mysql.jdbc.Driver was not found, trying direct instantiation.
2018-10-09 18:21:55 [main] ERROR com.zaxxer.hikari.pool.HikariPool - mainPool - Exception during pool initialization.
java.sql.SQLNonTransientConnectionException: Could not create connection to database server. Attempted reconnect 3 times. Giving up.
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:108)
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:95)
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:87)
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:61)
        at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:71)
        at com.mysql.cj.jdbc.ConnectionImpl.connectWithRetries(ConnectionImpl.java:932)
        at com.mysql.cj.jdbc.ConnectionImpl.createNewIO(ConnectionImpl.java:857)
        at com.mysql.cj.jdbc.ConnectionImpl.<init>(ConnectionImpl.java:444)
        at com.mysql.cj.jdbc.ConnectionImpl.getInstance(ConnectionImpl.java:230)
        at com.mysql.cj.jdbc.NonRegisteringDriver.connect(NonRegisteringDriver.java:226)
        at com.zaxxer.hikari.util.DriverDataSource.getConnection(DriverDataSource.java:117)
        at com.zaxxer.hikari.util.DriverDataSource.getConnection(DriverDataSource.java:123)
        at com.zaxxer.hikari.pool.PoolBase.newConnection(PoolBase.java:367)
        at com.zaxxer.hikari.pool.PoolBase.newPoolEntry(PoolBase.java:196)
        at com.zaxxer.hikari.pool.HikariPool.createPoolEntry(HikariPool.java:467)
        at com.zaxxer.hikari.pool.HikariPool.checkFailFast(HikariPool.java:541)
        at com.zaxxer.hikari.pool.HikariPool.<init>(HikariPool.java:115)
        at com.zaxxer.hikari.HikariDataSource.getConnection(HikariDataSource.java:112)
        at com.github.freeacs.dbi.util.ACSVersionCheck.versionCheck(ACSVersionCheck.java:50)
        at com.github.freeacs.dbi.Users.<init>(Users.java:45)
        at com.github.freeacs.stun.StunServlet.initializeDBI(StunServlet.java:66)
        at com.github.freeacs.stun.StunServlet.trigger(StunServlet.java:75)
        at com.github.freeacs.stun.StunServlet.init(StunServlet.java:50)
        at javax.servlet.GenericServlet.init(GenericServlet.java:244)
        at org.eclipse.jetty.servlet.ServletHolder.initServlet(ServletHolder.java:670)
        at org.eclipse.jetty.servlet.ServletHolder.initialize(ServletHolder.java:427)
        at org.eclipse.jetty.servlet.ServletHandler.initialize(ServletHandler.java:760)
        at org.springframework.boot.web.embedded.jetty.JettyEmbeddedWebAppContext$JettyEmbeddedServletHandler.deferredInitialize(JettyEmbeddedWebAppContext.java:46)
        at org.springframework.boot.web.embedded.jetty.JettyEmbeddedWebAppContext.deferredInitialize(JettyEmbeddedWebAppContext.java:36)
        at org.springframework.boot.web.embedded.jetty.JettyWebServer.handleDeferredInitialize(JettyWebServer.java:209)
        at org.springframework.boot.web.embedded.jetty.JettyWebServer.start(JettyWebServer.java:141)
        at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.startWebServer(ServletWebServerApplicationContext.java:300)
        at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.finishRefresh(ServletWebServerApplicationContext.java:162)
        at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:553)
        at org.springframework.boot.web.servlet.context.ServletWebServerApplicationContext.refresh(ServletWebServerApplicationContext.java:140)
        at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:759)
        at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:395)
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:327)
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:1255)
        at org.springframework.boot.SpringApplication.run(SpringApplication.java:1243)
        at com.github.freeacs.stun.App.main(App.java:22)
Caused by: com.mysql.cj.exceptions.CJCommunicationsException: Communications link failure

[... cut ...]
2018-10-09 18:26:57 [ActiveDeviceDetection] INFO  c.g.f.stun.ActiveDeviceDetection - ActiveDeviceDection: Have removed 0 devices from active devices map
2018-10-09 18:26:57 [ActiveDeviceDetection] INFO  c.g.f.stun.ActiveDeviceDetection - ActiveDeviceDetection: Have removed 0 devices from sentSyslog map - should be approx 1/60 of activeDevices.size() (0)
2018-10-09 18:26:57 [ActiveDeviceDetection] INFO  c.g.f.stun.ActiveDeviceDetection - ActiveDeviceDetection: Processed 0 active device syslog messages to the syslog server. Sent 0 syslog messages. SentSyslogMap.size() = 0, ActiveDevices.size() = 0
```

[image]

jarlah, 2018-10-09:
yey :) problem solved. . its a little bit flaky i can give you that. But it works at least ;) In a production environment i would rather use normal .deb installs, but its very cool that you use the docker version :+1: Closing for now, but if you feel something more is troublesome just open an issue and we can discuss it there.

closed 2018-10-09 by jarlah

UnAfraid, 2018-10-09:
What if you add healthcheck and make sure mysql is ready before other services are even started?

UnAfraid, 2018-10-09:
Something like
```yaml
services:
  db:
    image: freeacs/mysql
    command: --max_allowed_packet=32M
    restart: always
    mem_limit: 1000000000
    environment:
      - TZ
      - MYSQL_ROOT_PASSWORD=root
    ports:
      - 3306:3306
    volumes:
      - mysql-data:/var/lib/mysql
  healthcheck:
    test: "/usr/bin/mysqladmin -u root -proot ping"
    interval: 10s
    timeout: 10s
    retries: 3
    networks:
      mynetwork:
        aliases:
          - db.main
          - db.syslog

  stun:
    image: freeacs/freeacs-stun
    restart: always
    mem_limit: 2048000000
    depends_on:
      db:
        condition: service_healthy
    networks:
      - mynetwork
    environment:
      - SPRING_PROFILES_ACTIVE
      - MAIN_DATASOURCE_JDBCURL
      - MAIN_DATASOURCE_USERNAME
      - MAIN_DATASOURCE_PASSWORD
      - SYSLOG_SERVER_HOST
      - SERVER_PORT
      - TZ
    volumes:
      - stun_logs:/opt/docker/logs
#     - ./conf/stun/application-config.properties:/opt/docker/config/application-config.properties
#     - ./conf/stun/logback.xml:/opt/docker/config/logback.xml
```