freeacs/freeacs #191: Failed to execute operation: No such file or directory

penaivanalejandro, 2018-10-09:
This error its a new fresh installation using the command:

wget https://raw.githubusercontent.com/freeacs/freeacs/master/scripts/install.sh -v -O install.sh && chmod +x install.sh && sudo ./install.sh;

This happen on Ubuntu 16.04

```LOG
core installation start
Failed to execute operation: No such file or directory
Selecting previously unselected package freeacs-core.
(Reading database ... 93800 files and directories currently installed.)
Preparing to unpack freeacs-core_2.0.2_all.deb ...
Unpacking freeacs-core (2.0.2) ...
Setting up freeacs-core (2.0.2) ...
Creating system group: freeacs
Creating system user: freeacs in freeacs with freeacs-core daemon-user and shell /bin/false
Created symlink from /etc/systemd/system/multi-user.target.wants/freeacs-core.service to /lib/systemd/system/freeacs-core.service.
core installation complete
stun installation start
Failed to execute operation: No such file or directory
Selecting previously unselected package freeacs-stun.
(Reading database ... 93938 files and directories currently installed.)
Preparing to unpack freeacs-stun_2.0.2_all.deb ...
Unpacking freeacs-stun (2.0.2) ...
Setting up freeacs-stun (2.0.2) ...
Created symlink from /etc/systemd/system/multi-user.target.wants/freeacs-stun.service to /lib/systemd/system/freeacs-stun.service.
stun installation complete
tr069 installation start
Failed to execute operation: No such file or directory
Selecting previously unselected package freeacs-tr069.
(Reading database ... 94072 files and directories currently installed.)
Preparing to unpack freeacs-tr069_2.0.2_all.deb ...
Unpacking freeacs-tr069 (2.0.2) ...
Setting up freeacs-tr069 (2.0.2) ...
Created symlink from /etc/systemd/system/multi-user.target.wants/freeacs-tr069.service to /lib/systemd/system/freeacs-tr069.service.
tr069 installation complete
syslog installation start
Failed to execute operation: No such file or directory
Selecting previously unselected package freeacs-syslog.
(Reading database ... 94207 files and directories currently installed.)
Preparing to unpack freeacs-syslog_2.0.2_all.deb ...
Unpacking freeacs-syslog (2.0.2) ...
Setting up freeacs-syslog (2.0.2) ...
Created symlink from /etc/systemd/system/multi-user.target.wants/freeacs-syslog.service to /lib/systemd/system/freeacs-syslog.service.
syslog installation complete
web installation start
Failed to execute operation: No such file or directory
Selecting previously unselected package freeacs-web.
(Reading database ... 94341 files and directories currently installed.)
Preparing to unpack freeacs-web_2.0.2_all.deb ...
Unpacking freeacs-web (2.0.2) ...
Selecting previously unselected package freeacs-webservice.
Preparing to unpack freeacs-webservice_2.0.2_all.deb ...
Unpacking freeacs-webservice (2.0.2) ...
Setting up freeacs-web (2.0.2) ...
Created symlink from /etc/systemd/system/multi-user.target.wants/freeacs-web.service to /lib/systemd/system/freeacs-web.service.
Setting up freeacs-webservice (2.0.2) ...
Created symlink from /etc/systemd/system/multi-user.target.wants/freeacs-webservice.service to /lib/systemd/system/freeacs-webservice.service.
web installation complete
webservice installation start
Removed symlink /etc/systemd/system/multi-user.target.wants/freeacs-webservice.service.
(Reading database ... 94623 files and directories currently installed.)
Preparing to unpack freeacs-webservice_2.0.2_all.deb ...
Unpacking freeacs-webservice (2.0.2) over (2.0.2) ...
Setting up freeacs-webservice (2.0.2) ...
Created symlink from /etc/systemd/system/multi-user.target.wants/freeacs-webservice.service to /lib/systemd/system/freeacs-webservice.service.
webservice installation complete
shell installation start
Failed to execute operation: No such file or directory
Selecting previously unselected package freeacs-shell.
(Reading database ... 94623 files and directories currently installed.)
Preparing to unpack freeacs-shell_2.0.2_all.deb ...
Unpacking freeacs-shell (2.0.2) ...
Setting up freeacs-shell (2.0.2) ...
Failed to restart freeacs-shell.service: Unit freeacs-shell.service not found.
shell installation complete
monitor installation start
Failed to execute operation: No such file or directory
Selecting previously unselected package freeacs-monitor.
(Reading database ... 94688 files and directories currently installed.)
Preparing to unpack freeacs-monitor_2.0.2_all.deb ...
Unpacking freeacs-monitor (2.0.2) ...
Setting up freeacs-monitor (2.0.2) ...
Created symlink from /etc/systemd/system/multi-user.target.wants/freeacs-monitor.service to /lib/systemd/system/freeacs-monitor.service.
monitor installation complete
dpkg-query: no packages found matching nginx
Reading package lists... Done
Building dependency tree       
Reading state information... Done
```

jarlah, 2018-10-09:
Is something failing? It just means that the service is not already installed. Its trying to disable a service that does not exist

penaivanalejandro, 2018-10-09:
Good day @jarlah 

Nothing fails, but the message caught my attention and that is why I post it.

Thanks for the clarification.

closed 2018-10-09 by jarlah