freeacs/freeacs #195: Parameters can not be read (READALL)

penaivanalejandro, 2018-10-11:
What about @jarlah , I'm testing the version that is available at this moment and now the CPE that could read everything before (READALL), now can not be read by FreeACS..

But now with this version it only reads the basics but in the section of the Unit Dashboard it shows the status in 0.0, where before if it showed it in 10.0 with these CPEs.

This is the version that I have installed:

[image]

[image]

This is what can be read only in the web interface:

[image]

[image]

This is what the LOG show in the file tr069.log

```LOG
2018-10-11 10:06:31 [qtp1414934182-23] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node0gt4u8z65o1x7gmcprgp5qtug301 did not return a SessionData object from cache, must create a new SessionData object
2018-10-11 10:06:31 [qtp1414934182-23] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node0gt4u8z65o1x7gmcprgp5qtug301 created: 1539270391967, lastAccess:1539270391967, mxInactiveInterval:1800
2018-10-11 10:06:31 [qtp1414934182-23] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-10-11 10:06:31 [qtp1414934182-23] INFO  c.g.f.base.http.DigestAuthenticator - Send challenge to CPE, located on IP-address 127.0.0.1
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG c.g.f.base.http.DigestAuthenticator - Digest verification of CPE starts, located on IP-address 0:0:0:0:0:0:0:1
2018-10-11 10:06:32 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache.Cleanup.run() starts
2018-10-11 10:06:32 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 0 elements out of 2 in 0 ms.
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG c.g.f.base.http.DigestAuthenticator - Digest verification identifed unit id ADTN1000002F from CPE IP-address 0:0:0:0:0:0:0:1
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG com.github.freeacs.tr069.SessionData - Will load unit data
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0.35 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, up.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = 'ADTN1000002F' 
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG c.g.freeacs.base.db.DBAccessSession - Found unit ADTN1000002F, unittype 434RG, profile DefaultV4
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG com.github.freeacs.tr069.SessionData - Found unit in database - in total 7 params (7 unit params, 0 profile params (0 overridden), 0 always read params added)
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG com.github.freeacs.tr069.SessionData - Removed 7 system parameter from param-list, now contains 0 params
2018-10-11 10:06:32 [qtp1414934182-29] INFO  c.g.f.base.http.DigestAuthenticator - Authentication verified (CPE IP address: 0:0:0:0:0:0:0:1)
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG com.github.freeacs.common.util.Cache - Key 0:0:0:0:0:0:0:1"ADTN1000002F", realm will be removed from Cache (explicitely - not by Cache.Cleanup.run())
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG c.g.f.t.methods.HTTPRequestProcessor - Will process method Inform (incoming request/response from CPE)
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG c.github.freeacs.tr069.methods.INreq - Parsed INreq params, found keyroot:InternetGatewayDevice., parameterkey:(null), swver:J73.00.030.M4
2018-10-11 10:06:32 [qtp1414934182-29] DEBUG c.github.freeacs.tr069.ParameterKey - No device parameter values found, ACS parameterkey = "No data in DB"
2018-10-11 10:06:32 [qtp1414934182-23] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-10-11 10:06:32 [qtp1414934182-23] DEBUG c.g.f.t.methods.HTTPRequestProcessor - Will process method Empty (incoming request/response from CPE)
2018-10-11 10:06:32 [qtp1414934182-23] DEBUG com.github.freeacs.tr069.SessionData - DISCOVER parameter not found of value is null in discoverUnittype() 
2018-10-11 10:06:32 [qtp1414934182-23] INFO  c.g.freeacs.tr069.methods.EMDecision - EM-Decision is GPV since everything is normal and previous method was either IN or TC (updating LCT and possibly FCT)
2018-10-11 10:06:32 [qtp1414934182-23] DEBUG c.g.f.t.methods.HTTPResponseCreator - VendorConfigFile object will be requested (default behavior)
2018-10-11 10:06:32 [qtp1414934182-23] DEBUG c.g.f.t.methods.HTTPResponseCreator - 1 cpe-param (not found in database, but of special interest to ACS) added to the GPV-request
2018-10-11 10:06:32 [qtp1414934182-23] DEBUG c.g.f.t.methods.HTTPResponseCreator - Asks for 1 parameters in GPV-req
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG c.g.f.t.methods.HTTPRequestProcessor - Will process method GetParameterValues (incoming request/response from CPE)
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG c.g.freeacs.tr069.methods.GPVres - Will process XML: 702 char
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG c.g.freeacs.tr069.methods.GPVres - Response holds 0 parameters
2018-10-11 10:06:32 [qtp1414934182-26] WARN  c.g.freeacs.tr069.methods.GPVres - Number of parameters returned from CPE is less than asked for (1)
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG c.g.freeacs.tr069.methods.GPVres - Found 0 cpe-params (of special interest to ACS) in response
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG c.g.f.tr069.methods.GPVDecision - Mode was detected to be: REGULAR
2018-10-11 10:06:32 [qtp1414934182-26] ERROR c.g.f.tr069.methods.GPVDecision - The unittype does not contain PeriodicInformInterval, terminating the conversation.
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG c.g.f.tr069.methods.GPVDecision - GPV-Decision is Empty
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG c.github.freeacs.tr069.Provisioning - End of session is reached, will write queued unit parameters if unit (ADTN1000002F) is not null
2018-10-11 10:06:32 [qtp1414934182-26] INFO  com.github.freeacs.dbi.ACSUnit - Updated unit parameter System.X_FREEACS-COM.LastConnectTms with value 2018-10-11 10:06:32
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG com.github.freeacs.common.util.Cache - Key ADTN1000002FSESSION will be removed from Cache (explicitely - not by Cache.Cleanup.run())
2018-10-11 10:06:32 [qtp1414934182-26] DEBUG com.github.freeacs.common.util.Cache - Key node0gt4u8z65o1x7gmcprgp5qtug301SESSION will be removed from Cache (explicitely - not by Cache.Cleanup.run())
```

And the conversation show this:

```XML
2018-10-11 10:06:32 [qtp1414934182-29] INFO  Conversation - ============== FROM CPE ( ADTN1000002F ) TO ACS ===============
<SOAP-ENV:Envelope
    xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/"
    xmlns:SOAP-ENC="http://schemas.xmlsoap.org/soap/encoding/"
    xmlns:xsd="http://www.w3.org/2001/XMLSchema"
    xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
    xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
[... 289 lines cut ...]
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
	<soapenv:Header>
		<cwmp:ID soapenv:mustUnderstand="1">bogofabricated</cwmp:ID>
	</soapenv:Header>
	<soapenv:Body>
		<cwmp:InformResponse>
			<MaxEnvelopes>1</MaxEnvelopes>
		</cwmp:InformResponse>
	</soapenv:Body>
</soapenv:Envelope>
[      805.188] 2018-10-10T15:26:05-05 cwmpclient      acs_session.py(527): RUSAGE after POST to ACS: max RSS used=15440kB
[      805.276] 2018-10-10T15:26:05-05 cwmpclient      acs_session.py(618):
A XML parse error occured processing ACS HTTP response: Traceback (most recent call last):
  File "source/acs_session.py", line 580, in session_loop
  File "source/acs_session.py", line 696, in parse_acs_response
  File "source/cwmpmsgread.py", line 268, in read_incoming
  File "source/soapread.py", line 134, in read_incoming
CwmpParseError: First child element of SOAP body is unrecognized: urn:dslforum-org:cwmp-1-0:InformResponse. Is there a cwmp version namespace mismatch?
[      805.332] 2018-10-10T15:26:05-05 cwmpclient      acs_session.py(637): Session is over due to critical fault
[      805.335] 2018-10-10T15:26:05-05 cwmpclient      acs_session.py(271): Closing HTTP connection
[      805.338] 2018-10-10T15:26:05-05 cwmpclient      cwmp_session_handler.py(302): RUSAGE: after session.run() max RSS used=15440kB
[      805.341] 2018-10-10T15:26:05-05 cwmpclient      cwmp_session_handler.py(303): RUSAGE: after session.close() max RSS
```

I was using the different quirks that it brings in the properties of the tr069 app and I can not make it work.

jarlah, 2018-10-11:
Unit type is missing PeriodicInformInterval

jarlah, 2018-10-11:
Ok. So there might be a bug with how the code handles keyroot and retrieval of parameter values. About docker see the other issue for reply.

penaivanalejandro, 2018-10-11:
Yes, that are the step I use....

In the point number 3 (Discovery) I use the parameter:
System.X_FREEACS-COM.Device.Discover    

Enable whit value number 1

(If you want i can setup a CPE whit a URL/user/password do you give to my)

Thanks Sr.

jarlah, 2018-10-12:
It is reproducable. And i found an issue. A change in properties have creeped into tr069 on master. Discovery mode is on by default. Only affects docker installations

jarlah, 2018-10-12:
Seems easy to fix. And make unit test for. Not sure when it stopped work but i guess its easy to fix

jarlah, 2018-10-12:
I get exactly same bug as you do. It halts atter GPV due to missing Periodic inform interval param

jarlah, 2018-10-12:
waiting for this to build green, and then it will be merged. Then you can test if the newer device works.
For the other issue, in the mean time, please add manually a InternetGatewayDevice.ManagementServer.PeriodicInformInterval by using shell. you can download shell and run it locally on the server where you run docker for ex because mysql container exposes its port to the host system. please read the manual for how to use shell.

jarlah, 2018-10-12:
https://github.com/freeacs/freeacs/pull/198

jarlah, 2018-10-12:
if older devices will fail due to this increaed cwmp version change, then i must think more about this.

jarlah, 2018-10-12:
@penaivanalejandro test old units as well as new units, because i am not sure how old units will handle/respect the newer cwmp version number

jarlah, 2018-10-12:
@penaivanalejandro you might need to add a Device.ManagementServer.PeriodicInformInterval` for the new unit also. use shell as noted above. Without discovery, creation of unit type parameters can only be done in shell.

jarlah, 2018-10-12:
if i remember shell correctly command for changing context to a unit type is `cc ut:[Unit Type Name]`

jarlah, 2018-10-12:
then you can use 	`setparam <unittype-parameter-name> [D][A][C][S][I][M](R|RW|X)` as informed about when running help

jarlah, 2018-10-12:
so:
```
/>cc ut:Test
cc ut:Test
/ut:Test/>setparam InternetGatewayDevice.ManagementServer.PeriodicInformInterval RW
setparam InternetGatewayDevice.ManagementServer.PeriodicInformInterval RW
[1] The unit type parameter InternetGatewayDevice.ManagementServer.PeriodicInformInterval is added.
/ut:Test/>
```
for the new unit which does not use IGD keyroot change the above parameter to be prefixed with Device.

penaivanalejandro, 2018-10-12:
I put this in the shell:

```LOG
root@controller:/opt/freeacs-shell/bin# freeacs-shell
12:39:33,089 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback-test.xml]
12:39:33,089 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback.groovy]
12:39:33,089 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Found resource [logback.xml] at [jar:file:/opt/freeacs-shell/lib/com.github.freeacs.freeacs-shell-2.0.2.jar!/logback.xml]
12:39:33,098 |-INFO in ch.qos.logback.core.joran.spi.ConfigurationWatchList@7c75222b - URL [jar:file:/opt/freeacs-shell/lib/com.github.freeacs.freeacs-shell-2.0.2.jar!/logback.xml] is not of type file
12:39:33,127 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - debug attribute not set
12:39:33,127 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.ConsoleAppender]
12:39:33,131 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [STDOUT]
12:39:33,135 |-INFO in ch.qos.logback.core.joran.action.NestedComplexPropertyIA - Assuming default type [ch.qos.logback.classic.encoder.PatternLayoutEncoder] for [encoder] property
12:39:33,166 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.rolling.RollingFileAppender]
12:39:33,168 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [FILE]
12:39:33,174 |-INFO in c.q.l.core.rolling.TimeBasedRollingPolicy@1277181601 - No compression will be used
12:39:33,175 |-INFO in c.q.l.core.rolling.TimeBasedRollingPolicy@1277181601 - Will use the pattern tr069_%d{yyyy-MM-dd}.log for the active file
12:39:33,178 |-INFO in c.q.l.core.rolling.DefaultTimeBasedFileNamingAndTriggeringPolicy - The date pattern is 'yyyy-MM-dd' from file name pattern 'tr069_%d{yyyy-MM-dd}.log'.
12:39:33,178 |-INFO in c.q.l.core.rolling.DefaultTimeBasedFileNamingAndTriggeringPolicy - Roll-over at midnight.
12:39:33,180 |-INFO in c.q.l.core.rolling.DefaultTimeBasedFileNamingAndTriggeringPolicy - Setting initial period to Fri Oct 12 12:04:57 CDT 2018
12:39:33,181 |-INFO in ch.qos.logback.core.joran.action.NestedComplexPropertyIA - Assuming default type [ch.qos.logback.classic.encoder.PatternLayoutEncoder] for [encoder] property
12:39:33,183 |-INFO in ch.qos.logback.core.rolling.RollingFileAppender[FILE] - Active log file name: logs/shell.log
12:39:33,183 |-INFO in ch.qos.logback.core.rolling.RollingFileAppender[FILE] - File property is set to [logs/shell.log]
12:39:33,184 |-INFO in ch.qos.logback.classic.joran.action.LoggerAction - Setting level of logger [com.zaxxer.hikari] to WARN
12:39:33,184 |-INFO in ch.qos.logback.classic.joran.action.LoggerAction - Setting level of logger [com.github.freeacs] to DEBUG
12:39:33,184 |-INFO in ch.qos.logback.classic.joran.action.RootLoggerAction - Setting level of ROOT logger to WARN
12:39:33,184 |-ERROR in ch.qos.logback.core.joran.action.AppenderRefAction - Could not find an appender named [ASYNC_FILE]. Did you define it below instead of above in the configuration file?
12:39:33,184 |-ERROR in ch.qos.logback.core.joran.action.AppenderRefAction - See http://logback.qos.ch/codes.html#appender_order for more details.
12:39:33,184 |-ERROR in ch.qos.logback.core.joran.action.AppenderRefAction - Could not find an appender named [ASYNC_STDOUT]. Did you define it below instead of above in the configuration file?
12:39:33,184 |-ERROR in ch.qos.logback.core.joran.action.AppenderRefAction - See http://logback.qos.ch/codes.html#appender_order for more details.
12:39:33,184 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - End of configuration.
12:39:33,185 |-INFO in ch.qos.logback.classic.joran.JoranConfigurator@27f674d - Registering current configuration as safe fallback point

Loading class `com.mysql.jdbc.Driver'. This is deprecated. The new driver class is `com.mysql.cj.jdbc.Driver'. The driver is automatically registered via the SPI and manual loading of the driver class is generally unnecessary.
=== FUSION LOGIN === 
You may skip login to become Admin, just press ENTER
Username: />
  ____ _  _ ____ _ ____ _  _    ____ _  _ ____ _    _    
  |___ |  | [__  | |  | |\ |    [__  |__| |___ |    |
  |    |__| ___] | |__| | \|    ___] |  | |___ |___ |___ v2.3.41

  Fusion Shell provdes a variety of commands and possibilities for the advanced 
  user of Fusion. It is possible to make scripts to automate tedious tasks, and 
  the scripts can be made fairly advanced by the use of file input/output. 
  Furthermore it is possible to combine several scripts so the user can build 
  script libraries for optimal reuse. To learn all about Fusion Shell you need 
  to read Fusion Shell User Manual, but 4 commands is a must at this point:

  - Type 'help' to get context-sensitive help.
  - Type 'help generic' to get generic help.
  - Type 'help <name-of-command>' to get detailed command help.
  - Type 'exit' to exit.

  By using help the user should be able to learn most of the commands in Fusion 
  Shell, which will enable him to create/update/delete/inspect all parts of 
  the Fusion database. The argument in the 'help' command can be shortened as 
  long as it's not ambigious.

/>cc 434RG
Error: The argument  could not be parsed into a context switch
/>cc ut:434RG
/ut:434RG/>setparam InternetGatewayDevice.ManagementServer.PeriodicInformInterval RW
[... cut ...]
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
	<cwmp:ID soapenv:mustUnderstand="1">FREEACS-1539366297197</cwmp:ID>
</soapenv:Header>
<soapenv:Body>
		<cwmp:GetParameterValues>
			<ParameterNames soapenc:arrayType="xsd:string[1]">
				<string>InternetGatewayDevice.</string>
			</ParameterNames>
		</cwmp:GetParameterValues>
</soapenv:Body>
</soapenv:Envelope>
```

penaivanalejandro, 2018-10-12:
Hi...  good news...  i will change the parameter "READALL" to "REGULAR" and now i can see the equipment online i get the basic values:

[image]

Dashboard Unit:

[image]

Now how can i ReadAll?

jarlah, 2018-10-12:
well it came a bit further, it didn't fail due to missing PeriodicInformInterval param. I will test this with my own china unit. It seems its sending cwmp-1-0 from the device and thats how it roll with that device. So its not the same problem as with the newer cpe that reported cwmp-1-2 and failed because acs didnt comply to that. 

Btw, i have added AutonomousTransferComplete in master now, its building and is soon available on docker for pulling.

jarlah, 2018-10-12:
nice :+1:

jarlah, 2018-10-12:
oh and btw newer version of acs in docker will now use cwmp-1-2, Test that against the new unit

jarlah, 2018-10-12:
my gut feeling is that older units might get panic .. if so i need to check what unit reports in inform ... if its cwmp-1-0 or cwmp-1-2 and use that in the server responses ...

penaivanalejandro, 2018-10-12:
I understand that the Free ACS is 1.0 because of the XML that is sent from the Server.

I have a CPE that sends in its XML that is version 1.2 and I have not been able to read the parameters, nor basic (REGULAR).

This is LOG:

```LOG
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding to HTTPReqResData object to the list
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG c.g.f.base.http.DigestAuthenticator - Digest verification of CPE starts, located on IP-address 127.0.0.1
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG c.g.f.base.http.DigestAuthenticator - Digest verification identifed unit id IPHO0050A109 from CPE IP address 127.0.0.1
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG com.github.freeacs.tr069.SessionData - Will load unit data
2018-10-12 13:17:05 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache.Cleanup.run () starts
2018-10-12 13:17:05 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 0 elements out of 2 in 0 ms.
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG cgfdbi.UnitQueryCrossUnittype - [0.28 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, up.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = 'IPHO0050A109'
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG c.g.freeacs.base.db.DBAccessSession - Found unit IPHO0050A109, unittype 7259G, profile Default7259G
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG com.github.freeacs.tr069.SessionData - Found unit in database - in total 3 params (3 unit params, 0 profile params (0 overridden), 0 always read params added)
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG com.github.freeacs.tr069.SessionData - Removed 2 system parameter from param-list, now contains 1 params
2018-10-12 13:17:05 [qtp1232703108-22] INFO c.g.f.base.http.DigestAuthenticator - Authentication verified (CPE IP address: 127.0.0.1)
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG com.github.freeacs.common.util.Cache - Key 127.0.0.1 "IPHO0050A109", realm will be removed from Cache (explicitely - not by Cache.Cleanup .run ())
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG c.g.f.t.methods.HTTPRequestProcessor - Will process method Inform (incoming request / response from CPE)
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG c.github.freeacs.tr069.methods.INreq - Parsed INreq params, found keyroot: Device., Parameterkey :, swver: C5.6.0.39
2018-10-12 13:17:05 [qtp1232703108-22] DEBUG c.github.freeacs.tr069.ParameterKey - No device parameter values ​​found, ACS parameterkey = "No data in DB"
```

And the conversation is as follows:

```XML
2018-10-12 13:17:05 [qtp1232703108-22] INFO  Conversation - ============== FROM CPE ( IPHO0050A109 ) TO ACS ===============
<soapenv:Envelope xmlns:cwmp="urn:dslforum-org:cwmp-1-2" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"><soapenv:Header><cwmp:ID soapenv:mustUnderstand="1">bogofabricated</cwmp:ID></soapenv:Header><soapenv:Body><cwmp:Inform><DeviceId><Manufacturer>IPHOTONIX</Manufacturer><OUI>001C15</OUI><ProductClass>FTTP</ProductClass><SerialNumber>IPHO0050A109</SerialNumber></DeviceId><Event soapenc:arrayType="cwmp:EventStruct[03]"><EventStruct><EventCode>4 VALUE CHANGE</EventCode><CommandKey /></EventStruct><EventStruct><EventCode>2 PERIODIC</EventCode><CommandKey /></EventStruct><EventStruct><EventCode>10 AUTONOMOUS TRANSFER COMPLETE</EventCode><CommandKey /></EventStruct></Event><MaxEnvelopes>2</MaxEnvelopes><CurrentTime>2018-10-12T18:17:05+00:00</CurrentTime><RetryCount>0</RetryCount><ParameterList soapenc:arrayType="cwmp:ParameterValueStruct[13]"><ParameterValueStruct><Name>Device.RootDataModelVersion</Name><Value xsi:type="xsd:string">2.4</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.HardwareVersion</Name><Value xsi:type="xsd:string">-330-7259-013-A04</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.SoftwareVersion</Name><Value xsi:type="xsd:string">C5.6.0.39</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ProvisioningCode</Name><Value xsi:type="xsd:string" /></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ParameterKey</Name><Value xsi:type="xsd:string" /></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ConnectionRequestURL</Name><Value xsi:type="xsd:string">http://196.196.8.180:7547/cwmp/</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.AliasBasedAddressing</Name><Value xsi:type="xsd:boolean">0</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ConnectionRequestURL</Name><Value xsi:type="xsd:string">http://196.196.8.180:7547/cwmp/</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.URL</Name><Value xsi:type="xsd:string">http://201.172.206.16/tr069</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.Time.Status</Name><Value xsi:type="xsd:string">Synchronized</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.Routing.Router.1.IPv4ForwardingNumberOfEntries</Name><Value xsi:type="xsd:unsignedInt">1</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.IP.Interface.1.Status</Name><Value xsi:type="xsd:string">Up</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.IP.Interface.1.IPv4Add [... cut ...]
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
        <cwmp:ID soapenv:mustUnderstand="1">bogofabricated</cwmp:ID>
</soapenv:Header>
<soapenv:Body>
                <cwmp:InformResponse>
                        <MaxEnvelopes>1</MaxEnvelopes>
                </cwmp:InformResponse>
</soapenv:Body>
</soapenv:Envelope>
```

jarlah, 2018-10-12:
Just tested with my own china unit which i discovered is actually also 1-2. and it has previously worked with the old version of freeacs that used 1-0 (2.0.2). Master is currently bleeding edge. Soon the be released ;) Lots of changes there....

So, the change in master works with my CPE. But someone must verify that cpe that use 1-0 works.

penaivanalejandro, 2018-10-12:
> Just tested with my own china unit which i discovered is actually also 1-2. and it has previously worked with the old version of freeacs that used 1-0 (2.0.2). Master is currently bleeding edge. [...]

So I understand you made adjustments, do I run the install.sh script again? so that it is updated with the master version.

jarlah, 2018-10-12:
i have made adjustments with the rolling update docker version. I will soon make a new release of deb files, but first i need to know if it works. I dont know if you can test in docker, but if you can it would be wonderful ;)

penaivanalejandro, 2018-10-12:
Excellent...  

I been try with Docker....!  And confirm

Greetings

jarlah, 2018-10-12:
With units reporting cwmp-1-0? if so :+1:

I will let master get a good night sleep and maybe release something tomorrow evening, or sunday. Or some other day.

Its a lof of changes. Tr069 has been trimmed down to only 9MB in size ... monitor also. Two test rabbits for a more lightweight server technology.

penaivanalejandro, 2018-10-12:
Hi @jarlah ,

I confirm it's working with the parameters:

Secret
Provisioning  = REGULAR

and manually I add the parameter:

PeriodInformInterval = RW

Both version of cwmp  (1.0 and 1.2) but that's all i can do read REGULAR...
I can't readall like other version FreeACS (2.0)

This test was in docker
Greetings.

jarlah, 2018-10-13:
I am so tired of the message passing between stun server and web only to send a simple http request to the device. I mean whats the point. This will soon be fixed. Kick Jobs might still be triggered by stun server. But web will depend on stun and use its kick implementation. And we shouldnt have to wait. We kick and get response and rerender the Page. To see if kick was succes reload Page. Simple

jarlah, 2018-10-13:
Actually. You will get immediate response when kicking. But device might not connect immediately

jarlah, 2018-10-13:
will not change how kick works, but will port stun server to sparkjava in  #199

jarlah, 2018-10-14:
this has been verified in docker

jarlah, 2018-10-14:
I have verified that READALL works. But to kick the unit, when the unit does not report the connection request url, you must manually set some parameters. See images below.  I was able to click Read all button in web and then automatically get the parameters in one go ;)

[image]

[image]

jarlah, 2018-10-14:
1. PublicPort must be set
2. PublicProtocol must be set (http or https) .. i COULD have made a default here ...
2. ConnectionRequestUsername must be set (i set it manually because stupid china unit does not report it)

PublicIp was already set. And luckily correct.

jarlah, 2018-10-14:
it defaults to using the username as password when no ConnectionRequestPassword is set. I have some troubles with my china unit where when i set the password it saves it with space inside the device, bricking the device. So i use the defaulting to username as password in this case.

jarlah, 2018-10-14:
New release. see https://github.com/freeacs/freeacs/releases/tag/v2.0.3
Tested in docker and ubuntu.
To get READALL working, i suppose follow my guide further up, because it works for me.
If there are unique cpe bugs, the only way to debug those problems is to check logs in the CPE to see why it failed to accept response from acs .. i know that you in some cases have units which never reports back. Some units doesnt report back due to wrong cwmp version, but now thats fixed.

penaivanalejandro, 2018-10-15:
Hello @Jarl, good morning ...

I'm going to start trying this version 2.0.3, any detail I mention.

regards

penaivanalejandro, 2018-10-15:
I was testing today with 2 different CPEs (versions 1-0 and 1-2).

I perform the following steps:

1.- Create the type of unit.
2.- Create a different profile than Default.
3.- Create the Unit ID based on the serial number of the equipment.

(Up to this point we remain the same).

I had always tried first with the parameters that FreeACS brings:

- System.X_FREEACS-COM.Secret (password123)
- System.X_FREEACS-COM.ProvisioningMode (READALL)

Comments: When the CPE was connected, the conversation with the FreeACS started and the latter asked him about all the parameters and their values.

In case the previous configuration does not work, it used the following configuration:

- System.X_FREEACS-COM.Discover (1)
- System.X_FREEACS-COM.Secret (password123)
- System.X_FREEACS-COM.ProvisioningMode (READALL)

Now with this version 2.0.3 I only put the following configuration:

- System.X_FREEACS-COM.Discover (1)
- System.X_FREEACS-COM.Secret (password123)

And I can obtain all the parameters of the CPE (1-0 and 1-2), when obtaining the values, an interval time of more than 8000 seconds estimated in the CPE is fixed, to then enable the parameters:

InternetGatewayDevice.ManagementServer.ConnectionRequestURL (R)
InternetGatewayDevice.ManagementServer.ConnectionRequestPassword (RW)
InternetGatewayDevice.ManagementServer.ConnectionRequestUsername (RW)

In the parameters (RW) he wrote user and password values. To then restart the CPE and re-read the URL parameters and set the user / password.

With this I can now click on the "Read All" button via web and get all the CPE values ​​as well.

This works, since only with obtaining all the parameters of the CPE once the required ones are enabled.

Now my intention is to test this via webservice, to be able to obtain the values ​​of the enabled parameters and see if there is any way to use STUN via webservice to be able to set some value and this is sent using the webservice instead of the web page.

This issue can now be closed, if you agree ?.

jarlah, 2018-10-16:
Agree I think. Closing issue. But if its not resolved completely you are free to open New issue.

closed 2018-10-16 by jarlah