freeacs/freeacs #203: Can you contribute to this project or test it?

jarlah, 2018-10-15:
Hi all

I am doing this because i am one of the original developers and because i like software development. I am doing some professional paid support on it, but its basically mostly done in my spare time.

To continue this project, to drive it forward, we need more contributors. I have been trying to make this project more maintainable and more understandable for the purpose of attracting **you** ;)

Will you help improve the future of secure provisioning server technology?  Do you know a language like python or c/c++, and want to learn a little bit Java?`Join me in this project and we will make it better. 

Cheers

penaivanalejandro, 2018-10-15:
I can contribute very well in tests ...!
In the subject of programming I know the basics, I have tried to read and modify the code you have but I am honest I do not see myself capable.

Greetings.

jarlah, 2018-10-22:
@kangdengfeng definitely no c/c++ knowledge here besides personal adventures. Basically Java 8 here. If you know some low level or high level language, Java 8 will not be a big hurdle ;)

kangdengfeng, 2019-01-16:
> @kangdengfeng definitely no c/c++ knowledge here besides personal adventures. Basically Java 8 here. If you know some low level or high level language, Java 8 will not be a big hurdle ;)

Actually i am used to be an software engineer use C#, and now is an embedded software engineer use c/c++... actually  just know basic Java  grammar and little frame. But i am very interested in this, and i am following this project , will try my best to help

closed 2021-09-04 by jarlah