freeacs/freeacs #214: Add more unit tests in dbi (dao layer)

jarlah, 2018-10-22:
I have started to work on and will continue work on creating more tests. But we need more and better test coverage.

If you dont know much about the internals of freeacs nows the chance to do learn. Just checkout the project, comment here what you are going to test and fire away :)

jarlah, 2018-10-22:
The biggest hurdle is that you need to install an editor that supports sbt. But thats easy. Just install intellij community edition. Second Bugges hurdle, how do i write the tests? Look db/src/test/java :)

govi20, 2018-10-22:
If you are open for contribution, I would like to work on the second hurdle ;-) assign me this issue, if possible.

jarlah, 2018-10-22:
Nice  @govi20  :) just fire away ;) i am using sbt primarely btw, and you should either need to add an eclipse plugin in project/plugins.sbt if you are using eclipse, or install scala plugin in intellij (which gives you sbt support). Assigning you to it now, no stress. I see you are spring dev ;) only spring apps here are the old web module and the rewritten webservice module. For this case however it would be great to increase as said unit coverage in dbi module.

jarlah, 2018-10-22:
@govi20 i suggest you just start and I can add you to a team when you have contributed and merged your first issue :+1:

jarlah, 2018-10-23:
Look in contribution guide for how to contribute. In short you can make a new issue relating to the piece of code you are testing. And then open pull request referencing said issue

closed 2021-09-04 by jarlah