freeacs/freeacs #219: data model evolution: Refresh Unit-Type Parameters

lindererik, 2018-10-23:
Hello,

I would need to add Parameters to an already existing Unit-Type as there are new parameters for new SW version

Can I trigger re-discovery in some way ? Or do I have to delete the unit-type to rediscover ? (What would happen to exisiting units ?)

Thanks for any tip (I'm ready to hack the database in last resort  )

Cheers
erik

jarlah, 2018-10-23:
```
/**
   * The discovery flag, set to run GetParameterNames on device - update the unittype configuration.
   */
  public static final String DISCOVER = "System.X_FREEACS-COM.Discover" ;
```
Set it to 1.:)

jarlah, 2018-10-23:
At least thats the theory. I have no unit test for this scenario. Crossing fingers

jarlah, 2018-11-17:
No response in almost a month. Closing

closed 2018-11-17 by jarlah