freeacs/freeacs #265: Application level password failure

samgior10, 2018-11-26:
My credentials where created as a user, but I can't use them with the web GUI.
How can I access the web GUI and reset the password? FreeACS

jarlah, 2018-11-26:
Docker?

jarlah, 2018-11-26:
Tested on Ubuntu. Seems to work. In master and possibly docker there is now a little revolution going on in regards to default user and password. Thats why i Wonder if you are on bleeding edge or stable v2.1.0

jarlah, 2018-11-26:
i have now updated docker hub images. Please help me out on what platform you are using?

samgior10, 2018-11-28:
Hi Sorry for the delay response, ACS was loaded on the server and when I try and access the ACS web browser the password user created on the server is not recognised. "Bad Credentials"  I tried using admin/freeacs and a combination of uppercase and indicated in some blogs with no success.

jarlah, 2018-11-28:
ok. check in web.log. its in /opt/freeacs-web/logs/web.log

There is an auto generated password there. 

This feature is on its way out of the ocde. And the old behaviour is on its way back. with admin/freeacs.

New release coming today or tomorrow that removes the auto generated password in log feature.

Sorry for the troubles.

jarlah, 2018-11-28:
New release available. You can download and install the freeacs-web deb file manually with dpkg -i freeacs-web*.deb to get up and running again. To see more:

https://www.tapatalk.com/groups/freeacsforum/v2-1-2-t14.html#p66

closed 2018-11-30 by jarlah

samgior10, 2018-12-03:
Thanks for your prompt response, we haven't tested it yet.
I'll update you on the results.