freeacs/freeacs #272: ERROR c.github.freeacs.tr069.Provisioning - Something went wrong

penaivanalejandro, 2018-11-30:
Hi @jarlah 

Today I saw the need to reinstall the FreeACS from a new copy using the "install.sh"
I already had CPEs that authenticated with the tr069, at this time I still do not register them, so they can not be authenticated.
But when checking the LOG I see that it marks some errors, (you can confirm if it is normal to see these messages):
The log is from tr069.log

```LOG
2018-11-29 18:44:14 [qtp1101643590-713] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node02hnkh4oi5v5810asxst4470b207 did not return a SessionData object from cache, must create a new SessionData object
2018-11-29 18:44:14 [qtp1101643590-713] DEBUG c.g.freeacs.tr069.HTTPReqResData - Sessionid node02hnkh4oi5v5810asxst4470b207 created: 1543538654945, lastAccess:1543538654945, mxInactiveInterval:-1
2018-11-29 18:44:14 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache.Cleanup.run() starts
2018-11-29 18:44:14 [qtp1101643590-713] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-11-29 18:44:14 [qtp1101643590-713] INFO  c.g.f.base.http.BasicAuthenticator - Send challenge to CPE, located on IP-address 127.0.0.1
2018-11-29 18:44:14 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Key node01g32cl5iki80y12c4jciifaynv203SESSION with CacheValue com.github.freeacs.common.util.CacheValue@f577dae is to be removed
2018-11-29 18:44:14 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - CacheValue has a cleanupnotifier-class which will be executed
2018-11-29 18:44:14 [Cache Cleanup] ERROR c.g.f.base.SessionDataCacheCleanup - SessionData for null (key:node01g32cl5iki80y12c4jciifaynv203) was removed from cache after timeout - indicate a session which did not terminate correctly
2018-11-29 18:44:14 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 1 elements out of 2 in 1 ms.
2018-11-29 18:44:14 [qtp1101643590-704] DEBUG c.g.freeacs.tr069.HTTPReqResData - Adding a HTTPReqResData object to the list
2018-11-29 18:44:14 [qtp1101643590-704] DEBUG c.g.f.base.http.BasicAuthenticator - Basic verification of CPE starts, located on IP-address 0:0:0:0:0:0:0:1
2018-11-29 18:44:14 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache.Cleanup.run() starts
2018-11-29 18:44:14 [qtp1101643590-704] DEBUG c.g.f.base.http.DigestAuthenticator - Basic verification identifed unit id ADTN14170685 from CPE IP-address 0:0:0:0:0:0:0:1
2018-11-29 18:44:14 [qtp1101643590-704] DEBUG com.github.freeacs.tr069.SessionData - Will load unit data
2018-11-29 18:44:14 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Key 0:0:0:0:0:0:0:1Basic SVBITzAwNTBBMTA5OnBhc3N3b3JkMTIz with CacheValue com.github.freeacs.common.util.CacheValue@55847ce5 is to be removed
2018-11-29 18:44:14 [Cache Cleanup] DEBUG com.github.freeacs.common.util.Cache - Cache removed 1 elements out of 2 in 0 ms.
2018-11-29 18:44:14 [qtp1101643590-704] DEBUG c.g.f.dbi.UnitQueryCrossUnittype - [      0.39 ms] SELECT u.unit_id, u.profile_id, u.unit_type_id, up.unit_type_param_id, up.value FROM unit u LEFT JOIN unit_param up ON u.unit_id = up.unit_id WHERE u.unit_id = 'ADTN14170685'
2018-11-29 18:44:14 [qtp1101643590-704] WARN  com.github.freeacs.tr069.SessionData - Did not find unit in unit-table, nothing exists on this unit
2018-11-29 18:44:14 [qtp1101643590-704] ERROR c.github.freeacs.tr069.Provisioning - Authentication failed because unitid was not found (CPE IP address: 0:0:0:0:0:0:0:1) (username: ADTN14170685)
2018-11-29 18:44:14 [qtp1101643590-704] ERROR c.github.freeacs.tr069.Provisioning - Something went wrong
com.github.freeacs.tr069.exception.TR069AuthenticationException: Authentication failed because unitid was not found (CPE IP address: 0:0:0:0:0:0:0:1) (username: ADTN14170685)
        at com.github.freeacs.base.http.BasicAuthenticator.verify(BasicAuthenticator.java:145)
        at com.github.freeacs.base.http.BasicAuthenticator.authenticate(BasicAuthenticator.java:30)
        at com.github.freeacs.base.http.Authenticator.authenticate(Authenticator.java:98)
        at com.github.freeacs.tr069.Provisioning.doPost(Provisioning.java:178)
        at com.github.freeacs.common.spark.ResponseHelper.process(ResponseHelper.java:23)
        at com.github.freeacs.App.lambda$processRequest$3(App.java:117)
        at spark.RouteImpl$1.handle(RouteImpl.java:72)
        at spark.http.matching.Routes.execute(Routes.java:61)
        at spark.http.matching.MatcherFilter.doFilter(MatcherFilter.java:134)
        at spark.embeddedserver.jetty.JettyHandler.doHandle(JettyHandler.java:50)
        at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1568)
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:144)
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132)
        at org.eclipse.jetty.server.Server.handle(Server.java:503)
        at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:364)
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:260)
        at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:305)
        at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103)
        at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:118)
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:333)
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:310)
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:168)
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126)
        at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366)
        at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:765)
        at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:683)
        at java.lang.Thread.run(Thread.java:748)
Caused by: com.github.freeacs.base.NoDataAvailableException: Desired data is not available in the database
        at com.github.freeacs.tr069.SessionData.updateParametersFromDB(SessionData.java:159)
        at com.github.freeacs.base.http.BasicAuthenticator.verify(BasicAuthenticator.java:79)
        ... 26 common frames omitted
2018-11-29 18:44:14 [qtp1101643590-704] DEBUG c.github.freeacs.tr069.Provisioning - End of session is reached, will write queued unit parameters if unit (null) is not null
2018-11-29 18:44:14 [qtp1101643590-704] DEBUG com.github.freeacs.common.util.Cache - Key ADTN14170685SESSION will be removed from Cache (explicitely - not by Cache.Cleanup.run())
2018-11-29 18:44:14 [qtp1101643590-704] DEBUG com.github.freeacs.common.util.Cache - Key node02hnkh4oi5v5810asxst4470b207SESSION will be removed from Cache (explicitely - not by Cache.Cleanup.run())
```

jarlah, 2018-11-30:
Without staring at the log too much i can see that the unit did not exist in the db while trying to be authenticated. The something went wrong is just a try catch thing. It could have said Failed to provision unit.

jarlah, 2018-11-30:
Ah i understand now. Sorry me being slow. Yes all those messages are legit but shoukdnt need to be error. They are warn

jarlah, 2018-11-30:
The cache things should be debug actually

jarlah, 2018-11-30:
Do you want me to tune the log levels? You have reported such things several times. And its pretty legit from an "end users perspective". If something is not an actual error it should be warn

jarlah, 2018-11-30:
Hi :) Nice to hear that this was merely a confusement rather than a bug report. The project is open source and thus stack traces helps me a lot. However i will look into this issue with regards to not include stack trace for unauthorized or when unit does not exist. And give a human readable warn level log message instead.

closed 2018-11-30 by jarlah