freeacs/freeacs #277: Mishandling of short uinitId?

dror27, 2018-12-03:
on unitId.length() < 6 this line throws an IndexOutOfBound... exception. 

[or is is the case that there is a lower size limit enforced on unitId elsewhere?]

https://github.com/freeacs/freeacs/blob/fd955afe5b4b79545aa80cfad5e071574b353b74/tr069/src/main/java/com/github/freeacs/base/db/DBAccessSessionTR069.java#L39

jarlah, 2018-12-03:
Hah. Nice one. That code is actually wrong. What we actually want here is Math.min(s.length(), 6)

jarlah, 2018-12-03:
However i think it has been like this for a long time. But i will check and maybe fix it.

Or you could :)

jarlah, 2018-12-03:
There is no issue that the unitId is short. Its weird but not a problem. The unit should not be able to authenticate without a valid unitId anyway so, so it can never be null in a sane environment. You can of course disable auth in properties and restart and lo and behold, the code might not care about the unitIds length or nullness, I dont know actually. Note to self: fun exercise ;)

jarlah, 2018-12-03:
This fix and a couple of other fixes released:

https://github.com/freeacs/freeacs/releases/tag/v2.1.5

Can you test? Closing this issue. Please just reopen if it still doesnt work, open new issue or just comment here.

closed 2018-12-03 by jarlah

dror27, 2018-12-04:
Thanx for fixing. I was off line for a while and I see that you tested/closed. I will do a pull request (fix) next time. Enjoying your project.