freeacs/freeacs #278: Help page and about page not working (nginx + ssl related?)

jarlah, 2018-12-03:
Not sure if its actually nginx or ssl related, but its not working. And needs some fixes.

[image]

Daniel-Beardsmore, 2019-05-16:
Well, it’s definitely broken here in nginx with HTTPS, and since that is the only documentation at all for this thing …

I don’t have a way to test bypassing nginx though — I can access port 8081 fine in links on localhost but login attempts are just ignored. I cannot get to that server on that port externally.

jarlah, 2019-05-19:
@Daniel-Beardsmore The whole micro service and nginx thing is broken. I am currently working steadily on moving away from an enterprise like stack. Until I have made more progress on the new stack, which is soon ready for some testing, you can try to bypass the ningx layer,  by accessing the web directly from the server on port something something (8086? don't remember... its in cleartext in nginx config and application config files)

closed 2021-09-04 by jarlah