freeacs/freeacs #282: Does freeACS support IPv6?

disclean, 2018-12-19:
Good day,
It's nice to find here and I've one question.

Does freeACS support IPv6?
I'm trying to find some clue on web and source files but I couldn't.
I'll be nice if someone could say the answer clearly.

pingwiniasty, 2018-12-19:
What do you mean about supporting ipv6? 
Are you want to connect CPE with ACS by IPv6? Or send some parameters with IPv6 Address?

disclean, 2018-12-20:
Yes, the first one. Whether CPE and ACS connect each other by IPv6.
Probably ConnectionRequestURL must be IPv6 too.

jarlah, 2018-12-20:
Address syntax shouldn't necessary be a problem. However, there is a problem when it comes to detection logic inside the tr069 service for public ip and such; it will not recognize an ipv6 address as public even if it is. 

But, freeacs is now using sparkjava which i am not yet familiar with regarding support for ipv6. 

So the answer is, I dont know actually. Configure a ipv6 interface, try to start the services and check if they are reachable. The best answer is the one you get by trying ;)

disclean, 2018-12-20:
Well, you look like know the point.
I'm checking this issue in IPv6 environment.
The basic functions (provisioning, send&receive tr069 message, ...) work well.
But seems that the kick is not working properly. 
CPE could not get any signal from freeACS after kick so CPE does nothing and so freeACS also couldn't get response of it in 30s of course. 
The Device.ManagementServer.ConnectionRequestURL format is "http://[xx:xx:x:xx::x:x]:50003/creq", public IP. CPE is listening the 50003 port.

The kick function worked well in IPv4 environment
so I was wondering whether the problem is in my IPv6 environment or in freeACS.
(I'm using old version which uses install-or-update-freeacs-ubuntu.sh to install.)

jarlah, 2018-12-20:
I do not support the old installation, sorry. To get support for behaviour and to get fixes, you should use the script documented on this repos wiki page.

But we can look into this, and see if its possible, to support ipv6.

My guess is that the code isn't aligned for connecting to ipv6 urls. But I am just throwing balls at it now. I need to look at it and test it properly.

jarlah, 2018-12-20:
the new installation method does not rely on tomcat and each module is installed as independently deployed applications. This allows us to restart each module without affecting the others. And we don't have the bloat from a shared tomcat instance.

jarlah, 2019-02-08:
@pingwiniasty ipv6 works right? i think i fixed this problem with detecting if its public or not some time ago.

jarlah, 2019-02-08:
@penaivanalejandro what about you? are you using ipv6?

penaivanalejandro, 2019-02-08:
Hi @jarlah 

We are currently testing with IPv4.
At the beginning of March we started testing with dual stack which is when I could give feedback.

pingwiniasty, 2019-02-09:
@jarlah We are still not using ipv6

pingwiniasty, 2019-03-05:
We will test it maybe this week :)

pingwiniasty, 2019-03-19:
Currently we are trying to test it on one device, but it hangs when we provide IPv6 in TR069 url ;)

pmacangussu, 2019-07-15:
Hi Guys...
I have done many things with freeACS and it s just great. But I am still looking forward to set up IPV6 on my devices. When I READ ALL, it shows my devices parameter and its just equal on profile set. 
So, it seems not to be misspelled parameter. The device is a HS8545MS from huawei. As I said...i can fix everything I've tried but IPv6. 
Here is my strings to be fixed. and it DOESN'T show in syslog with ProvMsg.

InternetGatewayDevice.WANDevice.1.WANConnectionDevice.1.WANPPPConnection.1.X_HW_IPv6.IPv6Address.1.Origin

InternetGatewayDevice.WANDevice.1.WANConnectionDevice.1.WANPPPConnection.1.X_HW_IPv6.IPv6Prefix.1.Origin

And here is my 500 last line of tr-69-conversations.log
[tr069-conversation.log](https://github.com/freeacs/freeacs/files/3394182/tr069-conversation.log)

closed 2021-09-04 by jarlah