freeacs/freeacs #283: Missing appenders for freeacs-shell logback.xml

Daniel-Beardsmore, 2019-02-06:
Starting freeacs-shell triggers a flood of information, much of it pointless. (Does anyone need to be reminded that logs rotate at midnight?)

This includes:

… Could not find an appender named [ASYNC_FILE] …
… Could not find an appender named [ASYNC_STDOUT] …

These appenders are missing from the config file, and since this is buried inside the .jar file, I can't correct this in /opt/freeacs-shell/config/logback.xml … (it seems odd that these config files exist on disc if they are ignored and duplicate files are read from the .jar …)

FreeACS 2.1.5, Ubuntu 18 LTS, fresh installation.

jarlah, 2019-02-07:
Thanks for the report. If not someone else fixes it i will maybe look at it some Day.

jarlah, 2019-02-07:
https://github.com/freeacs/freeacs/pull/286

jarlah, 2019-02-09:
Fixed in: 
https://github.com/freeacs/freeacs/releases/tag/v2.1.6

jarlah, 2019-02-09:
Closing as its fixed, please test and reopen if its still not working

closed 2019-02-09 by jarlah

jarlah, 2019-02-12:
This has now been verified to be fixed.