freeacs/freeacs #284: freeacs-shell only runs from inside /opt/freeacs-shell

Daniel-Beardsmore, 2019-02-06:
Upon typing "freeacs-shell", following the flood of messages, it dies:

Exception in thread "main" java.lang.NullPointerException
        at com.github.freeacs.shell.ACSShellExec.main(ACSShellExec.java:10)

This happens if the working directory is not one of a set number of places including (and possibly limited to) /opt/freeacs-shell and /opt/freeacs-shell/config.

Line 10 seems to be 'SyslogClient.SYSLOG_SERVER_HOST = Properties.pr.getProperty("syslog.server.host");' and it's not clear from this why it would die. There are no prior stack frames in that exception.

FreeACS 2.1.5, Ubuntu 18 LTS.

Daniel-Beardsmore, 2019-02-06:
Ah, the PDF is wrong. The PDF does not match https://github.com/freeacs/freeacs/wiki/Use-Freeacs-Shell — it pretends that typing "freeacs-shell" actually works. (Probably for the same reason that it informs you that "You can check to see if tomcat is available by using the command.")

However, it should be really easy to dig Java out of its stupid hole. Replace the symlink at /usr/bin/freeacs-shell with a shell script thus:

#!/bin/sh
cd /opt/freeacs-shell
bin/freeacs-shell

Now it will actually work as documented, notwithstanding the flood of messages it throws out. (The sheer amount of rubbish it emits makes it hard to determine just where it's all going wrong.)

jarlah, 2019-02-07:
I like your direct attitude ;) yes i know its a lot of rubbish output. This is open for Contribution. I can fix it but best if someone throws in a pull request. Oh and i really really need to remove all references to tomcat which is no used

jarlah, 2019-02-07:
https://github.com/freeacs/freeacs/pull/286

jarlah, 2019-02-09:
Fixed in: 
https://github.com/freeacs/freeacs/releases/tag/v2.1.6

jarlah, 2019-02-09:
Closing as its fixed, please test and reopen if its still not working

closed 2019-02-09 by jarlah

Daniel-Beardsmore, 2019-02-11:
Now it's completely dead (this is with just the shell .deb installed, and with all the new .deb files installed, in Ubuntu 18 LTS):

user@host:~# freeacs-shell
Error: Could not find or load main class com.github.freeacs.shell.ACSShellExec
user@host:~# cd /opt/freeacs-shell/
user@host:/opt/freeacs-shell# freeacs-shell
Error: Could not find or load main class com.github.freeacs.shell.ACSShellExec
user@host:/opt/freeacs-shell# cd bin/
user@host:/opt/freeacs-shell/bin# freeacs-shell
Error: Could not find or load main class com.github.freeacs.shell.ACSShellExec
user@host:/opt/freeacs-shell/bin# ls
freeacs-shell  logs
user@host:/opt/freeacs-shell/bin# ./freeacs-shell
Error: Could not find or load main class com.github.freeacs.shell.ACSShellExec

The server has also had a reboot just in case, as well as a removal (dpkg -r FreeACS-Shell) and reinstallation of the package.

jarlah, 2019-02-11:
Try again. Details smetails. But yet another release.

https://github.com/freeacs/freeacs/releases/tag/v2.1.8

Daniel-Beardsmore, 2019-02-12:
That works! There is only one warning left now, and that's not related to the working directory.

(Interestingly, pressing ^C at the Username prompt generates a stack trace for "java.io.IOException: Resource temporarily unavailable".)

jarlah, 2019-02-12:
Ill leave the last "bug" be.

Thanks for testing!

I hope the stack works for you.