freeacs/freeacs #285: 404 from wget localhost/tr069

Daniel-Beardsmore, 2019-02-06:
The documentation (Fusion Installation.pdf) gives "wget localhost/tr069" as a test command. This returns 404.

This subdirectory is proxied into http://localhost:8085/tr069, which is also a 404.

In /var/log/freeacs-tr069, there are four files:

tr069-conversation.log -- empty
tr069-event.log -- empty
tr069-stability.log -- uhhhh?? (how unstable is something if it needs a stability log?)

tr069.log would, one presumes, have a log entry for each incoming request, but these 404s don't register -- that test does not cause anything to be logged. Moreover, this file was pre-filled with randomly-occurring "com.mysql.cj.exceptions.CJCommunicationsException: Communications link failure" errors -- however, all you get is stack traces, which are useless without a log entry that says what the program was doing at the time the crash occurred. "Communications link failure" ... with what? What protocol? What port? What destination?

Very odd ... it's worrying when the setup guide sends you to a 404 on a service that just logs random unexplained errors.

FreeACS 2.1.5, Ubuntu 18 LTS, fresh installation.

jarlah, 2019-02-07:
Thanks for the report. I have omitted normal get. Use curl - X POST -u admin:<password> <url>

I can add normal get back but it has no purpose. Try /tr069/ok

jarlah, 2019-02-07:
Stability log is not what it seems. Its legacy from the tomcat era. It will soon be removed. Or someone else can Just level=off it in logback.

jarlah, 2019-02-07:
https://github.com/freeacs/freeacs/pull/286

jarlah, 2019-02-07:
solved most or all of your issues. will make a new release 2.1.6 tomorrow. Thanks for the bug reports! 👍

Daniel-Beardsmore, 2019-02-08:
I don’t mind what the command is for testing it, so long as the setup guide and the software agree with each other ;-)

It was alarming to have some of the test commands fail, as it implied that the installation had gone awry.

jarlah, 2019-02-09:
Fixed in: 
https://github.com/freeacs/freeacs/releases/tag/v2.1.6

jarlah, 2019-02-09:
Closing as its fixed, please test and reopen if its still not working

closed 2019-02-09 by jarlah

Daniel-Beardsmore, 2019-02-11:
Half-way there. The PDF says:

"wget localhost/tr069"

This is still a 404, but wget localhost/tr069/ with a trailing slash works. (That is something that I had tried previously just in case, but it had not worked in 2.1.5.)

jarlah, 2019-02-11:
Try again. Details smetails. But yet another release.

https://github.com/freeacs/freeacs/releases/tag/v2.1.8

Daniel-Beardsmore, 2019-02-12:
That works!

jarlah, 2019-02-12:
Thanks for testing!