freeacs/freeacs #287: java.lang.OutOfMemoryError: Metaspace   ---->WebService

penaivanalejandro, 2019-02-08:
Hello @jarlah , I'm doing tests with the webservice and today I wake up with this error:
[image]

Do you know if there is a fix to increase the metaspace for this module?

jarlah, 2019-02-08:
i suppose you have some ram available. Just increase metaspacesize to 100 or 200 or something, like ```-XX:MaxMetaspaceSize=200m````. and if you have the memory, set ```-Xmx=1024m```. It has been running for 2 weeks i see, not good, but the memory parameters are adjustable for your liking. Its in application.ini under /opt for the module

jarlah, 2019-02-08:
fix it, let it run for another to or three weeks untouched, and we can take it from there. But i will take a deeper look at this problem, by adding best practice jvm params for memory management.

penaivanalejandro, 2019-02-08:
Thanks @jarlah 
Ready, I'm going to try these 3 weeks ...
Excellent support, thank you very much.

[image]

jarlah, 2019-02-12:
Ill leave this issue open since i need to fix the default config in webservice module and possibly others.

Or anyone else can do it. Really.

pingwiniasty, 2019-02-13:
1GB of memory and ~256M to MetaSpace?

pingwiniasty, 2019-03-19:
@penaivanalejandro Can you tell me, are you have some troubles with webservice? I ran it with ~20 CPEs ang 2GB of memory. After some time, webservice java process uses 100% of CPU

pingwiniasty, 2019-03-19:
It seems it is a problem with trying to fetch CPE which not exist in ACS.

penaivanalejandro, 2019-03-19:
Hi @pingwiniasty I have 8Gb of memory....  this is the status on the machine...

I have this CPE today:
[image]

systemctl status to webservice:
[image]

Hardware:
[image]

[image]

And I do a test to get a Unit:

[image]

[image]

jarlah, 2019-03-19:
Might not be related. But. 

I have a hunch. The dbi way of doing things is very impractical. Each user must have its own DBI instance which leads to many long living DBI instances.

We need to figure out how to do multi user on singleton dbi.

jarlah, 2019-03-19:
This multi user singleton pattern should also be considered for new web implementation.

PR #294 (2019-03-05): Patch issue 287
Incremets memory limits in default config.
Touches only senstive services 

Fixes #287
closed 2021-09-04 by jarlah