freeacs/freeacs #289: [tr069] ACS responding with method AutonomousTransferCompleteResponse when CPE sends method TransferComplete

toshht, 2019-02-20:
After doing a successful firmware upgrade initiated from the ACS, the CPE sends a TransferComplete message, but the ACS responds with an AutonomousTransferComplete message. This behavior triggers an error on the CPE since it was expecting a TransferComplete response.

This is the output from the log file /var/log/freeacs-tr069/tr069-conversation.log

2019-02-20 12:31:43 [qtp1947519330-1839] INFO  Conversation - ============== FROM CPE ( torres9 ) TO ACS ===============
<soapenv:Envelope xmlns:soap='http://schemas.xmlsoap.org/soap/encoding/' xmlns:xsd='http://www.w3.org/2001/XMLSchema' xmlns:cwmp='urn:dslforum-org:cwmp-1-0' xmlns:soapenv='http://schemas.xmlsoap.org/soap/envelope/' xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance'><soapenv:Body><cwmp:Inform><DeviceId><Manufacturer>MikroTik</Manufacturer><OUI>E48D8C</OUI><ProductClass>RB2011iL</ProductClass><SerialNumber>7DCF08DC745D</SerialNumber></DeviceId><Event soap:arrayType='cwmp:EventStruct[3]'><EventStruct><EventCode>7 TRANSFER COMPLETE</EventCode><CommandKey></CommandKey></EventStruct><EventStruct><EventCode>M Download</EventCode><CommandKey>6.41.1</CommandKey></EventStruct><EventStruct><EventCode>1 BOOT</EventCode><CommandKey></CommandKey></EventStruct></Event><MaxEnvelopes>1</MaxEnvelopes><CurrentTime>2019-02-20T12:31:42-05:00</CurrentTime><RetryCount>0</RetryCount><ParameterList soap:arrayType='cwmp:ParameterValueStruct[7]'><ParameterValueStruct><Name>Device.RootDataModelVersion</Name><Value xsi:type='xsd:string'>2.11</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.SoftwareVersion</Name><Value xsi:type='xsd:string'>6.41.1</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ProvisioningCode</Name><Value xsi:type='xsd:string'></Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.HardwareVersion</Name><Value xsi:type='xsd:string'>v1.0</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ParameterKey</Name><Value xsi:type='xsd:string'>No data in DB</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ConnectionRequestURL</Name><Value xsi:type='xsd:string'>http://192.168.13.108:7547/645370e942a7ff0477cb717e6d8e1bc69ba7</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.AliasBasedAddressing</Name><Value xsi:type='xsd:boolean'>0</Value></ParameterValueStruct></ParameterList></cwmp:Inform></soapenv:Body></soapenv:Envelope>

2019-02-20 12:31:43 [qtp1947519330-1839] INFO  Conversation - =============== FROM ACS TO ( torres9 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header/>
<soapenv:Body>
		<cwmp:InformResponse>
			<MaxEnvelopes>1</MaxEnvelopes>
		</cwmp:InformResponse>
</soapenv:Body>
</soapenv:Envelope>

2019-02-20 12:31:43 [qtp1947519330-1789] INFO  Conversation - ============== FROM CPE ( torres9 ) TO ACS ===============
<soapenv:Envelope xmlns:soap='http://schemas.xmlsoap.org/soap/encoding/' xmlns:xsd='http://www.w3.org/2001/XMLSchema' xmlns:cwmp='urn:dslforum-org:cwmp-1-0' xmlns:soapenv='http://schemas.xmlsoap.org/soap/envelope/' xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance'><soapenv:Body><cwmp:TransferComplete><CommandKey>6.41.1</CommandKey><FaultStruct><FaultCode>0</FaultCode><FaultString></FaultString></FaultStruct><StartTime>2019-02-19T20:51:33-05:00</StartTime><CompleteTime>2019-02-19T20:51:40-05:00</CompleteTime></cwmp:TransferComplete></soapenv:Body></soapenv:Envelope>

2019-02-20 12:31:43 [qtp1947519330-1789] INFO  Conversation - =============== FROM ACS TO ( torres9 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header/>
<soapenv:Body>
		<cwmp:AutonomousTransferCompleteResponse />
</soapenv:Body>
</soapenv:Envelope>

jarlah, 2019-02-20:
Don't remember all the context but i think that i implemented this due to another reported issue.

toshht, 2019-02-20:
I believe it was implemented on #197. But, the ACS should only respond with an AutonomousTransferComplete when an event code "10 AUTONOMOUS TRANSFER COMPLETE" arrives at the ACS.

In the case I described, the CPE is sending an event code "7 TRANSFER COMPLETE" and thus should receive a TransfereCompleteResponse.

toshht, 2019-02-21:
Today I experienced the same behavior during a TR069_SCRIPT execution. Is there an easy way to fix this?

2019-02-21 11:30:18 [qtp752007186-4540] INFO  Conversation - ============== FROM CPE ( torres10 ) TO ACS ===============
<soapenv:Envelope xmlns:soap='http://schemas.xmlsoap.org/soap/encoding/' xmlns:xsd='http://www.w3.org/2001/XMLSchema' xmlns:cwmp='urn:dslforum-org:cwmp-1-0' xmlns:soapenv='http://schemas.xmlsoap.org/soap/envelope/' xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance'><soapenv:Body><cwmp:Inform><DeviceId><Manufacturer>MikroTik</Manufacturer><OUI>E48D8C</OUI><ProductClass>RB2011iL</ProductClass><SerialNumber>7DCF08DC745D</SerialNumber></DeviceId><Event soap:arrayType='cwmp:EventStruct[2]'><EventStruct><EventCode>7 TRANSFER COMPLETE</EventCode><CommandKey></CommandKey></EventStruct><EventStruct><EventCode>M Download</EventCode><CommandKey>1.0.3</CommandKey></EventStruct></Event><MaxEnvelopes>1</MaxEnvelopes><CurrentTime>2019-02-21T11:30:17-05:00</CurrentTime><RetryCount>0</RetryCount><ParameterList soap:arrayType='cwmp:ParameterValueStruct[7]'><ParameterValueStruct><Name>Device.RootDataModelVersion</Name><Value xsi:type='xsd:string'>2.11</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.SoftwareVersion</Name><Value xsi:type='xsd:string'>6.41.3</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ProvisioningCode</Name><Value xsi:type='xsd:string'></Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.HardwareVersion</Name><Value xsi:type='xsd:string'>v1.0</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ParameterKey</Name><Value xsi:type='xsd:string'>No data in DB</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ConnectionRequestURL</Name><Value xsi:type='xsd:string'>http://192.168.13.108:7547/0ec0551e38ae88514c64df008bfafe581e59</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.AliasBasedAddressing</Name><Value xsi:type='xsd:boolean'>0</Value></ParameterValueStruct></ParameterList></cwmp:Inform></soapenv:Body></soapenv:Envelope>

2019-02-21 11:30:18 [qtp752007186-4540] INFO  Conversation - =============== FROM ACS TO ( torres10 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header/>
<soapenv:Body>
		<cwmp:InformResponse>
			<MaxEnvelopes>1</MaxEnvelopes>
		</cwmp:InformResponse>
</soapenv:Body>
</soapenv:Envelope>

2019-02-21 11:30:18 [qtp752007186-4539] INFO  Conversation - ============== FROM CPE ( torres10 ) TO ACS ===============
<soapenv:Envelope xmlns:soap='http://schemas.xmlsoap.org/soap/encoding/' xmlns:xsd='http://www.w3.org/2001/XMLSchema' xmlns:cwmp='urn:dslforum-org:cwmp-1-0' xmlns:soapenv='http://schemas.xmlsoap.org/soap/envelope/' xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance'><soapenv:Body><cwmp:TransferComplete><CommandKey>1.0.3</CommandKey><FaultStruct><FaultCode>0</FaultCode><FaultString></FaultString></FaultStruct><StartTime>2019-02-21T11:30:15-05:00</StartTime><CompleteTime>2019-02-21T11:30:17-05:00</CompleteTime></cwmp:TransferComplete></soapenv:Body></soapenv:Envelope>

2019-02-21 11:30:18 [qtp752007186-4539] INFO  Conversation - =============== FROM ACS TO ( torres10 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header/>
<soapenv:Body>
		<cwmp:AutonomousTransferCompleteResponse />
</soapenv:Body>
</soapenv:Envelope>

toshht, 2019-02-21:
I think I found where the problem is. I'm going to make the change, try to compile the project and I'll let you know if it solves the issue.

freeacs/tr069/src/main/java/com/github/freeacs/tr069/methods/TCDecision.java
.
.
.
public class TCDecision {
  public static void process(HTTPReqResData reqRes) {

    try {
      Fault fault = reqRes.getRequest().getFault();
      if (fault != null && !fault.getFaultCode().equals("0")) {
        String errormsg = "TC request reports a faultcode (" + fault.getFaultCode();
        errormsg += ") with faultstring (" + fault.getFaultString() + ")";
        Log.error(TCDecision.class, errormsg);
      }
    } finally {
      reqRes.getResponse().setMethod(TR069Method.AUTONOMOUS_TRANSFER_COMPLETE); <--- here should be TRANSFER_COMPLETE
    }
  }
}

toshht, 2019-02-22:
I compiled the code and run some tests including Firmware upgrade and Vendor Config, and all worked perfectly.

Is it possible to implement this change into the project?

jarlah, 2019-02-24:
Good work investigating. I would however check if we store the boot code in the session cache. If we are not we should do it. And then read the boot code in this method and check what method should be returned. I or someone else might have to dig a bit down in git history or unncomment some code and additionally make some adjustements

jarlah, 2019-02-24:
@toshht please check code in #291 to see if I have nailed it.

toshht, 2019-02-25:
@jarlah you made a lot of changes! Since you're the expert I'm just going to compile the code, run the same tests and I'll let you know how they went.

toshht, 2019-02-26:
@jarlah I ran some tests and it's working great! But now I have one issue and I don't know if is related to the changes made.

Running updates from the ACS to the CPE works fine. But, when I run a TR069_SCRIPT on the CPE, the ACS shows me an error indicating that the CPE is sending an empty response. I checked the tr069-conversation log and I can see that the CPE is sending a correct response to the server. I don't understand why the ACS is showing that error.

It doesn't matter what I send in the TR069 script, it always gives me the same error.

The only way to get this fixed is to delete the provisioning on the ACS, creating it again, and then the ACS recognizes the response from the CPE.

I uploaded the logs on a normal boot scenario and after TR069 script execution. Can you please look into it? Maybe I'm missing something and it's not related to the changes made.

Normal boot: https://pastebin.com/fVABNG7N
After TR069 script boot: https://pastebin.com/QP1mcdqi

jarlah, 2019-02-27:
Thank you for the logs. This is a typical programming error or logic error. Much simpler to debug :)

toshht, 2019-02-27:
Great! Is there any place I should start looking for the cause? Or how could I solve this?

jarlah, 2019-02-27:
```
  /**
   * Retrieves info from vendor-config-files and populate this map: Key: Name of config-file Value:
   * Version of config-file This method will crash with NP if the device returns a
   * VendorConfigFile.{i}.Name, but not the corresponding VendorConfigFile.{i}.Version
   */
  public Map<String, String> getConfigFileMap() {
    Map<String, String> cMap = new HashMap<>();
    for (Map.Entry<String, ParameterValueStruct> entry : cpeParams.entrySet()) {
      if (entry.getKey().contains(CONFIG_FILES) && entry.getKey().endsWith("Name")) {
        String namePN = entry.getKey();
        String namePV = entry.getValue().getValue();
        String versionPN = namePN.substring(0, namePN.length() - "Name".length()) + "Version";
        String versionPV = cpeParams.get(versionPN).getValue();
        cMap.put(namePV, versionPV);
      }
    }
    return cMap;
  }
```
this is the place where it chrashes . This code is .. put midly, not easy to read and relies on the current state of the class containing the method. We need to make unit tests for this method when we know what state caused the NullPointerException.
The line that crashed is:
```
        String versionPV = cpeParams.get(versionPN).getValue();
```
which tells us that there is no versionPN (whatever that is, difficult to understand out of the box, version parameter name?) and so it crashes on .getValue().... at least thats my theory

What i would do is start the server in debug mode, and use postman to send the series of payloads to the server. And try to step through. But I could rather just investigate the soap messages and look for abnormalities.

jarlah, 2019-02-27:
for example, a little training on making unit tests here, we can make a test like this:
```
package com.github.freeacs.tr069;

import org.junit.Test;

import java.util.Map;

import static org.junit.Assert.*;

public class CPEParametersTest {

    @Test
    public void getConfigFileMap() {
        // Given:
        CPEParameters cpeParameters = new CPEParameters("Device");
        // When:
        Map<String, String> result = cpeParameters.getConfigFileMap();
        // Then:
        assertNotNull(result);
    }
}
```
which succeeds, telling me that we need more context

jarlah, 2019-02-27:
See here, this is the context we need to add to the test:
```
                <ParameterValueStruct>
                    <Name>Device.DeviceInfo.VendorConfigFile.1.Name</Name>
                    <Value xsi:type='xsd:string'>Current ROS configuration</Value>
                </ParameterValueStruct>
                <ParameterValueStruct>
                    <Name>Device.DeviceInfo.VendorConfigFile.1.Description</Name>
                    <Value xsi:type='xsd:string'>Currently applied ROS configuration which will return dynamic /export
                        output when Uploaded.
                    </Value>
                </ParameterValueStruct>
                <ParameterValueStruct>
                    <Name>Device.DeviceInfo.VendorConfigFile.1.UseForBackupRestore</Name>
                    <Value xsi:type='xsd:boolean'>0</Value>
                </ParameterValueStruct>
```

jarlah, 2019-02-27:
There is no "Version" for ConfigFile.1

jarlah, 2019-02-27:
This fails:
```
    @Test
    public void onMissingVersion() {
        // Given:
        CPEParameters cpeParameters = new CPEParameters("Device.");
        cpeParameters.putPvs("Device.DeviceInfo.VendorConfigFile.1.Name", new ParameterValueStruct(
                "Device.DeviceInfo.VendorConfigFile.1.Name",
                "Current ROS configuration",
                "xsd:string"
        ));
        cpeParameters.putPvs("Device.DeviceInfo.VendorConfigFile.1.Description", new ParameterValueStruct(
                "Device.DeviceInfo.VendorConfigFile.1.Description",
                "Currently applied ROS configuration which will return dynamic /export output when Uploaded.",
                "xsd:string"
        ));
        cpeParameters.putPvs("Device.DeviceInfo.VendorConfigFile.1.UseForBackupRestore", new ParameterValueStruct(
                "Device.DeviceInfo.VendorConfigFile.1.UseForBackupRestore",
                "0",
                "xsd:boolean"
        ));
        // When:
        Map<String, String> result = cpeParameters.getConfigFileMap();
        // Then:
        assertNotNull(result);
    }
```
yay no need to postman this

jarlah, 2019-02-27:
which is correct if we read the javadoc for the method getConfigFileMap. @toshht why doesnt the cpe return the .Version parameter?

jarlah, 2019-02-27:
@toshht added tests for CPEParameters. But i need to trust the javadoc until proven wrong. So i added test to verify that it crashes when missing Version parameter.

https://github.com/freeacs/freeacs/pull/291/commits/ac425f3595eb373e2840b8a80098b8871f01707a

jarlah, 2019-02-27:
point is, this is only used to verify is script download is setup correctly.
this code:
```
String scriptVersionFromCPE = cpeParams.getConfigFileMap().get(name);
          if (svDB != null && !svDB.equals(scriptVersionFromCPE)) {
            // upgrade
            scriptVersionFromDB = svDB;
            scriptName = name;
            break;
          }
```
will not set scriptVersionFromDB to svDB if the script version parameter name does not exist in cpeParams. Right now it will crash, but I dont see why could just log it as Error and continue returning false in DownloadLogicTR069.java#96.

jarlah, 2019-02-27:
proposed fix is:
1. Make sure getConfigFileMap() doesnt crash
2. Change to code above to:
```
          String scriptVersionFromCPE = cpeParams.getConfigFileMap().get(name);
          if (svDB != null **&& scriptVersionFromCPE != null** && !svDB.equals(scriptVersionFromCPE)) {
            // upgrade
            scriptVersionFromDB = svDB;
            scriptName = name;
            break;
          }
```

jarlah, 2019-02-27:
but i dont understand the code in DownloadLogicTR069 well enough to be sure what is correct here. I need to extract some code and make some more unit tests. God, why could people just make more unit tests .. pleaaase.

jarlah, 2019-02-27:
I have made a couple of tests including a fix in DownloadLogicTr069 which will not make the server crash on missing Version parameter, but will instead log it as ERROR.

2019-02-27 20:09:03 [main] ERROR c.g.freeacs.tr069.GetScriptVersion - No script version found for Device.DeviceInfo.VendorConfigFile.1.Version

See tests and changes in https://github.com/freeacs/freeacs/pull/291/commits/80e8bd5ffa74415291b458c4eafe16536895c010

toshht, 2019-02-27:
> which is correct if we read the javadoc for the method getConfigFileMap. @toshht why doesnt the cpe return the .Version parameter?

I'm not sure about this. Maybe it's the implementation of TR069 on the CPE that doesn't return that parameter.

toshht, 2019-02-27:
> I have made a couple of tests including a fix in DownloadLogicTr069 that will avoid making the server crash on missing Version parameter, but will instead log it as ERROR.
> 
> 2019-02-27 20:09:03 [main] ERROR c.g.freeacs.tr069.GetScriptVersion - No script version found for Device.DeviceInfo.VendorConfigFile.1.Version
> 
> See tests and changes in [80e8bd5](https://github.com/freeacs/freeacs/commit/80e8bd5ffa74415291b458c4eafe16536895c010)

Great! I'll check the changes and test them. I'll let you know what comes up.

jarlah, 2019-02-27:
@toshht yes please checkout the branch feature/autonomousTransferCompleteFix. I have added more tests, and more refactoring after my last comment.

toshht, 2019-02-27:
> proposed fix is:
> 
>     1. Make sure getConfigFileMap() doesnt crash
> 
>     2. Change to code above to:
> 
> 
> ```
>           String scriptVersionFromCPE = cpeParams.getConfigFileMap().get(name);
>           if (svDB != null && scriptVersionFromCPE != null && !svDB.equals(scriptVersionFromCPE)) {
>             // upgrade
>             scriptVersionFromDB = svDB;
>             scriptName = name;
>             break;
>           }
> ```
> 
> or better:
> 
> ```
>           String scriptVersionFromCPE = cpeParams.getConfigFileMap().get(name);
>           if (scriptVersionFromCPE == null) {
>             Logger.error("Missing Version parameter value for " + name);
>           } else if (svDB != null && !svDB.equals(scriptVersionFromCPE)) {
>             // upgrade
>             scriptVersionFromDB = svDB;
>             scriptName = name;
>             break;
>           }
> ```

I think the second option is better.

jarlah, 2019-02-27:
@toshht its implemented like this now. But I guess you will reach into further problems along the way, but at least it wont crash.

jarlah, 2019-03-02:
merged #291. Tests are passing, and code looks code. So I plan to roll a minor release with these fixes.

jarlah, 2019-03-05:
released. Please verify latest release 
https://github.com/freeacs/freeacs/releases/tag/v2.2.0

closed 2019-03-05 by jarlah