freeacs/freeacs #290: Yeacomm P11 4G LTE Router | Envelope xml issue

niekvanderven, 2019-02-21:
FROM CPE (lan-router)  ( microtik302 ) TO ACS             : is running Ok to freeacs web page.

FROM CPE (4gRouter) ( test01.sdnbucks.com ) TO ACS  : envelope comes in (see part of log), but nothing to see on the web-page. No errors, and no parameters.

    the layout of the xml is different.

Is there a way to accommodate this on the ACS side or is this a fault by the Vendor?
Still if its a fault by the Vendor can we bypass it to get it to work until they fix it?

I have attached the tr069 model the CPE should support, auto-provisioning works but really limited to default parameters and never a write possible.
Thank you in advance!
ps if needed we can point a CPE towards a test URL if that helps the troubleshooting.

 

=================OK

2019-02-19 16:30:03 [qtp1255801684-251] INFO  Conversation - ============== FROM CPE ( microtik302 ) TO ACS ===============
<soapenv:Envelope xmlns
[CWMP-Data_Model-1.xlsx](https://github.com/freeacs/freeacs/files/2889300/CWMP-Data_Model-1.xlsx)
:soap='http://schemas.xmlsoap.org/soap/encoding/' xmlns:xsd='http://www.w3.org/2001/XMLSchema' xmlns:cwmp='urn:dslforum-org:cwmp-1-0' xmlns:soapenv='http://schemas.xmlsoap.org/soap/envelope/' xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance'><soapenv:Header><cwmp:ID soap:mustUnderstand='1'>FREEACS-1550590161273</cwmp:ID></soapenv:Header><soapenv:Body><cwmp:GetParameterValuesResponse><ParameterList soap:arrayType='cwmp:ParameterValueStruct[573]'><ParameterValueStruct><Name>Device.RootDataModelVersion</Name><Value xsi:type='xsd:string'>2.11</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.InterfaceStackNumberOfEntries</Name><Value xsi:type='xsd:unsignedInt'>5</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.SoftwareVersion</Name><Value xsi:type='xsd:string'>6.43.11</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ProvisioningCode</Name><Value xsi:type='xsd:string'></Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.Manufacturer</Name><Value xsi:type='xsd:string'>MikroTik</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ManufacturerOUI</Name><Value xsi:type='xsd:string'>E48D8C</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ModelName</Name><Value 

=================Not Ok

2019-02-19 15:27:02 [qtp1255801684-77] INFO  Conversation - ============== FROM CPE ( test01.sdnbucks.com ) TO ACS ===============
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/" xmlns:SOAP-ENC="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsi="http://www.w3.org/1999/XMLSchema-instance" xmlns:xsd="http://www.w3.org/1999/XMLSchema" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<SOAP-ENV:Header>
<cwmp:ID SOAP-ENV:mustUnderstand="1">6</cwmp:ID>
</SOAP-ENV:Header>
<SOAP-ENV:Body>
<cwmp:Inform>
<DeviceId xsi:type="cwmp:DeviceIdStruct">
<Manufacturer xsi:type="xsd:string">Shenzhen Tozed Technologies Co., Ltd</Manufacturer>
<OUI xsi:type="xsd:string">00037F</OUI>
<ProductClass xsi:type="xsd:string">CPE</ProductClass>
<SerialNumber xsi:type="xsd:string">S864262028824325</SerialNumber>
</DeviceId>
<Event SOAP-ENC:arrayType="cwmp:EventStruct[1]">
<EventStruct>
<EventCode xsi:type="xsd:string">2 PERIODIC</EventCode>
<CommandKey></CommandKey>
</EventStruct>
</Event>
<MaxEnvelopes xsi:type="xsd:unsignedInt">1</MaxEnvelopes>
<CurrentTime xsi:type="xsd:dateTime">2019-02-19T15:27:01Z</CurrentTime>
<RetryCount xsi:type="xsd:unsignedInt">0</RetryCount>
<ParameterList SOAP-ENC:arrayType="cwmp:ParameterValueStruct[17]">
<ParameterValueStruct>
<Name xsi:type="xsd:string">InternetGatewayDevice.DeviceSummary</Name>
<Value xsi:type="xsd:string">2</Value>
</ParameterValueStruct>
<ParameterValueStruct>
<Name xsi:type="xsd:string">InternetGatewayDevice.LANDeviceNumberOfEntries</Name>
<Value xsi:type="xsd:string">1</Value>
</ParameterValueStruct>
<ParameterValueStruct>
<Name xsi:type="xsd:string">InternetGatewayDevice.WANDeviceNumberOfEntries</Name>
<Value xsi:type="xsd:string">1</Value>

jarlah, 2019-02-24:
i am not completely sure what you are asking for here. I added code block around the xml, but still not sure what the real question is. Is the cpe sending an xml that is rejected by the acs? is the payload incomplete? And what is `[CWMP-Data_Model-1.xlsx](https://github.com/freeacs/freeacs/files/2889300/CWMP-Data_Model-1.xlsx)` in the middle of the first soap message?

appendixie, 2019-02-25:
CWMP-Data_Model-1.xlsx](https://github.com/freeacs/freeacs/files/2889300/CWMP-Data_Model-1.xlsx, this is from a normal-router with good communication, so here is no problem.
The problem is with the 4G router.
I add a picture, so you can see, that some parameter coming in, 

With a read all I normally get a column with the cpe values.
but that does not happen here.
Timestamps are green, and a number of parameters are coming in well.

Also a reset command gives nothing, even no error message.
It may be useful, as suggested by niekvanderven, to have the 4G router refer to your server.
Thanks for contributing ideas,
greetings, advanderven.

[image]
.

appendixie, 2019-02-28:
Ok Jahl,
I managed to add a few parameters, and after a read-all the CPE (current) value column appears.
But I also get an error, a reboot is needed, and it gives the following error.
[image]
I think this is easy to assess for you, thank you in advance.
gr. Ad.

jarlah, 2019-02-28:
Actually. Kick is very difficult to get working if you havent done it before. Actually i think its very difficult myself. I need to revisit the code. Then ill make a wiki page describe the different methods and parameter requirements.

appendixie, 2019-02-28:
Ok thanks for the quick response, I look forward to the wiki.
Kick, that worked well on the mikrotik router but unfortunately not with this 4G chinese wonder box.
gr.

pingwiniasty, 2019-03-04:
@appendixie You can simple check if your 4G router initializing connection to ACS by requesting ConncetionRequestURL from browser

Can you paste this parameter here?

appendixie, 2019-03-05:
pingwiniasty, thanks 
 The good thing: I do not have stun-errors anymore.

The 4G device sends:   <Name xsi:type="xsd:string">InternetGatewayDevice.ManagementServer.ConnectionRequestURL</Name>
<Value xsi:type="xsd:string">http://100.65.147.67:5400</Value> 

but this parameter do not shows on the web-page
My other device (cable-router) (Ok in ACS) does not use "xsd:string", but 'xsd:string'. 
  in the web-page,
   InternetGatewayDevice.ManagementServer.ConnectionRequestURL       keeps empty.
[image]
I think that the 4G router is talking differently and that the acs do not understand this.
The x-flags parameters from cpe are displayed good in the web page.

greetings Ad.

pingwiniasty, 2019-03-05:
@appendixie please change flag for `InternetGatewayDevice.ManagementServer.ConnectionRequestURL` to RA in UnitType and reboot device.

appendixie, 2019-03-05:
pingwiniasty, thanks again,
Your suggestion was already tried, but you never know ....
In the new photo is some text about a failed kick, and you can see that I do get info from the CPE.
gr. ad.
[image]

jarlah, 2019-03-05:
Actually i have the exact same problem with my own china device running easycwmp. Its impossible to get it to return connection params

jarlah, 2019-03-05:
But i will try the RA too. Havent been working with it for a while. But If acs asks and get less params back or empty values then we have no luck. Impossible to configure easycwmp on my china device. Maybe i should flash it.... With custom rom :P

appendixie, 2019-03-05:
Ok, nice to hear, that the expert ... has the same problems ... hi
I try something myself, but suggestions remain welcome.
But Jarl, in the last picture is some text (error message) which can not be ignored for a proper functioning.
I do not get back empty fields, but acs does not understand them.
this is a little part of the answer from cpe (indeed also chinese).
<ParameterValueStruct>
<Name xsi: type = "xsd: string"> InternetGatewayDevice.ManagementServer.URL </ Name>
<Value xsi: type = "xsd: string"> http: //xx.xx.248.187/tr069/prov </ Value>
</ ParameterValueStruct>

pingwiniasty, 2019-03-05:
`InternetGatewayDevice.ManagementServer.URL` is addres of ACS, not your CPE.

ACS needs `ConnectionRequestURL` to kick your device. The error message was correct

appendixie, 2019-03-05:
InternetGatewayDevice.ManagementServer.URL is addres of ACS, not your CPE.
 ==yes in the logfile it is my home-address==
ACS needs ConnectionRequestURL to kick your device. The error message was correct
==but now ?===
Can you paste full conversation log, after device reboot?
== here it comes==
[log.txt](https://github.com/freeacs/freeacs/files/2932426/log.txt)

pingwiniasty, 2019-03-05:
@appendixie you are paste log with "2 PERIODIC" event, I need "0 BOOTSTRAP" or "1 BOOT"
It comes when you reboot device or reset to factory defaults.
Please turn off your device, clear log, turn on device and check, what happens in log.

Maybe this device, sends some info, only with this events

pingwiniasty, 2019-03-05:
Sorry, I see `InternetGatewayDevice.ManagementServer.ConnectionRequestURL` in periodic.
Are you running in `discovery.mode = true`?
If no, please set it to true, restart tr069 modules and test this case:
1. Remove device from acs
2. Reboot it
3. Check if device available in ACS and check parameters

appendixie, 2019-03-05:
Ok,
discovery.mode = true.
i did a remove device, and made it runs with a other name to acs.
a new log after reset is the same as i uploaded before.
so, no BOOTSTRAP messages.
sorry.

pingwiniasty, 2019-03-05:
Are you have time tomorrow (I'm available at ~9AM  UTC+1) to connect your device to my development ACS? Maybe I find somethig?

appendixie, 2019-03-06:
Ok, that's good news. if I receive the connect string, then I will enter it in the 4g router.

pingwiniasty, 2019-03-06:
@appendixie please send me an email, is visible in my profile. I don't want share my test server in public

appendixie, 2019-03-06:
ok email was send to you.

pingwiniasty, 2019-03-06:
When I have enabled discovery mode, your device are responding on ACS GetParameterNames Query
```
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
	<cwmp:ID soapenv:mustUnderstand="1">FREEACS-1551865082125</cwmp:ID>
</soapenv:Header>
<soapenv:Body>
		<cwmp:GetParameterNames>
			<ParameterPath>InternetGatewayDevice.</ParameterPath>
			<NextLevel>false</NextLevel>
		</cwmp:GetParameterNames>
</soapenv:Body>
</soapenv:Envelope>
```

with empty. I don't know is this normal behaviour, I will read docs about it.

Easycwmp reponses on it with error, but next ACS query for GetParameterValues.

appendixie, 2019-03-06:
Ok Jakub,

yes I know this, and even more information is sent from the CPE, see the first message about this item on top of discussions.

 

I hope you find something more, I am patient, give me a sign if the router is allowed to be stopped.

 

Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

appendixie, 2019-03-06:
oeps, my mailserver is offline,
This is my replay to you:
Ok Jakub,
yes I know this, and even more information is sent from the CPE, see the first message about this item on top of discussions.

I hope you find something more, I am patient, give me a sign if the router is allowed to be stopped.

pingwiniasty, 2019-03-06:
IMO this is problem with cpe and TR069 implementation.
CPE is not responding to GPN or GPV queries...
To correct set parameters on your CPE, ACS need to know which parameters CPE already has.

I added to unit type some parameters and ACS trying to set it in CPE, but at first ACS queries CPE for current values 

```
2019-03-06 11:59:11 [qtp723861661-50] INFO  Conversation - =============== FROM ACS TO ( cpe101 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
	<cwmp:ID soapenv:mustUnderstand="1">FREEACS-1551869951544</cwmp:ID>
</soapenv:Header>
<soapenv:Body>
		<cwmp:GetParameterValues>
			<ParameterNames soapenc:arrayType="xsd:string[3]">
				<string>InternetGatewayDevice.ManagementServer.ConnectionRequestURL</string>
				<string>InternetGatewayDevice.ManagementServer.Password</string>
				<string>InternetGatewayDevice.ManagementServer.Username</string>
			</ParameterNames>
		</cwmp:GetParameterValues>
</soapenv:Body>
</soapenv:Envelope>

2019-03-06 11:59:12 [qtp723861661-45] INFO  Conversation - ============== FROM CPE ( cpe101 ) TO ACS ===============

2019-03-06 11:59:12 [qtp723861661-45] INFO  Conversation - =============== FROM ACS TO ( cpe101 ) ============
```

appendixie, 2019-03-06:
the actual values of the cpe (+/- 15 params) can be seen in conversation log.
And indeed the ACS can not deal with this.
But how to proceed?
gr. Ad.

pingwiniasty, 2019-03-06:
This is Inform Event, it's used to initiate connection to ACS

jarlah, 2019-03-06:
Maybe we should think about adding a mode that allows the ACS to do less queries, and more often make educated guesses about which parameters the cpe should have. Either based on inform params or via cwmp version the device reports using.

I dont know if its related. But i have got issues regarding that the ACS is a little bit too chatty

jarlah, 2019-03-06:
I am Just throwing ideas here. Not sure if its easily implementable or possible

pingwiniasty, 2019-03-06:
@jarlah but if CPE is not responding to GPN/GPV, ACS is unusable.

jarlah, 2019-03-06:
@pingwiniasty as I said .. throwing balls ;) you are defacto correct. I just said that you could potentially instrument a unit manually. But as you said, then you can just use something else than freeacs.

jarlah, 2019-03-06:
@pingwiniasty my point is, you are right and we are not discussing my curve balls. My china unit is also pretty darn stupid at times, it writes values with spaces, and bricks the devices. Nice.

appendixie, 2019-03-07:
Ok, it is too complicated for me now, I had hoped that I had done something wrong.
Ok, Jarl and Jakub both thank you for thinking along.

jarlah, 2019-03-08:
@appendixie sorry we couldn't help and/or went technical. If units does not respond properly on standard tr069 methods, then we can no do. Closing this issue now, but if it still contains issues that need to discuss you are free to reopen it, create a new issue or just continue discussion on the closed issue.

closed 2019-03-08 by jarlah