freeacs/freeacs #292: Centos Installation missing install.sql file

AndrewElz, 2019-02-27:
When using the Centos installation script, the installation fails at:

The FreeACS MySQL database user is OK.

Loads all FreeACS table defintions into MySQL
./install_centos.sh: line 72: install.sql: No such file or directory
The output from the installation of the tables indicate some
errors occurred:
------------------------------------------------
mysql: [Warning] Using a password on the command line interface can be insecure.
mysql: [Warning] Using a password on the command line interface can be insecure.
------------------------------------------------

Centos release is 

CentOS Linux release 7.5.1804 (Core)

Installing as root.

jarlah, 2019-02-27:
What happens around this point in the script?
```
  whattodownload="rpm|tables|pdf"
  curl -s https://api.github.com/repos/freeacs/freeacs/releases/latest | jq -r ".assets[] | select(.name | test(\"${whattodownload}\")) | .browser_download_url" > files.txt
  awk '{print $0;}' files.txt | xargs -l1 wget
  unzip "*.zip"
  rm -rf *.zip

  echo ""
  echo "All necessary FreeACS resources are available."
echo ""
```
whats inside the current directory? I just tried this code in my zsh here now, after installing jq, and it seems to download tables.zip and unzip and inflate install.sql.

AndrewElz, 2019-02-27:
Thanks, the problem was that unzip was missing.  Installation finished successfully, however when I try to connect I get 502 bad gateway.
When running journalctl -u freeacs* -f I see that freeacs is not starting up properly, there is a problem with connecting to the database:

localhost.localdomain freeacs-web[9978]: Caused by: java.sql.SQLNonTransientConnectionException: Could not create connection to database server. Attempted reconnect 3 times. Giving up.

This happens for STUN also.

I can connect to mysql using the acs user and the password created during installation:

mysql -uacs -p********************* acs
mysql: [Warning] Using a password on the command line interface can be insecure.
Reading table information for completion of table and column names
You can turn off this feature to get a quicker startup with -A

Welcome to the MySQL monitor.  Commands end with ; or \g.
Your MySQL connection id is 667
Server version: 5.7.25 MySQL Community Server (GPL)

Copyright (c) 2000, 2019, Oracle and/or its affiliates. All rights reserved.

Oracle is a registered trademark of Oracle Corporation and/or its
affiliates. Other names may be trademarks of their respective
owners.

Type 'help;' or '\h' for help. Type '\c' to clear the current input statement.

mysql>

In the FAQ there is reference to the xaps user which does not seem to be there.  I can also not find a *.web.property file to check if the user / pass is correct.

AndrewElz, 2019-02-28:
I have managed to get freeacs to start by changing the acs user password to acs using mysql.

It seems that the password used in the installation is not updated in the applicaiton-config files.

jarlah, 2019-02-28:
@AndrewElz it has been something that has been pushed down on my prio list. Passwords are printed at the end, and must be manually updated. If you have a scripted solution for it, feel free to contribute. CentOS script is more secure in that it doesnt use acs as password

jarlah, 2019-03-05:
released. Please verify latest release
https://github.com/freeacs/freeacs/releases/tag/v2.2.0
I have added unzip install and auto update of property files

closed 2019-03-05 by jarlah