freeacs/freeacs #298: Unit Jobs page - error 500

pingwiniasty, 2019-03-07:
When I trying to visit Unit-Jobs, server responses with 500 error.
I checked it and there is an error with default contructor in UnitJobPage, this class has 2 contructors.

getWebPage method in Main.java returns wrong contructor. It logic is broken, because always check first constructor.
https://github.com/freeacs/freeacs/blob/b778d8129a7ddaddcd198a861629c033a3c49f40/web/src/main/java/com/github/freeacs/web/app/Main.java#L59-L66

jarlah, 2019-03-07:
I have a fix. Refactor blunder.

pingwiniasty, 2019-03-07:
you are so fast :D

jarlah, 2019-03-07:
:D

I dont like the passing around datasource thing. But this is remnants of old code where i used spring controllers that auto wired datasource in constructor. If you want to help you can check if there is other similar constructors with equal uselessness :)

jarlah, 2019-03-07:
You can Just checkout branch and continue on it directly. You should be able to contribute on it

jarlah, 2019-03-07:
Merged pull request #299

pingwiniasty, 2019-03-07:
I checked other pages and I not found other constructors

jarlah, 2019-03-07:
done

closed 2019-03-07 by jarlah