freeacs/freeacs #302: Change url services

fernandezteran, 2019-03-11:
Hi, can you change the url of the services?

tr069 --- http://localhost:8085/web/index;
web --- http://localhost:8081/web2/;

pingwiniasty, 2019-03-11:
Hi, to provide clean urls, you can use nginx as a reverse proxy server. Look at config available in install script

```
events {
  worker_connections  19000;
}
http {
  server {
    listen       80;
    server_name  localhost;
    # For the geeks: "A man is not dead while his name is still spoken." -Terry Pratchett
    add_header X-Clacks-Overhead "GNU Terry Pratchett";
    location = / {
      return 301 /web/index;
    }
    location /tr069 {
      proxy_set_header        X-Real-IP       \$remote_addr;
      proxy_pass http://localhost:8085/tr069;
    }
    location /web/ {
      proxy_pass http://localhost:8081/web/;
    }
    location /monitor/ {
      proxy_pass http://localhost:8090/monitor/;
    }
    location /webservice/ {
      proxy_pass http://localhost:8088/webservice/;
    }
    location /syslog/ {
      proxy_pass http://localhost:8086/syslog/;
    }
    location /core/ {
      proxy_pass http://localhost:8083/core/;
    }
    location /stun/ {
      proxy_pass http://localhost:8087/stun/;
    }
  }
}
```

fernandezteran, 2019-03-11:
Thank you. I am going to try

closed 2019-03-19 by pingwiniasty