freeacs/freeacs #303: Simple Mikrotik provision

vittore, 2019-03-12:
Hi,

first of all thanks to all for yours efforts in this project.
I'm trying to work with Mikrotik router and freeacs.

Device is correctly identified, now I want to provision a simple setup: I want to setup periodic inform interval to 5 seconds.
I done this:

[image]

So I will expect this in my Mikrotik:

[image]

But I got this:
[image]

And in freeacs I got this:
[image]

Where is my mistake?

in botton there is transcript of /var/log/freeacs-tr069/tr069-conversation.log

A note: Mikrotik TR069 reference is https://wiki.mikrotik.com/tr069ref/current.html

Thanks in advance.
v.

```
2019-03-12 09:36:22 [qtp197026513-63] INFO  Conversation - ============== FROM CPE ( E48D8CA0D426 ) TO ACS ===============
<soapenv:Envelope xmlns:soap='http://schemas.xmlsoap.org/soap/encoding/' xmlns:xsd='http://www.w3.org/2001/XMLSchema' xmlns:cwmp='urn:dslforum-org:cwmp-1-0' xmlns:soapenv='http://schemas.xmlsoap.org/soap/envelope/' xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance'><soapenv:Body><cwmp:Inform><DeviceId><Manufacturer>MikroTik</Manufacturer><OUI>E48D8C</OUI><ProductClass>RB951Ui-2HnD</ProductClass><SerialNumber>6281058CD33D</SerialNumber></DeviceId><Event soap:arrayType='cwmp:EventStruct[1]'><EventStruct><EventCode>1 BOOT</EventCode><CommandKey></CommandKey></EventStruct></Event><MaxEnvelopes>1</MaxEnvelopes><CurrentTime>2019-03-12T14:42:52+01:00</CurrentTime><RetryCount>0</RetryCount><ParameterList soap:arrayType='cwmp:ParameterValueStruct[7]'><ParameterValueStruct><Name>Device.RootDataModelVersion</Name><Value xsi:type='xsd:string'>2.11</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.SoftwareVersion</Name><Value xsi:type='xsd:string'>6.44</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.ProvisioningCode</Name><Value xsi:type='xsd:string'></Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.HardwareVersion</Name><Value xsi:type='xsd:string'>v1.0</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ParameterKey</Name><Value xsi:type='xsd:string'>No data in DB</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ConnectionRequestURL</Name><Value xsi:type='xsd:string'>http://10.10.14.93:7547/8b7a55e8ae86e8e805397dd60bdaa27db07e</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.AliasBasedAddressing</Name><Value xsi:type='xsd:boolean'>0</Value></ParameterValueStruct></ParameterList></cwmp:Inform></soapenv:Body></soapenv:Envelope>

2019-03-12 09:36:22 [qtp197026513-63] INFO  Conversation - =============== FROM ACS TO ( E48D8CA0D426 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header/>
<soapenv:Body>
		<cwmp:InformResponse>
			<MaxEnvelopes>1</MaxEnvelopes>
		</cwmp:InformResponse>
</soapenv:Body>
</soapenv:Envelope>

2019-03-12 09:36:22 [qtp197026513-65] INFO  Conversation - ============== FROM CPE ( E48D8CA0D426 ) TO ACS ===============

2019-03-12 09:36:22 [qtp197026513-65] INFO  Conversation - =============== FROM ACS TO ( E48D8CA0D426 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
	<cwmp:ID soapenv:mustUnderstand="1">FREEACS-1552397782923</cwmp:ID>
</soapenv:Header>
<soapenv:Body>
		<cwmp:GetParameterValues>
			<ParameterNames soapenc:arrayType="xsd:string[6]">
				<string>Device.DeviceInfo.SoftwareVersion</string>
				<string>Device.DeviceInfo.VendorConfigFile.</string>
				<string>Device.ManagementServer.ConnectionRequestPassword</string>
				<string>Device.ManagementServer.ConnectionRequestURL</string>
				<string>Device.ManagementServer.ConnectionRequestUsername</string>
				<string>Device.ManagementServer.PeriodicInformInterval</string>
			</ParameterNames>
		</cwmp:GetParameterValues>
</soapenv:Body>
</soapenv:Envelope>

2019-03-12 09:36:22 [qtp197026513-55] INFO  Conversation - ============== FROM CPE ( E48D8CA0D426 ) TO ACS ===============
<soapenv:Envelope xmlns:soap='http://schemas.xmlsoap.org/soap/encoding/' xmlns:xsd='http://www.w3.org/2001/XMLSchema' xmlns:cwmp='urn:dslforum-org:cwmp-1-0' xmlns:soapenv='http://schemas.xmlsoap.org/soap/envelope/' xmlns:xsi='http://www.w3.org/2001/XMLSchema-instance'><soapenv:Header><cwmp:ID soap:mustUnderstand='1'>FREEACS-1552397782923</cwmp:ID></soapenv:Header><soapenv:Body><cwmp:GetParameterValuesResponse><ParameterList soap:arrayType='cwmp:ParameterValueStruct[8]'><ParameterValueStruct><Name>Device.DeviceInfo.SoftwareVersion</Name><Value xsi:type='xsd:string'>6.44</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.VendorConfigFile.1.Name</Name><Value xsi:type='xsd:string'>Current ROS configuration</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.VendorConfigFile.1.Description</Name><Value xsi:type='xsd:string'>Currently applied ROS configuration which will return dynamic /export output when Uploaded.</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.DeviceInfo.VendorConfigFile.1.UseForBackupRestore</Name><Value xsi:type='xsd:boolean'>0</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ConnectionRequestPassword</Name><Value xsi:type='xsd:string'></Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ConnectionRequestURL</Name><Value xsi:type='xsd:string'>http://10.10.14.93:7547/8b7a55e8ae86e8e805397dd60bdaa27db07e</Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.ConnectionRequestUsername</Name><Value xsi:type='xsd:string'></Value></ParameterValueStruct><ParameterValueStruct><Name>Device.ManagementServer.PeriodicInformInterval</Name><Value xsi:type='xsd:unsignedInt'>5</Value></ParameterValueStruct></ParameterList></cwmp:GetParameterValuesResponse></soapenv:Body></soapenv:Envelope>

2019-03-12 09:36:22 [qtp197026513-55] INFO  Conversation - =============== FROM ACS TO ( E48D8CA0D426 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
[... 58 lines cut ...]
2019-03-12 09:36:23 [qtp197026513-69] INFO  Conversation - =============== FROM ACS TO ( E48D8CA0D426 ) ============
<soapenv:Envelope xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:soapenc="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:cwmp="urn:dslforum-org:cwmp-1-0">
<soapenv:Header>
	<cwmp:ID soapenv:mustUnderstand="1">FREEACS-1552397783211</cwmp:ID>
</soapenv:Header>
<soapenv:Body>
		<cwmp:SetParameterValues>
			<ParameterList soapenc:arrayType="cwmp:ParameterValueStruct[1]">
				<ParameterValueStruct>
					<Name>Device.ManagementServer.PeriodicInformInterval</Name>
					<Value xsi:type="xsd:unsignedInt">96272</Value>
				</ParameterValueStruct>
			</ParameterList>
			<ParameterKey>No data in DB</ParameterKey>
		</cwmp:SetParameterValues>
</soapenv:Body>
</soapenv:Envelope>

2019-03-12 09:36:23 [qtp197026513-58] INFO  Conversation - ============== FROM CPE ( E48D8CA0D426 ) TO ACS ===============
<soapenv:Envelope xmlns:soap="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cwmp="urn:dslforum-org:cwmp-1-0" xmlns:soapenv="http://schemas.xmlsoap.org/soap/envelope/" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"><soapenv:Header><cwmp:ID soap:mustUnderstand="1">FREEACS-1552397783211</cwmp:ID></soapenv:Header><soapenv:Body>
<cwmp:SetParameterValuesResponse><Status>0</Status></cwmp:SetParameterValuesResponse></soapenv:Body></soapenv:Envelope>

2019-03-12 09:36:23 [qtp197026513-58] INFO  Conversation - =============== FROM ACS TO ( E48D8CA0D426 ) ============

```

jarlah, 2019-03-12:
you probably need to configure frequency and spread. I am not doing this everyday. But I think there is some mouse over tips. And there is also good documentation for this functionality in web docs or some other documentation.

pingwiniasty, 2019-03-12:
https://github.com/freeacs/freeacs/blob/master/web/docs/Fusion%20Web%20User%20Manual.pdf Page 22. This parameter can be configured in Unit or Unit Profile.

vittore, 2019-03-13:
Perfect. Resolved using  frequency and spread.

Thanks.
v.

closed 2019-03-13 by vittore