freeacs/freeacs #304: Wrong Download Firmware URL sent from ACS to CPE

dariosky75, 2019-03-12:
Hi all, 
thanks in advance for your support.

When I try upgrade CPE firmware it fails. 
I've notice that the URL sent over from the ACS to the CPE is wrong: http://**localhost**/tr069/file/SOFTWARE/v5.0.00/G640/001A77-G640-HTUAQERH6118576

I should be able to get localhost to be replaced with the target IP address(FreeACS server).
Is there some environment variable or configuration file I should modify?

BR,
Dario

pingwiniasty, 2019-03-12:
Hi @dariosky75, You need to provide correct url for your CPE in /opt/freeacs-web/config/application-config.conf in variable named `public.url`

dariosky75, 2019-03-14:
Hi @pingwiniasty tks a lot for the quick reply and for pointing me to the right direction.
I was able to fix this modifing public.url in **freeacs-tr069/config/application-config.conf**

pingwiniasty, 2019-03-14:
Sorry ^^' I though about tr069 but typed web...

closed 2019-03-14 by pingwiniasty