freeacs/freeacs #305: The device does not support PII

anteok, 2019-03-15:
ACS's syslog is full of messages like that:

`ProvMsg: PP:TR069, ST:ERROR, PO:EMPTY, SL:2395, PR:261, EV:2, PM:REGULAR, ER:CLIENT, EM:The device does not support PII `

CPE's syslog shows:
```
Mar 14 08:35:32 s501 daemon.notice easycwmpd: add event '2 PERIODIC'
Mar 14 08:35:32 s501 daemon.notice easycwmpd: start session
Mar 14 08:35:32 s501 daemon.notice easycwmpd: configured acs url _my_tr069server_url_
Mar 14 08:35:32 s501 daemon.notice easycwmpd: external script init
Mar 14 08:35:32 s501 daemon.notice easycwmpd: external: execute inform parameter
Mar 14 08:35:33 s501 daemon.notice easycwmpd: send Inform
Mar 14 08:35:33 s501 daemon.notice easycwmpd: receive InformResponse from the ACS
Mar 14 08:35:33 s501 daemon.notice easycwmpd: send empty message to the ACS
Mar 14 08:35:33 s501 daemon.notice easycwmpd: received GetParameterValues method from the ACS
Mar 14 08:35:33 s501 daemon.notice easycwmpd: external: execute get value Device.DeviceInfo.VendorConfigFile.
Mar 14 08:35:33 s501 daemon.notice easycwmpd: Fault in the param: Device.DeviceInfo.VendorConfigFile. , Fault code: 9005
Mar 14 08:35:33 s501 daemon.notice easycwmpd: send Fault: 9005: 'Invalid parameter name'
Mar 14 08:35:33 s501 daemon.notice easycwmpd: received GetParameterValues method from the ACS
Mar 14 08:35:33 s501 daemon.notice easycwmpd: external: execute get value Device.
Mar 14 08:35:35 s501 daemon.notice easycwmpd: send GetParameterValuesResponse to the ACS
Mar 14 08:35:36 s501 daemon.notice easycwmpd: receive empty message from the ACS
Mar 14 08:35:36 s501 daemon.notice easycwmpd: external: execute apply service
Mar 14 08:35:36 s501 daemon.notice easycwmpd: external script exit
Mar 14 08:35:36 s501 daemon.notice easycwmpd: end session success
```

discussion was started at the forum:
[(https://www.tapatalk.com/groups/freeacsforum/viewtopic.php?f=2&t=30&p=125#p125)]
 
following @pingwiniasty 's advice, I've added that string in freeacs-tr069/config/application-config.conf:

`quirks."TVIP STB" = ignorevendorconfigfile`

After that, error 9005 has disappeared, but the main problem is still actual - "The device does not support PII" error still appears.
I think I need some help.

pingwiniasty, 2019-03-15:
Please paste here conversation log of your device

anteok, 2019-03-15:
```
2019-03-15 10:36:12 [qtp1544703528-37] INFO  Conversation - ============== FROM CPE ( test ) TO ACS ===============
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
 <soap_env:Envelope
xmlns:soap_env="http://schemas.xmlsoap.org/soap/envelope/"
xmlns:soap_enc="http://schemas.xmlsoap.org/soap/encoding/"
xmlns:xsd="http://www.w3.org/2001/XMLSchema"
xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
xmlns:cwmp="urn:dslforum-org:cwmp-1-2">
  <soap_env:Header>
   <cwmp:ID soap_env:mustUnderstand="1">3</cwmp:ID>
  </soap_env:Header>
  <soap_env:Body>
   <cwmp:Inform>
    <DeviceId>
     <Manufacturer>TVIP</Manufacturer>
     <OUI>1027BE</OUI>
     <ProductClass>STB</ProductClass>
     <SerialNumber>1027BE0F0239</SerialNumber>
    </DeviceId>
    <Event soap_enc:arrayType="cwmp:EventStruct[1]">
     <EventStruct>
      <EventCode>2 PERIODIC</EventCode>
      <CommandKey />
     </EventStruct>
    </Event>
    <MaxEnvelopes>1</MaxEnvelopes>
    <CurrentTime>2019-03-15T07:36:10+00:00</CurrentTime>
    <RetryCount>0</RetryCount>
    <ParameterList soap_enc:arrayType="cwmp:ParameterValueStruct[28]">
     <ParameterValueStruct>
      <Name>Device.RootDataModelVersion</Name>
      <Value xsi:type="xsd:string">2.6</Value>
     </ParameterValueStruct>
     <ParameterValueStruct>
      <Name>Device.DeviceInfo.Description</Name>
      <Value xsi:type="xsd:string">TVIP S-Box s501</Value>
     </ParameterValueStruct>
     <ParameterValueStruct>
      <Name>Device.DeviceInfo.DeviceLog</Name>
      <Value xsi:type="xsd:string">
      /// huge syslog of CPE is following, I will send the whole file if you need ///
</Value>
     </ParameterValueStruct>
     <ParameterValueStruct>
      <Name>Device.DeviceInfo.HardwareVersion</Name>
      <Value xsi:type="xsd:string"></Value>
     </ParameterValueStruct>
     <ParameterValueStruct>
      <Name>Device.DeviceInfo.Manufacturer</Name>
      <Value xsi:type="xsd:string">TVIP</Value>
     </ParameterValueStruct>
     <ParameterValueStruct>
      <Name>Device.DeviceInfo.ManufacturerOUI</Name>
      <Value xsi:type="xsd:string">1027BE</Value>
     </ParameterValueStruct>
     <ParameterValueStruct>
      <Name>Device.DeviceInfo.MemoryStatus.Free</Name>
      <Value xsi:type="xsd:string">290496</Value>
     </ParameterValueStruct>
     <ParameterValueStruct>
      <Name>Device.DeviceInfo.MemoryStatus.Total</Name>
      <Value xsi:type="xsd:string">383168</Value>
     </ParameterValueStruct>
     <ParameterValueStruct>
      <Name>Device.DeviceInfo.ModelName</Name>
      <Value xsi:type="xsd:string">s501</Value>
     </ParameterValueStruct>
     <ParameterValueStruct>
      <Name>Device.DeviceInfo.ProcessStatus.ProcessNumberOfEntries</Name>
[... 102 lines cut ...]
		</cwmp:GetParameterValues>
</soapenv:Body>
</soapenv:Envelope>

2019-03-15 10:36:49 [qtp1544703528-40] INFO  Conversation - ============== FROM CPE ( test ) TO ACS ===============
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
 <soap_env:Envelope
xmlns:soap_env="http://schemas.xmlsoap.org/soap/envelope/"
xmlns:soap_enc="http://schemas.xmlsoap.org/soap/encoding/"
xmlns:xsd="http://www.w3.org/2001/XMLSchema"
xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
xmlns:cwmp="urn:dslforum-org:cwmp-1-2">
  <soap_env:Header>
   <cwmp:ID soap_env:mustUnderstand="1">FREEACS-1552635409784</cwmp:ID>
  </soap_env:Header>
  <soap_env:Body>
   <cwmp:GetParameterValuesResponse>
    <ParameterList soap_enc:arrayType="cwmp:ParameterValueStruct[0]" />
   </cwmp:GetParameterValuesResponse>
  </soap_env:Body>
 </soap_env:Envelope>

2019-03-15 10:36:49 [qtp1544703528-40] INFO  Conversation - =============== FROM ACS TO ( test ) ============

```

pingwiniasty, 2019-03-15:
Put it into pastebin or something. We need full log.

anteok, 2019-03-15:
[tr069-conversation.log](https://github.com/freeacs/freeacs/files/2969973/tr069-conversation.log)

anteok, 2019-03-15:
the log starts from rebooting CPE, initial connection and so on.

pingwiniasty, 2019-03-15:
Ok, give me some time, I need to analyse that log.

pingwiniasty, 2019-03-15:
Another question.
Are you have ACS configured in discovery mode or you added this device manually?

anteok, 2019-03-18:
I have added it manually, but my ACS is also configured in discovery mode.

pingwiniasty, 2019-03-18:
Can you delete whole unit and unit type and reboot device when you have discovery.mode enabled? 
ACS sending empty GetParameterNames Request, but CPE not respond on it. 
Also attach here conversation log.

anteok, 2019-03-19:
It seems to be it works correctly now - CPE has given lots of parameters. I could not make it to give it all manually.
[tr069-conversation.log](https://github.com/freeacs/freeacs/files/2981805/tr069-conversation.log)

pingwiniasty, 2019-03-19:
IMO discovery mode are best way to understand how ACS should work. But you devices not supports setting Periodic Inform by ACS. In previous unittype are you have any parameter of Device tree?

anteok, 2019-03-19:
No, I had not. I tried to add some manually, the manufacturer has a short how-to guide, but it was not successful at all.

anteok, 2019-03-19:
I think I'll close the issue. Thank you very much for spending time.

closed 2019-03-19 by anteok